MarlinFirmware/Marlin #12750: 3D touch ERROR ENDER 3 MARLIN TH3D U1.R2.7

Gomufas, 2018-12-28:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

I just instaled a 3D touch on my ender 3 but I havent been able to make it work 
BLtouch settings do appear in the LCD and i can stow and deploy.
when i send G28 it homes X,Y when Z is probing it stops and prompts the following error
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
I read that somethimes the probe isnt TRIGGERED so i tested with M119 and it was working

already tried using both custom probe and bltouch in the firmware but cant find the issue, i started following this guide 
https://www.youtube.com/watch?v=ocAeZRPVusw&t=620s
but ive read several guides trying to fix the issue
things that ive tried:
-#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN was already defined
-already changed the z offset for the probe
-simulated the bltouch using the Z endstop 
- I tried to enable pins_debugging but wasnt able to find it in the firmware

GMagician, 2018-12-28:
What result did you get with M119? Since trigger is very fast you should never see it

GMagician, 2018-12-28:
When Z start homing, can probe fully deploy?

Gomufas, 2018-12-28:
it seems to probe twice, the first time it did the second it didnt

GMagician, 2018-12-28:
So it deploy move Z down, trigger, stow, z raise, deploy again and move down then error?
Please detail sequence if not this way

Gomufas, 2018-12-28:
once it's centered it seems like it starts with a self-test, then deploys, moves down, trigger, moves up a little bit, then deploys(seems like it doesn't deploy enough to the full length), instantly stows. then moves up a little bit again

GMagician, 2018-12-28:
What about Z_CLEARANCE_DEPLOY_PROBE and Z_CLEARANCE_BETWEEN_PROBES values? maybe are to little to properly let probe deploy after trigger

Gomufas, 2018-12-28:
HAD 
  #define Z_CLEARANCE_DEPLOY_PROBE   5
  #define Z_CLEARANCE_BETWEEN_PROBES 3 
changed both to 10 
error changed 
Error:STOP called because of BLTouch error - restart with M999

after uploading, I did M502 and M500 as recommended in some threads 

just in case I've a piece of foam above the bed in case it doesn't stop on time

GMagician, 2018-12-28:
Such error raises when bltouch report triggered when it has to deploy.
If you execute M119 when stowed what do you get? and when manually deployed?

Gomufas, 2018-12-28:
echo:enqueueing "M280 P0 S90"
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED
ok
echo:enqueueing "M280 P0 S10"
Reporting endstop status
x_min: open
y_min: open
z_min: open
ok

Gomufas, 2018-12-28:
Don't know if it could be due to it not restarting the probe when moving to another location. at the moment after probing it goes up while led is blinking and then it resets before probing again. when it moves to the second location it seems like its not resetting the probe

GMagician, 2018-12-28:
That's not good. M119 should always return open

Gomufas, 2018-12-28:
I accidentally ordered two probes, I checked with both and it's not working. G28 and first corner work properly

Gomufas, 2018-12-29:
> That's not good. M119 should always return open
I put custom probe and manually changed some settings to BLTOUCH do not know if that could be the issue

GMagician, 2018-12-29:
Usually probe should report "open" because "triggered" lasts for 10ms only and you can't catch it.
Triggered is also reported when there is an error. Now, when you deploy manually led is blinking?

swilkens, 2018-12-30:
Have you verified with vanilla marlin in stead of TH3D?

Gomufas, 2019-01-03:
sorry for the late reply I was on vacation. I tried using vanilla marlin and it's working perfectly now 
thank you both for the help

closed 2019-01-03 by Gomufas