MarlinFirmware/Marlin #12751: [FR] EEPROM compatability check with new firmware

smellykid101, 2018-12-28:
### Description

Upon firmware change (ie: 1.1.8 -> 1.1.9) introduce a feature that will confirm the EEPROM has parameters stored in the same location. If not, suggest to the user to reinitialize in order to prevent errors.

### Steps to Reproduce

1. Install Marlin 1.1.8
2. Set new values
3. Save to EEPROM
4. Install Marlin 1.1.9 (bugfix)
5. Perform homing

**Expected behavior:**

Axis should home.

**Actual behavior:** 

Unexpected z-axis behavior; continuous movement away from z-min probe. X and Y axis do not move.

#### Additional Information

Hardware: FLSUN Cube printer (MKS Gen L) with BLTouch z-probe and TMC2130 (SPI) on all axis.

What fixed this problem for me was initializing EEPROM, restoring failsafe and storing these settings when using bugfix-1.1.9. I believe the problem was because settings are stored in different locations/formats within the EEPROM. Storing & checking the EEPROM methods revision number could prevent users from experiencing unexpected behaviour. This check would prompt the user to reinitialize for the appropriate EEPROM revision.

Example:
Old EEPROM:
[variable A, variable B, variable C , variable D]
New EEPROM:
[variable A, variable B, variable C1, variable C2 , variable D]
The new EEPROM would be expecting the value of variable C2 but receives the value of variable D from the old EEPROM's format style.

Proposal of new format:
[EEPROM_FORMAT_REVISION, variable A, variable B, variable C1, variable C2 , variable D]
Upon boot, the system would check EEPROM_FORMAT_REVISION and compare to the expected revision number. If different, block operations (or use failsafe settings) and report to user through serial debugging and LCD screens.

[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/2715669/ConfigFiles.zip)

boelle, 2019-02-21:
@smellykid101 problem solved?

smellykid101, 2019-02-21:
> @smellykid101 problem solved?

I haven't tried any recent nightly builds to see if this issue was fixed. I simply reinitialized the eeprom and restored the new firmware's defaults to fix this problem.

boelle, 2019-02-21:
oki, then please click green close button below

you can always open new issue if you have problems

smellykid101, 2019-02-21:
> oki, then please click green close button below
> 
> you can always open new issue if you have problems

This is a feature request. So that others do not experience bugs in the future.

AnHardt, 2019-02-21:
We have [EEPROM_VERSION](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/configuration_store.cpp#L40) since several years. But sometimes someone forgets to increase that number when the format is changed.

Requesting a already integrated feature does not make much sense.
We'll take it as a bug report.

smellykid101, 2019-02-22:
> We have [EEPROM_VERSION](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/configuration_store.cpp#L40) since several years. But sometimes someone forgets to increase that number when the [...]

Thank you for clearing that up. I'll close this issue. Thank you!

closed 2019-02-22 by smellykid101