MarlinFirmware/Marlin #12754: Delta printer problem - go down after G28 (-Z_PROBE_OFFSET_FROM_EXTRUDER)

e-Mole, 2018-12-28:
After homing (G28) my delta printer effector goes down (the downward shift is value of Z_PROBE_OFFSET_FROM_EXTRUDER).  But the printer height remains unchanged, so printing will start "under" the print bed (the print head crashes into the bed). I'm using a mechanical probe like allen key (Z_PROBE_OFFSET_FROM_EXTRUDER -3.90).
I changed this code in motion.cpp:
 
#
      #if HAS_BED_PROBE
        - zprobe_zoffset
      #endif

to older version:

 #
      #if HAS_BED_PROBE
        - zprobe_zoffset + Z_PROBE_OFFSET_FROM_EXTRUDER
      #endif

The problem disappeared.
Is this bug or my wrong configuration?

LVD-AC, 2018-12-30:
You need to re-calibrate the delta_height after the latest fix.

e-Mole, 2019-01-01:
@LVD-AC : Thanks for comment. I did the recalibration, but the result is the same. I did recalibrate the height classically with "paper test". Would it mean I have first to measure the distance of the probe and then subtract it from total height? This is very impractical... How to proceed correctly?

LVD-AC, 2019-01-02:
Issue a probe command G30 X0 Y0 and use the return value of that to set the DELTA_HEIGHT. Or use G33 P1 to do this automatically.

Only with printers that do not have a probe, that probe point is replaced by a paper test and the return value of the paper test is used. If you have a probe and still use a paper test instead of a probe point command to set the height you need to add (subtract for negative) the value of Z_PROBE_OFFSET_FROM_EXTRUDER to it.

boelle, 2019-02-21:
@e-Mole still a problem?

closed 2019-03-03 by thinkyhead