MarlinFirmware/Marlin #12755: probing offset calculated incorrectly

Dloranger, 2018-12-29:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

alekc, 2019-01-06:
I encountered the same issue, and I think you might have an error on ```Y_BED_SIZE - MIN_PROBE_EDGE - Y_PROBE_OFFSET_FROM_EXTRUDER``` formula.

I think it should be ```Y_BED_SIZE - MIN_PROBE_EDGE + Y_PROBE_OFFSET_FROM_EXTRUDER``` because if the sensor is in front of the nozzle, it should be a negative value. It will be able to reach all positions in front of the hotbed, but it wont reach the back, so if the delta between sensor and nozzle is -55, for back probe it's 220-10(min_probe_edge) + (-55)

boelle, 2019-02-21:
@Dloranger still a problem?

Dloranger, 2019-02-21:
I edited my code at the time to account for the discrepancy and have not reprogrammed my printer since so I can't comment if the issue still existing in the source code.

The issue should be easily reproducible for testing purposes.  I may be able to do it in a few weeks, I am preparing to upgrade my hardware to add more motors and will need to reflash the controller (ramps) when the new hardware is installed.

boelle, 2019-02-21:
maybe click the green close button and open a new issue in a few weeks if problem is still there

remember to use bugfix 2.0

Dloranger, 2019-02-21:
I just reviewed the source code and I don't see that it has been changed to account for the error, based on this I will say the issue is still present

boelle, 2019-06-22:
missing label

shitcreek, 2019-07-10:
Have you checked with 2.0?

Dloranger, 2019-07-11:
The code mentoned remains unchanged in the 2.0 branch so I have to assume the bug is still there.

thinkyhead, 2019-07-11:
The values in the configuration are intended to be changed, if and only if the values set automatically by `Conditionals_post.h` are not exactly right for your needs. In that case, they should be changed conscientiously and not just un-commented without thought. Otherwise the probe will go off the bed.

closed 2019-07-11 by thinkyhead