MarlinFirmware/Marlin #12756: Communicate with i2c twibus, Arduino to Arduino

bulior, 2018-12-29:
Is there a way to send gcodes like Homing (G28 X) from one Arduino or Microcontroller to An Arduino Mega with Marlin over i2c? 

Until now we can send something and Marlin would receive it but the Command wouldn't be executed. The received string would be printed in Serialmotior. How can we execute the received String? 

Best regards 

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

bulior, 2018-12-29:
Is this a void and in which library is it?
Twinbus.cpp?

GMagician, 2018-12-29:
You have to modify marlin firmware to get info via i2c and call such function when you have read commands

bulior, 2018-12-30:
Thanks @GMagician!

We have solved the problem and now we can receive Gcodes with an i2c Master.  
We customized twibus.cpp TWIBUS::echdata with your command:

> // static
> void TWIBus::echodata(uint8_t bytes, const char prefix[], uint8_t adr) {
>   echoprefix(bytes, prefix, adr);
> 
>    String gcode = "";
>   
>   while (bytes-- && Wire.available())
>   {
>     gcode = gcode + char(Wire.read());
>     //SERIAL_CHAR(Wire.read());
>   }
>   
>    int gcode_length = gcode.length() + 1;
>    char gcode_array[gcode_length];
>    gcode.toCharArray(gcode_array, gcode_length);
>    enqueue_and_echo_command(gcode_array);
> 
>   SERIAL_EOL();
> 
> }

miltostsalekkas, 2018-12-30:
Thank you all very much.It works fine!

boelle, 2019-02-21:
@bulior problem fixed?

miltostsalekkas, 2019-02-21:
Yes.

boelle, 2019-02-21:
click green close button below then

closed 2019-02-22 by bulior