MarlinFirmware/Marlin #12757: re-arm pins file tmc2208

boelle, 2018-12-29:
quick Q:

are there any reason for tmc2208 not being in the pins file yet?

i ASSUME that marlin does not support the ethernet addon and was going to use those pins for the 2208's, but are there anything wrong in doing so?

gloomyandy, 2018-12-29:
Do you have things working with the pins defined above? I thought that you needed an interrupt capable pin for software serial input (and I think only port 0 and port 1 have interrupt capable pins).

boelle, 2018-12-29:
nope, it will be another 4 days before i have the re-arm in my hands

just preparing things and making a list of other parts i need

gloomyandy, 2018-12-29:
Sorry there is a typo in my post above, only ports 0 and 2 have gpio interrupt pins, so I'm not sure if the ones you have selected will work or not. I'm sure I read something that said the RX pin must be able to generate an interrupt (which makes sense).

boelle, 2018-12-29:
yeah, and it also explains why 2208's have not beed added to the list :-( a shame thou as these runs cooler and can handle more current compared to the 2130's

but well i will then order the 2130's as that is the only option of semi silent operation

Nuttavoot, 2018-12-30:
So Re-ARM can not be use with TMC2208, just know that, I thought it was a drop-in compatible, almost buy them.

gloomyandy, 2018-12-30:
You can use a TMC2208 with Re-arm no problem if you use it in stand alone mode. If you want to use the more advanced options then you/someone needs to work out how to connect it to the ReArm this may or may not be possible depending upon what other features you are using (as there may not be enough interrupt enabled pins available to allow the use of the serial connections). How easy/hard this is will depend upon how many TMC2208s you want to use and what other devices you have connected (as things like displays etc. will use some of the pins). At the moment it seems that no-one has worked out a good configuration.

Nuttavoot, 2018-12-30:
> You can use a TMC2208 with Re-arm no problem if you use it in stand alone mode. If you want to use the more advanced options then you/someone needs to work out how to connect it to the ReArm this [...]

Do I need to change the driver to TMC2208 in the configuration.h file ? I tried that and it compiled fail with error like TMC2208 needs some pins.

gloomyandy, 2018-12-30:
If you are using the TMC2208 in standalone mode you need to define it as TMC2208_STANDALONE not as TMC2208

closed 2018-12-30 by boelle

necris38, 2019-02-01:
Hello,
I try (hard!) for a week to configure my Re-arm/RAMPS1.6 board with UART TMC2208.
Did you finally successfull with your configuration ?
I think we can find free pin on re-arm for RX/TX but I don't know where define it as in "pins_RAMPS_RE_ARM.h" there is no more "#if HAS_DRIVER(TMC2208)" section.
a little help would be appreciated because I'm starting to get a little confuse !

boelle, 2019-02-01:
it's not possible, there are not enough interrupt able pins left

at most you can drive 1 stepper with serial

i have ditched the idea and will... at some point.... go for tmc2130

boelle, 2019-02-01:
there are plenty of free pins... but they are not interrupt able :-(

necris38, 2019-02-01:
Arff :-(
what a pitty !
Standalone mode is so restrictive...
Do you think HYBRID_THRESHOLD is usuable in STANDALONE mode ?

necris38, 2019-02-01:
If I want to drive 1 stepper, wich pin is interrupt capable ?

boelle, 2019-02-01:
no idea.... and i did not investigate any further since only 1 stepper is a waste of time

necris38, 2019-02-01:
OK.
Thanks for your answer.

necris38, 2019-02-01:
in Datasheet https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf
you can read:
[image]

So... ??

boelle, 2019-02-01:
your guess is as good as mine, maybe @thinkyhead knows

necris38, 2019-02-02:
[image]

gloomyandy, 2019-02-02:
Yes we are well aware of how the various pins can be used. The issue is are any suitable pins available on the re-arm/ramps board (especially if you have other devices like an LCD attached) and if they are does the software serial and TMC library actually work correctly. I don't think anyone has actually come up with an answer for this so far. Feel free to investigate and post your results.

necris38, 2019-02-02:
I would love to investigate too but I am a little beginner. Do you know in which file I have to declare the TMC2208 library to start my research?

gloomyandy, 2019-02-02:
The TMC library is already included, it will be used when define a TMC device in your configuration, so you don't need to declare it anywhere. If you want to look at the source code for that library you can find it here: https://github.com/teemuatlut/TMCStepper
Be aware though that Marlin is a pretty complex project and may not be easy to understand if you have not had much experience with this sort of hardware and software.

necris38, 2019-02-02:
No experience... but very curious ! 😄

chrisqwertz, 2019-04-25:
I got 4x TMC2208 working on my RAMPS1.4 + Re-Arm setup!

chrisqwertz, 2019-04-25:
Very important: Make sure that you use the latest TMCStepper master.zip !
https://github.com/teemuatlut/TMCStepper/archive/master.zip

The half duplex mode has been added just recently (d27257900a14ee8cd8df7d5c9b01f1f60c3c4d0b)

ej0rge, 2019-05-04:
> I got 4x TMC2208 working on my RAMPS1.4 + Re-Arm setup!
>...
> Can we mark this as solved?

Just successfully testing a similar setup here with 3x tmc2208 and an lcd-for-melzi hooked up as though it were an rrd full graphic controller (which needed P2_11) 

#if HAS_DRIVER(TMC2208)
  /**
   * TMC2208 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */

  /**
   * Software serial
   */

   // P2_08 E1-Step
   // P2_13 E1-Dir

  #define X_SERIAL_TX_PIN    P2_06
  #define X_SERIAL_RX_PIN    P2_06

  #define Y_SERIAL_TX_PIN    P2_13 // E1-Dir
  #define Y_SERIAL_RX_PIN    P2_13

  #define Z_SERIAL_TX_PIN    P2_08 // E1-Step
  #define Z_SERIAL_RX_PIN    P2_08

#endif