MarlinFirmware/Marlin #12759: X-axis not working after auto home prior to printing

DavidThijs, 2018-12-29:
### Description

A G28 command in printer code isn't correctly executed when the printer was homed manually from the display menu prior to printing and then a manual move of x axis was performed from the menu. This issue was not present in Marlin 1.9 but I did found it in the Bugfix 2.0 branch.

### Steps to Reproduce

Printer configured with safe homing on in the middle of the bed and BLtouch is configured

1. Auto Home printer from menu
2. Move x axis to a random position using the display menu function to move the x-axis
2. Print any file that contains a G28 start instruction followed by a G29 bilinear bed leveling as startup instruction

**Expected behavior:

Printer executes a proper G28 command that includes X axis and Y axis movement to the center of the bed and then probes for bed Z level prior to do a bilinear bed leveling and printing the object.

**Actual behavior:

Printer moves Z axis and then probes at the actual position of the X axis, altough the X-axis wasn't homed prior to probing. Worst case result is that the probe crashes at the last known location which in my case was left of the bed bending the structure until the Z steppers give up.

When only executing a G29 instruction after a auto-home, the printer probes 3 times on the same x-axis position but for 3 different Y positions. So it doesn't move away from the middle of the bed to probe left and right for each of the Y positions.

#### Additional Information

[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/2716673/Marlin_config.zip)
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/2716675/pins_RAMPS.zip)

shitcreek, 2019-01-05:
Have you checked this page: http://marlinfw.org/docs/features/auto_bed_leveling.html ?

boelle, 2019-02-21:
@DavidThijs problem fixed?

DavidThijs, 2019-02-23:
@boelle Yes, I think it is and I believe the issue was the communication with the TMC2130. I replaced the wiring for the SPI and since then I didn't observe the behaviour anymore. But I'm not sure. Anyway, this thread can be closed now.

closed 2019-02-23 by DavidThijs