MarlinFirmware/Marlin #12760: Not an issue just want to know how

tibuck, 2018-12-29:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

tibuck, 2018-12-29:
How do i add a Mcode or Gcode to invert stepper motor direction on the fly ?

GMagician, 2018-12-29:
Some slicer let you mirror you object (Cura)

GMagician, 2018-12-29:
once you mirror your object you may use G92 in your gcode to shift origin

GMagician, 2018-12-29:
I think there is nothing in marlin that let you switch on fly motor direction

tibuck, 2018-12-29:
I already use the G92 to set the new 0.0 as a offset but regardless, the printers movements will always be in the direction set by the firmware, Marlin in this case, "There is nothing in marlin" YET, there is nothing till, someone asks or needs it lol, But im sure there probably was at one point,

Knaudler, 2019-01-07:
hi
only way in marlin is m92 with negative values, but dont think it works on the fly

tibuck, 2019-01-07:
> hi
> only way in marlin is m92 with negative values, but dont think it works on the fly

No it doesn't work for what i need, But i made a temp fix by adding a cable connector to that motor, I just flip the connections when i need it to go one way and flip it round to go the other. Not ideal but works till someone smarter then me can figure out how to make this a custom Gcode. But thank you for the suggestion nonetheless.

Roxy-3D, 2019-01-09:
One way to do it would be to tweak the macros used to feed steps to the stepper motors.  The IDEX machines have a 'Duplication Mode' that feeds pulses to both (independent) extruders.    If you look at the macros to see how that is done...    It is almost trivial to make the macros feed pulses that move the extruder(s) in an opposite direction.

The big problem is when you flip directions, you might want to update (or mirror) your current position so that the end stops are the 'correct' distance from where ever the nozzle is located.

closed 2019-03-03 by thinkyhead