MarlinFirmware/Marlin #12762: Delta printer problem - "stuttering" in peripheral bed zones

e-Mole, 2018-12-29:
### Description

When fast moving in peripheral zones (delta printer) - "stuttering" occurs (strange creaking, skipping steps (?)) - the exact position is lost. But this occurs only near of bed edge.
Tested with ver. 2.0.x and 1.1.9 (this problem does not happen in version 1.1.9).

### Steps to Reproduce

HW config: Arduino MEGA + RAMPS 1.4 + A4988,
bed diameter 265 mm,
diag_rod 278 mm,
segments/s 100

F8000
G0 Z100 X0Y-120 F8000 (stuttering)
G0 Z100 X0Y-110 F8000 (o.k)

F6000
G0 Z100 X0Y-120 F6000 (stuttering)
G0 Z100 X0Y-110 F6000 (o.k)

F4000
G0 Z100 X0Y-120 F4000 (o.k)
G0 Z100 X0Y-110 F4000 (o.k)

Where could the problem be? Something in the configuration or bug in code?
[Marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/2716724/Marlin-config.zip)

e-Mole, 2019-01-03:
Please, have any idea?

boelle, 2019-02-21:
@e-Mole 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.
Before posting a bug report please test with `bugfix-2.0.x` to check if problem is gone

thinkyhead, 2019-03-03:
It sounds like planner starvation. All you can really do is lower your print speed until the symptom abates.

closed 2019-03-03 by thinkyhead

e-Mole, 2019-03-10:
@thinkyhead Unfortunately, the current maximum speed on my deltas is 60 mm / sec, making printing slower by half. That's really bad.

boelle, 2019-03-10:
> All you can really do is lower your print speed

or you can upgrade to re-arm instead of Mega (the mega is VERY SLOW)

e-Mole, 2019-05-03:
@boelle Hi, today I tested last 2.x version and problem with "stuttering" is gone! Thanks to all developers :-).