MarlinFirmware/Marlin #12764: TMC2130 and Marlin 2.0 can't compile

pulster, 2018-12-29:
### Description

I'm trying to compile Marlin 2.0 and activating TMC2130 on Y axis. I cannot compile. I've included the error. 

I can compile if I use TMC2130_STANDALONE, but not when using TMC2130.

My pins file for GT2560_REV_A is the standard one given with Marlin, with removed Y_MAX_PIN and replaced by Y_CS_PIN.

#### Additional Information

[Configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/2716768/Configuration_files.zip)

Here is the error message:

Arduino: 1.9.0-beta (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/src/inc/../HAL/HAL_AVR/HAL.h:36:0,
                 from sketch/src/inc/MarlinConfig.h:30,
                 from sketch/src/Marlin.h:24,
                 from sketch/src/Marlin.cpp:31:
sketch/src/inc/../pins/pins_GT2560_REV_A.h:96:33: error: pasting "/* Must be enabled at startup to keep power flowing*/" and "_DDR" does not give a valid preprocessing token
 #define SUICIDE_PIN        54   // Must be enabled at startup to keep power flowing
                                 ^
sketch/src/inc/../HAL/HAL_AVR/fastio_AVR.h:82:42: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                          ^
sketch/src/inc/../HAL/HAL_AVR/fastio_AVR.h:100:35: note: in expansion of macro 'SET_OUTPUT'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                   ^
sketch/src/Marlin.h:242:27: note: in expansion of macro 'OUT_WRITE'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                           ^
sketch/src/Marlin.h:242:37: note: in expansion of macro 'SUICIDE_PIN'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                                     ^
sketch/src/inc/../pins/pins_GT2560_REV_A.h:96:33: error: pasting "/* Must be enabled at startup to keep power flowing*/" and "_PIN" does not give a valid preprocessing token
 #define SUICIDE_PIN        54   // Must be enabled at startup to keep power flowing
                                 ^
sketch/src/inc/../HAL/HAL_AVR/fastio_AVR.h:82:61: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                                             ^
sketch/src/inc/../HAL/HAL_AVR/fastio_AVR.h:100:35: note: in expansion of macro 'SET_OUTPUT'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                   ^
sketch/src/Marlin.h:242:27: note: in expansion of macro 'OUT_WRITE'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                           ^
sketch/src/Marlin.h:242:37: note: in expansion of macro 'SUICIDE_PIN'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                                     ^
In file included from sketch/src/inc/../HAL/HAL_AVR/HAL.h:36:0,
                 from sketch/src/inc/MarlinConfig.h:30,
                 from sketch/src/Marlin.h:24,
                 from sketch/src/Marlin.cpp:31:
sketch/src/inc/../pins/pins_GT2560_REV_A.h:96:33: error: pasting "/* Must be enabled at startup to keep power flowing*/" and "_RPORT" does not give a valid preprocessing token
 #define SUICIDE_PIN        54   // Must be enabled at startup to keep power flowing
                                 ^
sketch/src/inc/../HAL/HAL_AVR/fastio_AVR.h:77:48: note: in definition of macro '_WRITE'
 #define _WRITE(IO,V)          do{ if (&(DIO ## IO ## _RPORT) < (uint8_t*)0x100) _WRITE_NC(IO,V); else _WRITE_C(IO,V); }while(0)
                                                ^
sketch/src/inc/../HAL/HAL_AVR/fastio_AVR.h:100:51: note: in expansion of macro 'WRITE'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                                   ^
sketch/src/Marlin.h:242:27: note: in expansion of macro 'OUT_WRITE'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                           ^
sketch/src/Marlin.h:242:37: note: in expansion of macro 'SUICIDE_PIN'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                                     ^
sketch/src/inc/../pins/pins_GT2560_REV_A.h:96:33: error: pasting "/* Must be enabled at startup to keep power flowing*/" and "_WPORT" does not give a valid preprocessing token
[... 106 lines cut ...]
                                     ^
sketch/src/inc/../pins/pins_GT2560_REV_A.h:96:33: error: pasting "/* Must be enabled at startup to keep power flowing*/" and "_PIN" does not give a valid preprocessing token
 #define SUICIDE_PIN        54   // Must be enabled at startup to keep power flowing
                                 ^
sketch/src/inc/../HAL/HAL_AVR/fastio_AVR.h:74:48: note: in definition of macro '_WRITE_C'
   DIO ## IO ## _RPORT = port_bits & _BV(DIO ## IO ## _PIN); /* Atomically toggle the output port bits */ \
                                                ^
sketch/src/inc/../HAL/HAL_AVR/fastio_AVR.h:89:31: note: in expansion of macro '_WRITE'
 #define WRITE(IO,V)           _WRITE(IO,V)
                               ^
sketch/src/inc/../HAL/HAL_AVR/fastio_AVR.h:100:51: note: in expansion of macro 'WRITE'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                                   ^
sketch/src/Marlin.h:242:27: note: in expansion of macro 'OUT_WRITE'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                           ^
sketch/src/Marlin.h:242:37: note: in expansion of macro 'SUICIDE_PIN'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                                     ^
In file included from sketch/src/module/stepper.h:232:0,
                 from sketch/src/Marlin.cpp:36:
sketch/src/module/stepper_indirection.h:59:26: fatal error: TMCStepper.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

pulster, 2018-12-29:
Nope, running IDE 1.9.0 (requirement for for Marlin 2.0).

Afrobaron, 2018-12-29:
try moving the comments anyway.  I would bet it might do the trick.

pulster, 2018-12-29:
You're right, it worked!

Thank you very much!

closed 2018-12-29 by pulster