MarlinFirmware/Marlin #12766: Sanity Check error similar to "[2.0.x] Fix bad sanitycheck" #12748

xeddog, 2018-12-30:
### Description

As title says.  Build using platformio fails when G26 MESH validation is enabled and "AUTO_BED_LEVELING_BILINEAR" is also enabled:  Screenshot included.

### Steps to Reproduce
Just run Build in platformio again

**Expected behavior:** [What you expect to happen]
Successful Build
**Actual behavior:** [What actually happens]
Build fails as shown in screenshot

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.

[image]

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2716834/Configuration.zip)

Roxy-3D, 2018-12-30:
The Sanity Check used to be correct and allow G26 to be built in the Bi-Linear bed leveling systems.    Some thing (simple???) must of broke...

GMagician, 2018-12-30:
It's the same error I got and I hope it has been fixed by my PR

xeddog, 2018-12-30:
The only reason I opened this issue is because #12748 Specifically talked about MESH levelling and I just wanted to make sure BiLinear was also included.

boelle, 2019-02-21:
@xeddog problem fixed?

xeddog, 2019-02-22:
I completely forgot this was even opened.  Sorry.  And yes, the problem was resolved.

closed 2019-02-22 by xeddog