MarlinFirmware/Marlin #12771: Marlin 1.1.9. - Repetier-Host/Server/Slic3r echo:Unknown command: "M73 P27 R69 Q27 S69"

Ianmcmill, 2018-12-30:
P3Steel
Repetier-Host 2.1.3
Repetier-Server 0.90.7
Marlin 1.1.9
Slic3r 1.3.1-dev

While printing I get this message in Repetier-Host 
`echo:Unknown command: "M73 P27 R69 Q27 S69"`

Print is running fine.

Larger log:

```
16:43:48.043 : echo:Unknown command: "M73 P28 R68 Q28 S68"
16:43:52.085 : X:71.42 Y:124.27 Z:6.45 E:0.00 Count X:8659 Y:14426 Z:25840
16:43:52.526 : X:75.17 Y:79.59 Z:6.45 E:0.00 Count X:8388 Y:14827 Z:25840
16:43:55.256 : echo:Unknown command: "M73 P28 R68 Q28 S68"
16:44:03.136 : echo:Unknown command: "M73 P28 R68 Q28 S68"
16:44:04.854 : Slic3r command:C:\Program Files\Repetier-Host\Slic3r\slic3r-console.exe --gui
16:44:08.932 : <Slic3r> Argument "" isn't numeric in numeric eq (==) at C:/PROGRA~1/REPETI~1/Slic3r/lib/Slic3r/GUI/MainFrame.pm line 103.
16:44:11.336 : echo:Unknown command: "M73 P28 R68 Q28 S68"
16:44:19.095 : X:131.53 Y:92.94 Z:6.45 E:0.00 Count X:15025 Y:11068 Z:25840
16:44:19.657 : echo:Unknown command: "M73 P28 R68 Q28 S68"
16:44:26.426 : echo:Unknown command: "M73 P28 R67 Q28 S67"
16:44:28.625 : X:131.53 Y:124.27 Z:6.45 E:0.00 Count X:14659 Y:14437 Z:25840
16:44:29.266 : X:125.15 Y:79.49 Z:6.45 E:0.00 Count X:14661 Y:14757 Z:25840
16:44:31.794 : X:132.51 Y:73.42 Z:6.45 E:0.00 Count X:15068 Y:9744 Z:25840
16:44:32.215 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:44:32.885 : X:128.00 Y:73.44 Z:6.45 E:0.00 Count X:14038 Y:9996 Z:25840
16:44:34.403 : X:71.93 Y:73.44 Z:6.45 E:0.00 Count X:8997 Y:9878 Z:25840
16:44:34.513 : X:75.06 Y:78.94 Z:6.45 E:0.00 Count X:8799 Y:9743 Z:25840
16:44:36.926 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:44:37.037 : X:75.09 Y:67.32 Z:6.45 E:0.00 Count X:8684 Y:9151 Z:25840
16:44:39.663 : X:62.91 Y:66.89 Z:6.45 E:0.00 Count X:7909 Y:9272 Z:25840
16:44:42.616 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:44:45.024 : X:137.09 Y:68.74 Z:6.75 E:0.00 Count X:14910 Y:9274 Z:25840
16:44:45.134 : Printing layer 22 of 93
16:44:45.244 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:44:46.893 : X:130.00 Y:87.95 Z:6.75 E:0.00 Count X:14487 Y:11613 Z:27040
16:44:48.324 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:44:48.555 : X:72.10 Y:90.05 Z:6.75 E:0.00 Count X:8292 Y:11393 Z:27040
16:44:53.122 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:44:55.953 : X:75.58 Y:87.51 Z:6.75 E:0.00 Count X:14302 Y:11711 Z:27040
16:44:56.493 : X:73.31 Y:74.57 Z:6.75 E:0.00 Count X:11972 Y:11213 Z:27040
16:44:59.566 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:45:06.796 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:45:07.658 : X:124.94 Y:78.94 Z:6.75 E:0.00 Count X:14016 Y:9898 Z:27040
16:45:07.985 : X:127.80 Y:73.55 Z:6.75 E:0.00 Count X:13991 Y:10084 Z:27040
16:45:08.202 : X:132.53 Y:73.67 Z:6.75 E:0.00 Count X:13995 Y:10292 Z:27040
16:45:11.275 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:45:13.684 : X:128.41 Y:120.81 Z:6.75 E:0.00 Count X:14665 Y:13856 Z:27040
16:45:19.473 : echo:Unknown command: "M73 P29 R67 Q29 S67"
16:45:21.436 : X:66.69 Y:81.88 Z:6.75 E:0.00 Count X:9650 Y:11121 Z:27040
16:45:22.314 : X:75.86 Y:81.55 Z:6.75 E:0.00 Count X:8644 Y:10687 Z:27040
16:45:26.361 : X:68.40 Y:121.56 Z:6.75 E:0.00 Count X:8333 Y:13602 Z:27040
16:45:27.234 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:45:29.201 : X:66.69 Y:74.28 Z:6.75 E:0.00 Count X:7892 Y:10311 Z:27040
16:45:30.964 : X:68.40 Y:123.89 Z:6.75 E:0.00 Count X:8625 Y:14789 Z:27040
16:45:32.485 : X:72.44 Y:73.55 Z:6.75 E:0.00 Count X:8923 Y:9911 Z:27040
16:45:32.707 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:45:34.013 : X:131.67 Y:122.85 Z:6.75 E:0.00 Count X:14375 Y:14680 Z:27040
16:45:39.046 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:45:39.371 : X:62.84 Y:68.55 Z:6.75 E:0.00 Count X:7911 Y:9089 Z:27040
16:45:44.514 : X:137.17 Y:66.96 Z:7.05 E:0.00 Count X:15133 Y:9260 Z:27040
16:45:44.632 : Printing layer 23 of 93
16:45:44.745 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:45:46.417 : X:130.00 Y:87.95 Z:7.05 E:0.00 Count X:14487 Y:11613 Z:28240
16:45:47.938 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:45:47.938 : X:72.10 Y:90.05 Z:7.05 E:0.00 Count X:8292 Y:11393 Z:28240
16:45:52.613 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:45:55.342 : X:75.58 Y:87.51 Z:7.05 E:0.00 Count X:14302 Y:11711 Z:28240
16:45:55.894 : X:73.31 Y:74.57 Z:7.05 E:0.00 Count X:12015 Y:11212 Z:28240
16:45:58.956 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:46:06.075 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:46:11.095 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:46:11.529 : X:131.08 Y:122.07 Z:7.05 E:0.00 Count X:14334 Y:13746 Z:28240
16:46:14.385 : X:134.01 Y:73.59 Z:7.05 E:0.00 Count X:14964 Y:10403 Z:28240
16:46:18.753 : echo:Unknown command: "M73 P30 R66 Q30 S66"
16:46:20.822 : X:75.11 Y:79.32 Z:7.05 E:0.00 Count X:9217 Y:10693 Z:28240
16:46:24.004 : X:68.33 Y:95.59 Z:7.05 E:0.00 Count X:8086 Y:11515 Z:28240
16:46:24.873 : echo:Unknown command: "M73 P31 R66 Q31 S66"
16:46:27.062 : X:70.33 Y:122.08 Z:7.05 E:0.00 Count X:8333 Y:13671 Z:28240
16:46:29.249 : X:75.06 Y:78.94 Z:7.05 E:0.00 Count X:8970 Y:9915 Z:28240
16:46:30.573 : X:71.59 Y:123.45 Z:7.05 E:0.00 Count X:8375 Y:14745 Z:28240
16:46:30.903 : echo:Unknown command: "M73 P31 R65 Q31 S65"
16:46:32.206 : X:131.59 Y:123.45 Z:7.05 E:0.00 Count X:14375 Y:14745 Z:28240
16:46:37.244 : X:62.91 Y:66.89 Z:7.05 E:0.00 Count X:7918 Y:9257 Z:28240
16:46:37.244 : echo:Unknown command: "M73 P31 R65 Q31 S65"
16:46:42.478 : X:137.09 Y:68.59 Z:7.35 E:0.00 Count X:14916 Y:9259 Z:28240
16:46:42.478 : Printing layer 24 of 93
16:46:42.590 : echo:Unknown command: "M73 P31 R65 Q31 S65"
16:46:44.358 : X:130.00 Y:87.95 Z:7.35 E:0.00 Count X:14487 Y:11613 Z:29440
16:46:45.761 : echo:Unknown command: "M73 P31 R65 Q31 S65"

```

AnHardt, 2018-12-30:
https://reprap.org/wiki/G-code#M73:_Set_build_percentage -> Don't care.

swilkens, 2018-12-30:
Support for M73 isn't enabled by default in Marlin. You can enable it in `Configuration_adv.h`

Change
```
// Add an 'M73' G-code to set the current percentage
//#define LCD_SET_PROGRESS_MANUALLY
```

To
```
// Add an 'M73' G-code to set the current percentage
#define LCD_SET_PROGRESS_MANUALLY
```

Ianmcmill, 2018-12-30:
Thanks! Good to know.

closed 2018-12-30 by Ianmcmill