MarlinFirmware/Marlin #12773: Faulty Tronxy X5S configuration

themotty, 2018-12-30:
The default configuration for the Tronxy X5S does not work on a recently bought unit.  The LCD content is corrupted. I don't know if this affects older units or not.

The default config selects REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER but correctly it should be LCD_FOR_MELZI.

Other less critical corrections in the config:
#define POWER_SUPPLY 0
#define TEMP_SENSOR_0 501
#define X_BED_SIZE 330
#define Y_BED_SIZE 334

In the attached configuration.h I have some additional changes too.

[Configuration.Tronxy_X5S.h.zip](https://github.com/MarlinFirmware/Marlin/files/2717420/Configuration.Tronxy_X5S.h.zip)

ej0rge, 2019-01-01:
Should also comment out #define DISPLAY_CHARSET_HD44780 JAPANESE since it doesn't do anything for graphical lcds and takes up precious space that could be used for other features on a 128k mcu. 

Tronxy used to say that "11" was the correct thermistor for their printers but that was before 501 existed so *shrug*.

themotty, 2019-01-01:
Thank you for the response.

I checked the temperature measurement accuracy by measuring the heater block with an infra thermometer. I think with the original config (thermistor type 1) it over reads. Comparing the thermistor tables at 200C, type 11 would be over reading even more. Compared to type 1 type 501 lowers the temperature by 13-14C at 200C which seems to be spot on.

I tried commenting DISPLAY_CHARSET_HD44780, but it didn't seem to do anything regarding free flash space. It seems that this define is not used with graphical displays.

ej0rge, 2019-01-01:
Right, it does nothing but take up space if you aren't using a character display, but it's enabled by default in a lot of configurations. 

I just mean, for the sake of clarity, it should be disabled in example configurations for specific printers that we know don't use character displays. Particularly printers with 128k controllers, in case people want to play with linear advance, add a bltouch, etc.

themotty, 2019-01-01:
What I meant is that based on Your suggestion I did try commenting the #define DISPLAY_CHARSET_HD44780, but it made no change to the used flash space as reported by Arduino. Seems like this define is irrelevant in case of a graphical display. According to the SanityCheck.h this define is only required for the ULTRA_LCD display, but I didn't go into more depth.

I have access to thermal camera and another type of IR thermometer. I will check again the accuracy of the hotend temperature measurement on the weekend.

ej0rge, 2019-01-02:
Oh ok. I hadn't checked that. 

I don't know much about FLIR imaging but I do know that different materials have different coefficients of emission when it comes to heat vs. ir output. Most infrared thermometers are preset to 0.95 but rough aluminum is about 0.07 and polished aluminum is about 0.3. 

The point is that, best case scenario with an ir thermometer, there is still some math that has to be done to arrive at a realistic number. And even then they are not what you'd call reference accurate. 

I actually know the quality control lead at fluke's thermal metrology division so there's an outside chance i could actually get one of the thermistors that came with my tronxy measured with world class equipment, but i fear that it may be a fool's errand because i don't imagine that they are super consistent with their sourcing.

boelle, 2019-02-21:
@themotty problem fixed?

themotty, 2019-02-21:
Yes I think so. In my issue report I didn't really ask for help, I tried to submit correction to the X5S default config, which I think should be included.

boelle, 2019-02-21:
oki please click green close button below

closed 2019-02-22 by themotty