MarlinFirmware/Marlin #12779: MKS Sbase v1.3 marlin usb driver

Thowra, 2018-12-31:
### Description
I installed Markin 2.0 on the MKS Sbase motherboard (LPC1768).
Windows shows "USB device marlin".
I've tried several drivers, but it doesn't work with one of them.
Operating System: Windows 7.
Which driver is good for it?

[image]

gloomyandy, 2018-12-31:
You could try installing the smoothieware driver... https://github.com/MarlinFirmware/Marlin/issues/8618

The board is recognised automatically under Windows 10 (and Linux based systems).

Thowra, 2018-12-31:
hi,
I tried this driver, but not working.
This link is died
http://marlinfw.org/docs/basics/install_arm.html

gloomyandy, 2018-12-31:
Did you look at the issue I pointed you at above? That has a link to the smoothieware driver...
http://smoothieware.org/windows-drivers

gloomyandy, 2018-12-31:
If you have solved your problem please be sure to close the issue.

closed 2018-12-31 by Thowra