MarlinFirmware/Marlin #12780: Autohome after disabled stepper motors

markdesouza, 2019-01-01:
### Description

If the stepper motors have been disabled, its reasonable to assume the bed or print head has been manually moved ie x, y and z should be viewed as undefined.
This has implications for instance when performing a filament change, and the printer attempts to home to the top left automatically.

As an aside I am unsure why the printer homes to the top left as opposed to the autohome location of bottom left when performing a filament change.
 
### Steps to Reproduce

1. Turn on printer (head is at bottom left)
2. Disable stepper motors
3. Manually move bed so print head is at top left (ie to clean the bed)
3. Select filament change

**Expected behavior:** [What you expect to happen]

Printer should perform an autohome to verify its position

**Actual behavior:** [What actually happens]

Printer over estimates the amount of distance to be travelled and grinds the belts

#### Additional Information

N/A

boelle, 2019-02-21:
@markdesouza problem fixed?

markdesouza, 2019-02-22:
@boelle I was running  v1.1.9 when I reported the issue and haven't tried the development branch as I'm using the printer for work. Happy to test if there is a new release

boelle, 2019-02-22:
try bugfix 2.0

boelle, 2019-03-06:
@thinkyhead i think we can close this one

markdesouza, 2019-03-07:
please do, I'm traveling atm and haven't had the chance to test.

boelle, 2019-03-07:
@markdesouza only you can close by clicking the close button below

you can always open new issue later

closed 2019-03-07 by markdesouza