MarlinFirmware/Marlin #12781: [2.0.x] Teensy 3.6 SD init error

kgy2002, 2019-01-01:
### Description

I created a Teensy 3.6 to Mega2560 header board that interfaces with the standard RAMPS with an attached RepRap Discount Full Graphic Smart Controller. The board was designed to be used with TMC2130 drivers on the RAMPS running 256 microsteps natively, and from what I can tell, the SPI bus communication works since the printer moves the proper distances in that mode and the M122 debug information shows everything as being OK.

However, for the life of me I haven't been able to get the SD card reader on the LCD working.
My setup involves all the SPI components (4x TMC2130 and SD) sharing the SCK, MOSI, and MISO pins on the RAMPS with the separate SPI CS pins going to independent I/O.

**Steps to Reproduce:**
Downloaded a fresh Marlin 2.0.x install, disconnected the SPI pins on the drivers so only the SD is using those pins, and set only these parameters in the Configuration.h:
  #define MOTHERBOARD BOARD_TEENSY35_36
  #define SDSUPPORT
  #define SPI_SPEED SPI_EIGHTH_SPEED (tried all 4 speeds)
  #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

SD still behaves the same and can't initialize.

**Expected behavior:**
Working SD

**Actual behavior:**
Getting various errors when attempting to initialize the SD card:
Tried a 2GB standard SD card which tends to only get SD init fail errors. (FAT and FAT32)
Tried several 4GB and 16GB SDHCs which tend to get openRoot failed errors. (FAT32)
Occasional volume.init failed error messages too.

**Additional Info:**
Once or twice I was able to initialize the SD but was unable to read any files at all.
Same behavior with a manual SD_DETECT toggle switch connected.
Verified that the 4 SD SPI signals are continuous all the way back to the Teensy.
Tried several different CS pins including the previously working CS pin for the drivers, which I think involves changing spi_pins.h?, since it seems like SS_PIN is also defined there for the Teensy HAL.

[pins_TEENSY35_36.h.txt](https://github.com/MarlinFirmware/Marlin/files/2719143/pins_TEENSY35_36.h.txt)

Ludy87, 2019-01-01:
try for the TMC's   
```c++
#define TMC_USE_SW_SPI
#define TMC_SW_MOSI
#define TMC_SW_MISO
#define TMC_SW_SCK
```

kgy2002, 2019-01-01:
I removed the TMCs to try to isolate any potential issues on the SPI line already. With just the SD physically connected and the TMCs not defined, the SD slot still doesn't work.

ryanhatfield, 2019-02-11:
If you connected the teensy to the Mega2560 Ramps pinout, what pins did you map to on the Teensy?

Could this be a hardware issue? 
You mentioned you verified the SD signals, did you do that with a multimeter? or did you test with an oscilloscope?

boelle, 2019-02-21:
@kgy2002 problem fixed?

kgy2002, 2019-02-22:
Bought a new 16gb PNY SDHC and it seemed to resolve the issue.

closed 2019-02-22 by kgy2002