MarlinFirmware/Marlin #12784: [FR 2.0.x] NOZZLE PARK Z Value absolute instead of relative

FanDjango, 2019-01-01:
When performing a filament change (for example on filament runout), the nozzle gets parked.

When setting the Nozzle Park position like thus:

```
#define NOZZLE_PARK_POINT { (X_MIN_POS), (Y_MAX_POS), 1000 }
```

the "1000" signifies a 1000mm **relative** Z motion (which is translated into a move only as far as the upper bounds allow (Z_MAX_POS).

So far so good.

In my case I would like the Nozzle to rise to a **specific** Z Position (so a relative move won't help me unless I am happy with Z_MAX_POS, that's why I used "1000" as a workaround.

To make this easier, I propose the following addition:

In pause.cpp, I have coded:

```
  // Park the nozzle by moving up by z_lift and then moving to (x_pos, y_pos)
  if (!axis_unhomed_error())
  #ifdef NOZZLE_PARK_POINT_Z_ABS
    Nozzle::park(1, park_point);
  #else
    Nozzle::park(2, park_point);
  #endif
```

Thus, if you **#define NOZZLE_PARK_POINT_Z_ABS** in the configuration, you can specify an absolute value for NOZZLE_PARK_POINT if needed instead of the value being treated as a relative one.

FanDjango, 2019-01-10:
I forgot to explain why an absolute position (Z) is nice: One of the reasons to put the nozzle to a defined place (away from the bed) is that it is an excellent opportunity to clean the nozzle manually a little bit before continuing with the new filament. I hate dark gunge left over on the nozzle to contaminate a lighter colored new filament, for example.

boelle, 2019-06-22:
missing label

thisiskeithb, 2021-11-28:
Closing because original requester deleted their account.

closed 2021-11-28 by thisiskeithb