MarlinFirmware/Marlin #12785: Thermistor/temperature problem

frankitunes, 2019-01-01:
### Description

Hi and Happy New year!

I am trying desperately to install Marlin (1.1.9) but there is an issue with the thermistors behavior. I tried to find an answer on several facebook groups and on RepRap forum (seehttps://reprap.org/forum/read.php?415,842736

[Conf.h.zip](https://github.com/MarlinFirmware/Marlin/files/2719301/Conf.h.zip)

) but nobody had this issue therefore I cannot go ahead to find a solution. Maybe it is a bug in Marlin ? I would really appreciated if someone can help me.

I have a Anet A8 which is working properly under A8 stock firmware so I don't think it is a hardware issue or a wrong wiring. 

Problems are :
1. The temp shown for the hot end it too low (9° whilst it is 18° in the room). Temp is fine for the bed.
2. If I warm the bed to 30° it goes up to 30° which is fine but if I warm up the hot endto 200°, the hot end warm which is also ok. BUT the temp shown for the bed increases to 110° without warming the bed above 30°. The bed is still around 30° then decrease a bit because it is not warming anymore dur to the temp shown at 110°.
3. If on the motherboard I connect the bed temp connector to the hot end entry, the temp shown for the hot end (which is in reality the bed) is 27° which is fine...
4. If I flash back the stock firmware on my anet, the 2 temp shown are ok.

I become crazy because I really want to use Marlin for security reasons and to be able to use a bltouch.

Any idea of the issue ? In config.h I have selected 5 et 5 and tried several others but without any success.

Any help will be appreciated. Thanks a lot in advance.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

FanDjango, 2019-01-01:
**Until I replaced the Anet A8 controller board with a better one, I ran Marlin 1.1.9. Just like you, I defined Motherboard to be ANET A10 etc.etc. and had no problems. So your symptoms are funny, I agree.**
Initially I thought maybe you have a wiring cross-connection somewhere on the sensors. But going to the stock firmware, you say, solves the problem.
Hmmm. 30° and 200° - the 110° would be somewhat like having the two sensors in parallel, averaging the two inputs. But with the stock firmware it does not happen?
A sensor broken? But with the stock firmware it does not happen?
I can understand your dilemma.
I checked the pin definitions for ANET A10 Arduino 1280 and Analog in for HE and BED is 7, 6 - that would be correct. I don't see anything else at the moment.

frankitunes, 2019-01-01:
FanDjango, thanks for your time! Yes, it is working fine with stock firmware on my Anet A8 (see picture of my current print). I will try to reinstall Marlin 1.1.9 and get more info of the problem.
Thanks again.
[image]
[image]

RudyBenner, 2019-01-01:
Compiles good, code looks right in that section.

I have a feeling you have somehow, by some miracle, wired the two thermistors in series.
That is preposterous, I will admit. Yet much of what you describe supports my hunch.
(Hunches are the subconscious trying to get our attention.)
Trying to come up with a way that could happen. Have to admit I am not getting traction.
The epiphany will strike at an inopportune moment, in traffic.

Sound like a wiring problem. Nominal resistance is 100k ohms @ room temp. NTC
(negative temperature coefficient)
Get someone else to check your wiring, you have a blind spot. Go have a beer or tea.

Would a copy of the schematic be of any help to you? See the attached file.

[ANET3D_Board_Schematic.pdf](https://github.com/MarlinFirmware/Marlin/files/2719458/ANET3D_Board_Schematic.pdf)

rudy

RudyBenner, 2019-01-01:
**I take back what I said about your config files.** The configuration_adv.h looks screwy. Configuration.h looks like it only had one change.
I suggest starting with the provided example files.
rudy

FanDjango, 2019-01-01:
Consider setting the bed thermistor and then the hotend thermistor to one or the other of the "dummy debug" settings, thereby producing fixed values for first the one and the other. Test bed and then hotend separately first. Try to isolate the problem or get a better overview of what is happening in combination with what.
```
 *   998 : Dummy Table that ALWAYS reads 25°C or the temperature defined below.
 *   999 : Dummy Table that ALWAYS reads 100°C or the temperature defined below.
 *
 * :{ '0': "Not used", '1':"100k / 4.7k - EPCOS", '2':"200k / 4.7k - ATC Semitec 204GT-2", '3':"Mendel-parts / 4.7k", '4':"10k !! do not use for a hotend. Bad resolution at high temp. !!", '5':"100K / 4.7k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '501':"100K Zonestar (Tronxy X3A)", '6':"100k / 4.7k EPCOS - Not as accurate as Table 1", '7':"100k / 4.7k Honeywell 135-104LAG-J01", '8':"100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT", '9':"100k / 4.7k GE Sensing AL03006-58.2K-97-G1", '10':"100k / 4.7k RS 198-961", '11':"100k / 4.7k beta 3950 1%", '12':"100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT (calibrated for Makibox hot bed)", '13':"100k Hisens 3950  1% up to 300°C for hotend 'Simple ONE ' & hotend 'All In ONE'", '20':"PT100 (Ultimainboard V2.x)", '51':"100k / 1k - EPCOS", '52':"200k / 1k - ATC Semitec 204GT-2", '55':"100k / 1k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '60':"100k Maker's Tool Works Kapton Bed Thermistor beta=3950", '61':"100k Formbot / Vivedino 3950 350C thermistor 4.7k pullup", '66':"Dyze Design 4.7M High Temperature thermistor", '70':"the 100K thermistor found in the bq Hephestos 2", '71':"100k / 4.7k Honeywell 135-104LAF-J01", '147':"Pt100 / 4.7k", '1047':"Pt1000 / 4.7k", '110':"Pt100 / 1k (non-standard)", '1010':"Pt1000 / 1k (non standard)", '-4':"Thermocouple + AD8495", '-3':"Thermocouple + MAX31855 (only for sensor 0)", '-2':"Thermocouple + MAX6675 (only for sensor 0)", '-1':"Thermocouple + AD595",'998':"Dummy 1", '999':"Dummy 2" }
 */
#define TEMP_SENSOR_0 5 //§§§§ Anet A8 original thermistor
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_5 0
#define TEMP_SENSOR_BED 5 //§§§§ Anet A8 original thermistor
#define TEMP_SENSOR_CHAMBER 0

// Dummy thermistor constant temperature readings, for use with 998 and 999
#define DUMMY_THERMISTOR_998_VALUE 25
#define DUMMY_THERMISTOR_999_VALUE 100

```
If you have setup the firmware updater in octoprint, you can compile a change in the marlin firmware and upload to the printer in a jiffy. Gives you upto 10 trial runs with different settings per hour, no need to unplug anything, just click-click.
Also: consider using Bugfix-2.0, that's closer to where the action is right now. I am using it and I prefer it to 1.9.x. I am worried that your pin-definitions are somehow warped, that would be one of the last ideas to consider and using 2.0.x would get you to the current pin/board files. Note though, that I also started off with 1.9.x and did not have this problem, so you can see this is only a shot in the dark.

frankitunes, 2019-01-01:
> Your configuration.h file looks like only 1 change has been made.
> Your confuguration_adv.h looks like something from hell.
> 
> I suggest you start again, using the example files provided.
> 
> rudy

Thanks Rudy for your funny advices. I prefer beer than tea...
The files I provided are the one from "\Marlin-1.1.x\Marlin\example_configurations\Anet\A8" and as u notice I have made 1 modif in "Configuration.h" and nothing in the other.
Thanks.

frankitunes, 2019-01-01:
> Consider setting the bed thermistor and then the hotend thermistor to one or the other of the "dummy debug" settings, thereby producing fixed values for first the one and the other. Test bed and [...]

I am sorry I do not really understand what you mean... could you explain what to do step by step ? Thanks.

frankitunes, 2019-01-01:
Btw; i also tried with the "Marlin-bugfix-1.1.x" version but nothing better.. I dont know how to try with "Marlin-bugfix-2.0.x" because there are only a coule of files provided and nothing for Anet A8, should I keep all other files from the previous version ?
Thanks again for your time.

FanDjango, 2019-01-01:
Ok, stick with 1.9.x right now.
How do you currently compile your firmware (after modifying configuration.h)? arduino IDE 1.8.8?
How do you currently upload a newly compiled firmware to the printer? USB cable from arduino IDE or via octoprint?
So anyhow, do the following:
Set the thermistor values like follows (instead of type "5"):
```
 *   998 : Dummy Table that ALWAYS reads 25°C or the temperature defined below.
 *   999 : Dummy Table that ALWAYS reads 100°C or the temperature defined below.
 *
 * :{ '0': "Not used", '1':"100k / 4.7k - EPCOS", '2':"200k / 4.7k - ATC Semitec 204GT-2", '3':"Mendel-parts / 4.7k", '4':"10k !! do not use for a hotend. Bad resolution at high temp. !!", '5':"100K / 4.7k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '501':"100K Zonestar (Tronxy X3A)", '6':"100k / 4.7k EPCOS - Not as accurate as Table 1", '7':"100k / 4.7k Honeywell 135-104LAG-J01", '8':"100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT", '9':"100k / 4.7k GE Sensing AL03006-58.2K-97-G1", '10':"100k / 4.7k RS 198-961", '11':"100k / 4.7k beta 3950 1%", '12':"100k / 4.7k 0603 SMD Vishay NTCS0603E3104FXT (calibrated for Makibox hot bed)", '13':"100k Hisens 3950  1% up to 300°C for hotend 'Simple ONE ' & hotend 'All In ONE'", '20':"PT100 (Ultimainboard V2.x)", '51':"100k / 1k - EPCOS", '52':"200k / 1k - ATC Semitec 204GT-2", '55':"100k / 1k - ATC Semitec 104GT-2 (Used in ParCan & J-Head)", '60':"100k Maker's Tool Works Kapton Bed Thermistor beta=3950", '61':"100k Formbot / Vivedino 3950 350C thermistor 4.7k pullup", '66':"Dyze Design 4.7M High Temperature thermistor", '70':"the 100K thermistor found in the bq Hephestos 2", '71':"100k / 4.7k Honeywell 135-104LAF-J01", '147':"Pt100 / 4.7k", '1047':"Pt1000 / 4.7k", '110':"Pt100 / 1k (non-standard)", '1010':"Pt1000 / 1k (non standard)", '-4':"Thermocouple + AD8495", '-3':"Thermocouple + MAX31855 (only for sensor 0)", '-2':"Thermocouple + MAX6675 (only for sensor 0)", '-1':"Thermocouple + AD595",'998':"Dummy 1", '999':"Dummy 2" }
 */
#define TEMP_SENSOR_0 999
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_5 0
#define TEMP_SENSOR_BED 998
#define TEMP_SENSOR_CHAMBER 0

// Dummy thermistor constant temperature readings, for use with 998 and 999
#define DUMMY_THERMISTOR_998_VALUE 70
#define DUMMY_THERMISTOR_999_VALUE 200
```
Compile, upload and you should see hotend at 200 and bed at 70, it would prove software is ok and need to check wiring and so on.

frankitunes, 2019-01-02:
Hi, to answer your questions before I was compiling/flashing through Arduino and a USB cable. Now I tried through octoprint (did not know this - very helpfull -> THX) and it is the same...
Tried with 999/998 + 70/200 and I see 200° / 70°, I don't understand... When you say check wiring you mean wiring of the MB? Thanks.

FanDjango, 2019-01-02:
Ok, so now we have confirmed your firmware workflow is correct and we can do some checks (remembering: stock firmware ok, marlin not ok). 

It's either a bug in Marlin connected to some special setting in the config files or it is because the Anet A8 that you have got does not correspond to the config files.

So, I have checked your config files against the example configuration for Anet A8. **configuration.h** has only on line changed: the printer name. **configuration_adv.h** has no changes.

Now to do some experiments, here are some things to try: (you will need to search for where these lines are in configuration.h)

How about (I remember that the Anet V1.0 controller supported this):
```
#define PIDTEMPBED
//#define BED_LIMIT_SWITCHING
...
and
...
  #define DEFAULT_bedKp 407.37 //§§§§ my values, you can use them for now
  #define DEFAULT_bedKi 56.51  //§§§§
  #define DEFAULT_bedKd 734.18 //§§§§
```

Can you have a good look at your Anet controller board and check if it says Anet V1.0 on the board? Could it be a much newer version of their controller that Marlin does not know about?

frankitunes, 2019-01-02:
Thanks again for your help! 
I will try this back at home. Version of the board is ANET3D 1-5. (see picture).

[image]

Just to be 100% sure, you want me to uncomment this "//#define PIDTEMPBED" (remove the // ) and comment this "#define BED_LIMIT_SWITCHING" (add // ) ? Right?

and put your PID values just a few lines below ?

to be like this : 

//===========================================================================
//============================= PID > Bed Temperature Control ===============
//===========================================================================

/**
 * PID Bed Heating
 *
 * If this option is enabled set PID constants below.
 * If this option is disabled, bang-bang will be used and BED_LIMIT_SWITCHING will enable hysteresis.
 *
 * The PID frequency will be the same as the extruder PWM.
 * If PID_dT is the default, and correct for the hardware/configuration, that means 7.689Hz,
 * which is fine for driving a square wave into a resistive load and does not significantly
 * impact FET heating. This also works fine on a Fotek SSR-10DA Solid State Relay into a 250W
 * heater. If your configuration is significantly different than this and you don't understand
 * the issues involved, don't use bed PID until someone else verifies that your hardware works.
 */
#define PIDTEMPBED

//#define BED_LIMIT_SWITCHING

/**
 * Max Bed Power
 * Applies to all forms of bed control (PID, bang-bang, and bang-bang with hysteresis).
 * When set to any value below 255, enables a form of PWM to the bed that acts like a divider
 * so don't use it unless you are OK with PWM on your bed. (See the comment on enabling PIDTEMPBED)
 */
#define MAX_BED_POWER 255 // limits duty cycle to bed; 255=full current

#if ENABLED(PIDTEMPBED)

  //#define PID_BED_DEBUG // Sends debug data to the serial port.

  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from FOPDT model - kp=.39 Tp=405 Tdead=66, Tc set to 79.2, aggressive factor of .15 (vs .1, 1, 10)
  #define DEFAULT_bedKp 407.37
  #define DEFAULT_bedKi 56.51
  #define DEFAULT_bedKd 734.18

  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from pidautotune
  //#define DEFAULT_bedKp 97.1
  //#define DEFAULT_bedKi 1.41
  //#define DEFAULT_bedKd 1675.16

  // FIND YOUR OWN: "M303 E-1 C8 S90" to run autotune on the bed at 90 degreesC for 8 cycles.
#endif // PIDTEMPBED

FanDjango, 2019-01-02:
AHA! Anet V1-5. Don't do anything yet unless you feel like it. Let's do some research first to see if anyone else is having trouble as with respect to Anet V1.0 ( = "ANET 10" ). My Anet controller is one of the first, it doesnt even have a version. The example configs for ANET A8 are not very new and maybe they have changed something - it is a moving target. It is difficult to imagine what the change could be - have you checked all the other pins functions, is all the movement of all motors working correct and do all the endstops work?
Also, when you compile the marlin in arduino ide, have you set the right board? installed all the updates that arduino suggests? when the compile is finished, how many % of the ram space is used?
Can you make a better photo of this area:? 
[image]

FanDjango, 2019-01-02:
Ok, after some surfing I think the board should be pin compatible with the older ones. I am running out of ideas, I am afraid. You need to do some more playing around and testing, like you did in your first post of this issue.

FanDjango, 2019-01-02:
Merde. All looks good.

frankitunes, 2019-01-02:
Yes, shit! Ia it worth to try what you proposed regarding PIDTEMPBED?

frankitunes, 2019-01-02:
BTW, does a new mb could correct the issue ? Maybe, maybe not...

FanDjango, 2019-01-02:
Why double wires to the bed? Safety?
[image]

frankitunes, 2019-01-02:
I got it like this and most of people I saw on facebook groups have the same.

[image]

frankitunes, 2019-01-02:
> New MB?
>
> So, new MB will solve the problem? **I don't think you should give up so soon** but also remember some day you will want a better one. It would be nice to identify what the actual problem is. Depends how much time you have.

I have time, meanwhile anet works pretyt well with stock firmware but I would like to use bltouch and features from Marlin + safety reasons.

But I don't know how to solve my issue, I am a geek rather good in IT and electronics but not a pro!

> 
> The connector for your HotEnd Thermistor: The single crimp terminals cannot be touching something by mistake? (does not explain why the stock firmware works, though).
> You wrote:
> 
> > If on the motherboard I connect the bed temp connector to the hot end entry, the temp shown for the hot end (which is in reality the bed) is 27° which is fine...
> 
> How about you swap both thermistor plugs. Then heat bed to (example) 30°, then heat hotend to 60°. How does that behave?
> 
> Do you have a Multimeter with a fine enough probe to take voltage measurements on the board? In that case I would like the voltages for the HE and the BED wires taken on the outside terminal.
> 

I am currently printing something, I will do these tests afterwards and keep you posted.

> Re PIDTEMPBED: If you have time. It won't hurt. But I can't believe it would change anything. Or may exaclty that is the difference, hmmm. **Is there anyway in your wiring, that one of the heater connections is influencing one of the thermistor connections?** Before or after the MOSFETs. And depending on PIDTEMPBED Marlin is doing different way of heating. And due to your possible wiring fault, it reacts differently from the stock firmware. Marlin is reading the thermistor value at a different time than stock firmware, maybe.

I will also try this.

FanDjango, 2019-01-02:
👍 Mine has only 4 pins for the bed. They did this, I suppose, to be less of a fire hazard - that connector is one of the first to get hot and burn the house down.

frankitunes, 2019-01-02:
I read that solder the wires could be hazardous. What's your view?

frankitunes, 2019-01-03:
Hello, I have tried to swap thermistors at room temp and at 30/60.... at least results are consistents! but strange, a good exercise to train our logical.

[image]

FanDjango, 2019-01-03:
That table is VERY well done. Now, let's think about what it tells us. Gimme some hours and a beer :-)

FanDjango, 2019-01-03:
To help with the thinking:
So we need to look at: (this is by the square cpu chip)
[image]
the yellow marked pins.
And this is next to the two white connectors:
[image]

[image]
And check on the Motherboard:
[image]
So: What causes that table (with Marlin) and with the stock firmware, you state that the table is "normal" (let's believe that for the moment).
Now beer.

FanDjango, 2019-01-03:
There is a Marlin Debug command we should use to interrogate pins directly. Thats M43 and in the firmware you need to enable PINS_DEBUGGING.
```
//#define PINS_DEBUGGING
```
But: Will it tell you analog pins? Please test it - I cannot right now because I have a long print running, just like you before.
What we are looking for, trying to prove/unprove, is that maybe the pins assignements you are using in configuration.h (board setting), Arduino (board setting) are wrong somehow.

**It can't be a fault on the motherboard if the stock firmware works.**

**Unplugging either EXT or BED reflects on the reported temperature, so you would assume that both are being "touched" by the connection** But it is possible for ATMEGA Analog Pins to have cross-talk to an open pin - it would follow (but not exactly, always less than) the connected pin.

frankitunes, 2019-01-03:
sorry, i do not follow you... what u want me to try/test?

FanDjango, 2019-01-03:
Look up M43 in [http://marlinfw.org/docs/gcode/M043.html](url)
But to able to use it, you need to re-compile Marlin by enabling PIN_DEBUGGING in configuration_adv.h and upload. Then in OctoPrint, in the terminal, you can enter M43 etc. etc. Play with it...

FanDjango, 2019-01-03:
Hold all your activity for the moment. I think I have found something.
In the source code of Marlin 1.1.x, in pins_ANET_10.h, there are the following lines:
```
#ifndef __AVR_ATmega1284P__
  #error "Oops!  Make sure you have 'Anet V1.0', 'Anet V1.0 (Optiboot)' or 'Sanguino' selected in the 'Tools -> Boards' menu and ATmega1284P selected in 'Tools -> Processor' menu."
#endif

#ifndef BOARD_NAME
  #define BOARD_NAME "Anet"
#endif

...
//
// Temperature Sensors
//
#define TEMP_0_PIN          7   // Analog Input (pin 33 extruder)
#define TEMP_BED_PIN        6   // Analog Input (pin 34 bed)
...
```
temp_0_pin IS 7 (arduino numbering) and temp_bed_pin IS 6, That is correct. But according to the schematic, it is connected to pins 30 and 31. So we need to address that.
Wrong board in Arduino. I am researching right now, but we on the right track.

FanDjango, 2019-01-03:
Ok.
In file **pins_ANET_10.h** you will find
```
//
// Temperature Sensors
//
#define TEMP_0_PIN          7   // Analog Input (pin 33 extruder)
#define TEMP_BED_PIN        6   // Analog Input (pin 34 bed)
```
Please replace that by
```
//
// Temperature Sensors
//
#define TEMP_0_PIN          4   // Analog Input (pin 33 extruder)
#define TEMP_BED_PIN        3   // Analog Input (pin 34 bed)
```
recompile, upload to printer and meanwhile I am switching from beer to bordeaux. If you have good news, there will be a Veuve Cliquot.

frankitunes, 2019-01-03:
Sorry no veuve-cliquot for time being... but a good boreaux is not so bad!

[image]

FanDjango, 2019-01-03:
No, dead end.
I am back to beginning.
Question: All other functions are working in Marlin - Movement of Axes and Endstops?

FanDjango, 2019-01-03:
Yeah, I thought I had found something, but it was not logical so I deleted the post.
You got a crazy problem there, you must admit.

FanDjango, 2019-01-03:
One interesting little fact came out of it, maybe it helps you find something:
[image]

frankitunes, 2019-01-04:
Hi, to answer your question, yes in Marlin everything else seems to work well ie x/y/z movements and endstops...

maciejmucha, 2019-01-04:
Hi, have you checked the temperature values with Octoprint or Pronterface?

frankitunes, 2019-01-04:
Of course, temps are consistent between octoprint, pronterface and the anet's screen.

FanDjango, 2019-01-04:
I have a new Anet A6 (I ordered it as a test bed for my planned new 32 bit controller) that just arrived today. Building it  aleady half finished by now and by tomorrow, ready for testing. It at this moment also has the Anet V1.5 controller board, I will hook it up and try my luck with Marlin. I will update you as soon as I have done that. Maybe I will have the same problem as you, we will see.

FanDjango, 2019-01-05:
Ok, the Anet A6 is built and the Anet V1.5 controller is connected with preliminary wiring. It's all hanging loose. Stock Anet V2.0 Firmware, I tested the functions, all ok.
Then I did the following to install Marlin:
I uninstallted Arduino via the Windows 10 Setup/Apps.
I deleted Documents/Arduino/*** completely
I deleted userid/AppData/Local/Arduino15/** completely
Rebooted PC to be totally sure (not needed really but so what)
Fresh install of Arduino
Copy the anet board defs into Documents/arduino/hardware
Start Arduino and check that the Anet V.10 board (not optiboot) can be selected
Next:
Download current bugfix-2.0 Marlin (zip file) from github
Expand the Zipfile
dive into the folder until you get to Marlin/src/config/examples/Anet/A8
copy both config files you find there into Marlin, thereby overwriting the config files originally there (those were the defaults).
I didn't change ANYTHING in the config files for the test.
1st compile failed because u8glib was missing. Went into Arduino library manager, searched for it, installed it and compiled again. No error this time.
Attached USB cable to Printer (got no octoprint on this one). Uploaded, Success - Marlin 2.0 bugfix is on the printer. 
Tested heating and motion: All is well. No problem like yours.

Now you try same.

FanDjango, 2019-01-05:
Oh, I used A6. He should use A8. He's got an A8. This test-printer of mine is an A6.

frankitunes, 2019-01-05:
OK let's do it! but in the "Marlin-bugfix-2.0.x" zipfile there are less files than in the other versions, what should I do ?

[image]

RudyBenner, 2019-01-05:
The same files are there, just in different places. Get used to it.

FanDjango, 2019-01-05:
Its all now under /src in many subdirectories. It has been "reorganised" to allow for multiple hardware platforms. You will for example find the pin files under /src/pins 
Just use the marlin.ino as usual after setting up the configuration.g and configuration_adv.h by getting these from the **watch out** folder under src/config... as I wrote. NOT from the old examples of the V 1. Marlin. The new example folder

frankitunes, 2019-01-05:
ok, ok, after cmpilation I got this from Arduino... is it a problem?

[log.txt](https://github.com/MarlinFirmware/Marlin/files/2729810/log.txt)

FanDjango, 2019-01-05:
No, those warnings are not a problem. But I am sad that you see them because if you do a clean new install of Arduino and have deleted all configuration stuff before install a new clean one, you should not be seeing these compiler warnings. Have you turned them on in the Arduino preferences?
Any way, the compile has succeeded. So do the upload,
Before I compile and upload, by the way, I put a number in the printer name (Like Anet A8-1, Anet A8-2) each time I upload so I know that what I see in the printer is what I think I compiled.

frankitunes, 2019-01-05:
Cool! My printer is currently printing... therefore I will try the next step (flash) tomorrow morning...

FanDjango, 2019-01-05:
I repeat:
**But I am sad that you see them because if you do a clean new install of Arduino and have deleted all configuration stuff before install a new clean one, you should not be seeing these compiler warnings. Have you turned them on in the Arduino preferences?**

frankitunes, 2019-01-06:
No good news this morning, still the same...

1. Heat the hotend at 100°, the bed jump to 85° (without heating).
2. Increase HE to 200°, the bed jump to 110° and stabilize.
3 Decrease HE to 100°, then turn off HE...

[image]

frankitunes, 2019-01-06:
Then I have decided to have a look at the mb to see if there something suspicious but I did not find anything stange except the general low quality of the soldering but for that price I do not think we should expect perfection.

FanDjango, 2019-01-06:
1. Does not explain why it works fine with the stock firmware. Really True? Hmmmm. Otherwise, use new mainboard.
2. Really a clean new install of Arduino? Clean Anet example file? Last chance for something to go wrong.

No more ideas here.

frankitunes, 2019-01-06:
What means the messages received aftere the compilation ?

force-e, 2019-01-23:
Guys, I had exactly the same issue on my Anet A8, after switching from stock firmware to Skynet: when printing for longer than aprox. 20 minutes my bed temperature would suddenly show very high values and the thermal runaway kicked in. Suspected my thermistor went bad so I did the bed/hotend thermistor swap and the problem persisted. Measured resistances and voltages and everything seemed to be okay.

I read you mentioned the PID tuning but did not read a confirmation that you actually did it. By default the bed PID control is disabled. I gave it a try, enabled bed PID, performed the PID tuning and stored the values and my printer seems to be printing fine now! So before buying a new mainboard, I'd suggest you to give this a try too!

NOTE: I am on Skynet 2.3.2

Be aware that you need to comment out the BED_LIMIT_SWITCHING define or you'll get a conflict error. Use the default PID settings for the bed and then perform autotune to determine the proper settings for your printer. Store them in your configuration.h, reupload and give it a try.

frankitunes, 2019-01-26:
Thanks for your feedback. I did PID tuning but it hasn't change anythin. By the way I have ordered a RAMPS to replace my stock MB... It is an upgrade I wanted to do anyway.

frankitunes, 2019-02-11:
Just finished installing a ramps 1.4 on my anet a8 and it is working like a charm...

closed 2019-02-20 by thinkyhead