MarlinFirmware/Marlin #12790: Issue with marlin randomly disconnecting during print

Cont3mplation, 2019-01-01:
I quite recently purchased an mks gen L v1.0 board to replace an old anet a8 one from a reprap printer I'm working on. I paired this board with DRV8825 stepper drivers. The uploading of marlin 1.1.x with all settings correct went smooth, but when I now go to attempt to print anything, at random points in the print, either 10 seconds in or even 3 seconds before ending, it will give the error that the heating has runaway and the printer has been shutdown. On looking at it after receiving this error the temperature is no longer being updated and the printer won't take any movement orders until it is reconnected or left for a short while. This would make it seem as though the serial communication between the board and the raspberry pi zero W running it (on octoprint) has stopped. This issue is present also when it's being controlled over usb with a regular pc and repetier host. I read on one of the few posts with the same issue that they found it to be electromagnetic interference that caused the issue, so I've moved my printer over a metre away from my pc, and have used a 30cm shielded usb cable to my pi. Am I missing something in the firmware as I never had the issue on my old board?

Steps to Reproduce:

1. Upload file to repetier host/octoprint after turning on and connecting printer.
2. Select print and allow printer to heat up.
3. Allow printing to start as normal.
4. Random crashing of serial communication with:
{  Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Thermal Runaway, system stopped! Heater_ID: 0 - Printer halted. kill() called!  }
error displayed.

edit: I have already attempted to turn off the windows auto usb disabling feature when using a regular pc and repetier host but this made no difference. Changing baud rate also had no effect.

kAdonis, 2019-01-01:
> Thermal Runaway, system stopped! Heater_ID: 0 - Printer halted. kill() called!

 The serial connection is closed on purpose by Marlin for safety reasons, after it detects a thermal runaway. That is what the error report means.
You probably dont have a problem with electromagnetic interference but with your hotend heater.
Please post a screenshot of your temperature graph (while printing) and a zip of your Configuration files as requested in the Issue-report template

Cont3mplation, 2019-01-02:
I have noticed that in repetier host the temperature will randomly jump down to temperatures between like 35 and 130 degrees for less than a second and jump back up, will try printing again and will post results

kAdonis, 2019-01-02:
Temperature jumps are an indication of bad wiring, check  thermistor connectors and cable

Cont3mplation, 2019-01-02:
I'll be honest they're connected at the moment with relatively loose jumper cables as I haven't had a chance to solder on proper connectors. Will do this just now and try again, hopefully it was just me being this dumb causing the problem

Cont3mplation, 2019-01-02:
Fitted the new connectors and without even starting the heating process the temperatures are much more stable, hopes are high

Cont3mplation, 2019-01-02:
Okay so less issues with temperature jumps, however recieved the same error perhaps 30 seconds in: 
[image]

Cont3mplation, 2019-01-02:
And this is the exact error:
[image]

kAdonis, 2019-01-02:
your configs? please

Cont3mplation, 2019-01-02:
Sorry still pretty new to this stuff, where abouts would I find those?

kAdonis, 2019-01-02:
Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
the files you compiled Marlin with

Cont3mplation, 2019-01-02:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2719742/configs.zip)

Cont3mplation, 2019-01-02:
I tried setting mintemp to 0 but that didn't fix the issue either so returned it to 170 in the config

kAdonis, 2019-01-02:
I see you have the default PID values defined

```
  #define DEFAULT_Kp 22.2
  #define DEFAULT_Ki 1.08
  #define DEFAULT_Kd 114
```
I guess you never run PID Autotune? 
You should do that, to get the correct PID values for your printer (M303)
please read the Marlin documentation, especially:
http://marlinfw.org/docs/configuration/configuration.html
https://reprap.org/wiki/PID_Tuning
Im afk now, but I will come back here

Cont3mplation, 2019-01-02:
Have autotuned PID and while it's more efficient it still recieves the same error from time to time seemingly

Cont3mplation, 2019-01-02:
Having said this the printer just finished the first full print it's managed in a few days without this error, will try a couple more and if all is good seems the pid correction fixed it

Cont3mplation, 2019-01-02:
Halfway through a two hour print on the raspberry pi webserver and everything is running perfectly. Going to go so far as to say this is probably the issue fixed. Thanks for the help!

kAdonis, 2019-01-02:
I'm glad I could help

closed 2019-01-02 by Cont3mplation