MarlinFirmware/Marlin #12791: [sbase] UBL/Bilinear Causes motors to stutter.

forkoz, 2019-01-01:
Downloaded the jan 1st commits and decided to give UBL and Bilinear a go, I was using linear for a while.

Made my grid and set it to print. First move to wipe after homing was stuttering all over the place. Motion is no longer smooth at all and everything vibrates.

Did something happen to these recently, I don't remember having this issue. I went back to linear and just like that its gone...

forkoz, 2019-01-01:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2719717/Marlin.zip)

This didn't attach because github hates firefox now.

Roxy-3D, 2019-01-02:
Did you do a M502 & M500 after turning on UBL/Bi-Linear?   If not...  That would explain the behavior.

forkoz, 2019-01-02:
Yes, saved and wiped "eeprom". I tried both leveling systems and wiped each time, the Z offset actually reset when I flashed the FW too.

boelle, 2019-02-21:
@forkoz problem fixed?

closed 2019-03-03 by thinkyhead