MarlinFirmware/Marlin #12792: Configuration.h says Marlin defaults  Z_MIN_PROBE_PIN to Z_MAX_PIN, but that doesn't seem to work.

sidhant, 2019-01-02:
Configuration.h says

> (By default Marlin assumes the Z-Max endstop pin.)
>  * To use a custom Z Probe pin, set Z_MIN_PROBE_PIN below.

As such, I proceeded to leave it undefined, assuming that Marlin somewhere defines something on the lines of #define Z_MIN_PROBE_PIN Z_MAX_PIN, however it turns out that pin_RAMPS.h has the following:
`#ifndef Z_MIN_PROBE_PIN`
` #define Z_MIN_PROBE_PIN  32`
`#endif`
Result was my probe not working as it was plugged into Z_MAX_PIN.

In otherwords, I would expect that if configuration says there is a default, that is is defined before pin_RAMPS can detect if it is #undef'd?

RudyBenner, 2019-01-02:
/**
 * Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *
 * Enable this option for a probe connected to the Z Min endstop pin.
 */
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

this is the default setting

RudyBenner, 2019-01-02:
Seems like there is a contradiction. 

/**
 * Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *
 * Enable this option for a probe connected to the Z Min endstop pin.
 */
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is the default

sidhant, 2019-01-03:
@RudyBenner Not sure I understand your comment. I am talking about the Z_MIN_PROBE_ENDSTOP and its behavior. The config comments suggests that Z_MIN_PROBE_ENDSTOP defaults to the Z-max endstop pin, however that is not the case.

boelle, 2019-01-06:
@sidhant still having issues?

boelle, 2019-02-21:
maybe click the green close button

sidhant, 2019-02-21:
Oops, sorry :-) Thank you for the reminder.

sidhant, 2019-02-21:
On second thought.. you do realize that the core issue with the firmware remains unresolved? My workaround resolved *my* issue.

closed 2019-02-21 by sidhant

boelle, 2019-02-21:
if others have the issue they will say it here then :-D

boelle, 2019-02-21:
even if issue is closed

thinkyhead, 2019-03-03:
Configuration of Z probes has been modified, along with commentary.

volconst, 2019-05-17:
I was having the same issue - documentation is wrong for boards using pins_RAMPS.h. Why pin 32 is the default there?