MarlinFirmware/Marlin #12793: [2.0] Advance pause doesn't work on 2nd Extruder (E1)

Le-zOU, 2019-01-02:
### Description

### Steps to Reproduce

On Geeetech A20M, the unload filament option in LCD, works just fine for E0, but has not effect when choosing E1.

**Expected behavior:** unloads filament on E1

**Actual behavior:** nothing

#### Additional Information
configuration.h and adv_configuration.h
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2720121/Marlin.zip)

boelle, 2019-01-06:
maybe make the title say: [Bugfix 2.0][BUG] Advance pause doesn't work on 2nd Extruder (E1)

boelle, 2019-02-21:
@Le-zOU still a problem?

boelle, 2019-03-06:
@Le-zOU did you have a look?

Le-zOU, 2019-03-09:
sorry @boelle , have a look at what ?
I'm about 1 week behind current version/commits. I plan to flash on monday probably with latest commits at this point

boelle, 2019-03-09:
have a look at what you said 2/1-2019, a little over 2 months ago

Le-zOU, 2019-03-09:
Oh yeah I had a look but we'll didn't find anything although I'm not a developer :p

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@Le-zOU  does it work on 2nd extruder now?

Le-zOU, 2019-09-25:
I'll have to check and probably update the FW on my printer, it's been a while I haven't used it. I was fighting other issues and gave up for other projects (restoring a 1966 Ford Mustang)

boelle, 2019-09-25:
restoring the mustang gets a thumbs up from me 👍 
i think we can wait a bit more

shitcreek, 2019-10-10:
Tested with latest build on A20M and it works flawlessly.
LMK if it's still an issue for you and thread needs to be reopened.

closed 2019-10-10 by shitcreek

Le-zOU, 2019-10-10:
Can you define "latest" ?
Thank you.
I've yet to take care of my A20M, but I'm glad it seems solved.

shitcreek, 2019-10-10:
A build from 2 days ago, so technically not the latest but I don't think any changes has been made to the relevant codes.

Le-zOU, 2019-11-29:
Just FYI, it works fine on a built that I use and is several month old. so keep it closed. (I finally got round to fix my A20M)