MarlinFirmware/Marlin #12794: M30 Fails and is missing newline

msticninja, 2019-01-02:
Can't find any other reports of this since 2012, so here goes: M30 fails in marlin 2.0 on Re-arm, and is missing newline on M30 fail. Serial monitor must be reconnected, and SD card may need to be chkdsk'ed and, if so, will find deleted file as a lost chunk.
May only happen with advanced ok enabled, haven't had time to recompile without it yet.

#### Additional Information

Send: M30 /2020ra~1.gco
Recv: Deletion failed, File: 2020ra~1.gco.ok P31 B63

[Marlin-conf.zip](https://github.com/MarlinFirmware/Marlin/files/2720266/Marlin-conf.zip)

boelle, 2019-02-21:
@msticninja problem fixed?

msticninja, 2019-02-21:
Not for me. I have just stopped using M30, and delete from the FlashAir interface(it is a wifi enabled SDcard). It may be fixed in newer versions, I haven't compiled new FW since January, but I also haven't heard of anybody doing anything to fix the problem, either. I've only seen one other person that encountered the problem, so...

boelle, 2019-02-22:
same if you use bugfix 2.0?

msticninja, 2019-02-22:
It is bugfix 2.0. Pretty sure that's the only branch that supports the Re-Arm(LPC1768).

boelle, 2019-02-22:
neopixels are not supported yet, they need to make an 32 bit library for it

msticninja, 2019-02-22:
Yup, that's the fix I'm waiting for to recompile.

boelle, 2019-02-22:
so they have made a neopixel library?

personally i have looked in to neopixels and re-arm, but also waiting for it

msticninja, 2019-02-22:
No, apologies for being confusing. Since I don't strictly need M30 to function, I haven't actively tried to fix it. I've been waiting to see if someone can submit a working PR to adafruit's neopixel library to add support for LPC176X. LadyAda already said they'd accept a PR at https://github.com/adafruit/Adafruit_NeoPixel/issues/188, but they don't have the hardware to do it themselves. Until then, I don't have enough reason to compile new firmware, just to see if M30 is fixed.

boelle, 2019-06-22:
missing label

shitcreek, 2019-09-17:
Cleaning house. LMK if this issue needs to me reopened.

closed 2019-09-17 by shitcreek