MarlinFirmware/Marlin #12796: MKS-SBASE - REPRAP_DISCOUNT_SMART_CONTROLLER not working

gtrevi, 2019-01-02:
I have an MKS-SBASE board, and since days I cannot get, a REPRAP_DISCOUNT_SMART_CONTROLLER (non-graphic), to show characters (it lits up bus just shows full chars).

This is what I've done:

1) Display tested working before/after with the previous Trigorilla (from my Anycubic Kossel Linear Plus).

2) Flipped 180 degrees both EXP1 & EXP2 casings on the MKS board (so I can easily swap cables & LCD with the Trigorilla in the meanwhile). This is the equivalent od desolderong them and resoldering them backwards... as MKS should have done from the beginning!!

3) For the cleanest solution, I soldered the 3 wires as documented in the last picture of this thread, https://github.com/MarlinFirmware/Marlin/issues/8498, from P1.22/P1.23/P2.12 to their respective NC pin on EXP1.

4) I changed the spi_pins.h as below:

`...
// gtrevi: added-start ****************************************
#if MOTHERBOARD == BOARD_MKS_SBASE
  #define LPC_SOFTWARE_SPI

  #define SCK_PIN           P1_22
  #define MISO_PIN          P1_23
  #define MOSI_PIN          P2_12
  #define SS_PIN            P0_28
#else
// gtrevi: added-end ****************************************

  #define LPC_SOFTWARE_SPI
 
  /** onboard SD card */
  //#define SCK_PIN           P0_07
  //#define MISO_PIN          P0_08
  //#define MOSI_PIN          P0_09
  //#define SS_PIN            P0_06
  /** external */
  #ifndef SCK_PIN
    #define SCK_PIN           P0_15
  #endif
  #ifndef MISO_PIN
    #define MISO_PIN          P0_17
  #endif
  #ifndef MOSI_PIN
    #define MOSI_PIN          P0_18
  #endif
  #ifndef SS_PIN
    #define SS_PIN            P1_23
  #endif
#endif
...`

The Marlin code from github is the latest-stable (downloaded yeasterday).
Everything compiles fine from VSCode and succesfully deploys into the MKS board (becoming firmware.cur): platform is LPC1768, board is BOARD_MKS_SBASE and the enabled LCD in configuration.h is REPRAP_DISCOUNT_SMART_CONTROLLER (I picked-up the entire configuration.h/adv from the Anycubuc Kossel example).

What am I missing to get this "holy LCD controller" to start working?? :)

Many thanks for any help.

RudyBenner, 2019-01-02:
config files please?

forkoz, 2019-01-02:
I never got mine working either and just got the graphical version. It only needs the connectors flipped.

aamcle, 2019-01-04:
I can't get it going, struggling with the graphical version but not tried sealing both connectors (as above) yet.

Aamcle

gloomyandy, 2019-01-04:
The DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER definately works with the SBase (that's what I have), you will probably have to rotate both connectors. 

You do not need a custom cable, but without one if you want to use the SD card reader attached to the display and you also want to have an SD card in the slot on the main board you will not get correct insert/remove detection of either card (as they use a shared pin). The SD reader will still work but you will have to tell Marlin when you insert/eject the card. This may or may not be important to you. The default settings for the SBase board does not use the external SD card reader at all. Instead the on board SD card is used my Marlin and the same card is made available to your PC as a USB drive. This setup seems to work well for a lot of people. There are various options for how to handle the two SD cards in the pins.h file.

aamcle, 2019-01-04:
Thankyou for your reply.

The first thing I'll try is rotating both connectors so far I have rotated
only connection 1.

To try to avoid these problems I bought a genuine MKS display to match the
MKS board But examining the board suggests that MKS had some other company
make the display and they just sold it!

Best Regards     Alan.

forkoz, 2019-01-04:
LCD 2004 is the REPRAP_DISCOUNT_SMART_CONTROLLER. (https://image.dhgate.com/0x0p/f2/albu/g7/M01/08/25/rBVaSVtdr7uAR90KAADGKC6B4PU541.jpg) The working one is DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. (https://reprap.org/mediawiki/images/thumb/9/93/RRD_FULL_GRAPHIC_SMART_CONTROLER_INFO.JPG/190px-RRD_FULL_GRAPHIC_SMART_CONTROLER_INFO.JPG)

The MKS TFT is the one they make I think and it has separate fw plus works by sending gcode over serial. Hope that's not what you got.

aamcle, 2019-01-04:
I have both :)

Atb

gtrevi, 2019-01-04:
> I never got mine working either and just got the graphical version. It only needs the connectors flipped.

Yeah... just gave up too and ordered a GLCD as well... and of course it worked oob on a second MKS SBASE I could try it on, having already flipped in advance the plastic connector housing.

boelle, 2019-01-06:
@gtrevi so this issue is solved? since it has nothing to do with marlin maybe close it ?

gtrevi, 2019-01-06:
> @gtrevi so this issue is solved? since it has nothing to do with marlin maybe close it ?

Personally, I wouldn't call it solved: the issue of having the REPRAP_DISCOUNT_SMART_CONTROLLER working on an MKS SBASE with Marlin is still open...

The fact that I gave-up and went for the DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER upgrade is not a solution in my humble opinion: one from the community may not be available (or cannot) do this type of upgrade and will be anyways stuck...

I'd let to the Marlin team wither supporting or not solving the initial issue, and eventually closing it as a non-supported scenario.

aamcle, 2019-01-06:
I've have also caved in and bought a tft32 it will be here on Monday.

Aamcle

boelle, 2019-01-06:
Are we sure its a problem with marlin? are we 100% sure that its not a hardware problem?

gloomyandy, 2019-01-06:
@aamcle The tft32 may not be a good choice. It is basically a separate (small) computer that talks to Marlin via a serial port and sends gcode commands to it. As a result it does not have the same "tight binding" that other displays have and may not be able to access some of the features.

gtrevi, 2019-01-06:
> @aamcle The tft32 may not be a good choice. It is basically a separate (small) computer that talks to Marlin via a serial port and sends gcode commands to it. As a result it does not have the same "tight binding" that other displays have and may not be able to access some of the features.

Totally agree: I have a TFT32 as well, and just few commands are useful, especially if you have a delta... also, I haven't been able to fine the source code to eventually customize it; you can only rough-tweak with some configurations.
It ended up in my “useless” electronics box, too bad I also bought some other accessories.

A Raspberry w/ OctoPI would be a hugely better investment in my opinion.

p3p, 2019-01-06:
> Personally, I wouldn't call it solved: the issue of having the REPRAP_DISCOUNT_SMART_CONTROLLER working on an MKS SBASE with Marlin is still open...

There is no issue with Marlin, I have a test setup working with the 4 bit bus those character displays use, as long as you can find pins to use. Unfortunately the SBase, and ReARM, maybe all the smoothieboard based boards do not populate the entire extension headers where the displays connect. If someone wants to use hardware the boards don't natively support without hardware modification it's left upto the user to configure the pins file in a way that does not conflict with the rest of their hardware and wire it up.

gtrevi, 2019-01-06:
> Are we sure its a problem with marlin? are we 100% sure that its not a hardware problem?

Well, the two displays are still both working, and the HW problem on my first SBASE was certainly caused by the wiring above (because those pins don’t work anymore, as they did before).
That wiring was got from an issue that proposed it as a solution, which should in care corrected.

The point and question is: 

1- does Marlin support REPRAP_DISCOUNT_SMART_CONTROLLER  on an MKS SBASE, and if so with what exact HW modifications?

2- If not (or there's no intention do support it), perhaps Sanitizer should intercept the MKS-SBASE+REPRAP_DISCOUNT_SMART_CONTROLLER combo in the configuration, and clearly advise the user that this configuration is not supported.

p3p, 2019-01-06:
> 1- Does Marlin support REPRAP_DISCOUNT_SMART_CONTROLLER on an MKS SBASE, and if so, with what exact HW modifications.

The LPC176x platform supports it, I made sure of it as it was requested, so as long as the pins can be connected, and the pins file configured, it will work. 

Although I admit I don't use one so there could have been a regression from my last test after the framework was pulled out of Marlin so I'l try and find time to hook the display up again.

gjdodd, 2019-01-07:
I read through most of this thread, and saw no mention of the fact you might need to edit your pins file for the MKS SBASE as you need to say what SD type you are using, the default is using the onboard sd, there are three define you need to either enable or disable. I think these came in when the new framework was included.

//#define LPC_SD_CUSTOM_CABLE // Use a custom cable to access the SD
//#define LPC_SD_LCD          // Marlin uses the SD drive attached to the LCD
#define LPC_SD_ONBOARD        // Marlin uses the SD drive attached to the control board

RudyBenner, 2019-01-07:
Line 165 in the appropriate pins file. Thanks.
rudy

gloomyandy, 2019-01-07:
You should think about why you want to change those lines before you do it though. The SBase has an onboard SD card, this card is used for firmware updates and can be shared with your PC (or other system) as a USB drive. This allows fast access from the PC to this "drive", by default Marlin will also be able to access this drive. This allows you to slice on your PC, save the file directly to the SD card and then use the Marlin menu (or a remote control program) to mount the SD card and print the file. For many this is a fast efficient workflow. 

If you tell Marlin to use the LCD based SD card (if you have one), then this card will not be shared with the PC (the on board card will), also because only a single pin is used to detect SD card insertion then Marlin will not be able to detect a card being inserted/removed from the reader if you have a card in both slots. So to get "normal" SD card usage you should probably not have an SD card in the on board SD card reader if you want to use the SD card reader on the LCD. If you want to use both cards and have insert/remove detection then you need a custom cable for the LCD that uses a different pin for insert detect (this is not necessarily the same as the custom cable that used to be needed to get the LCD SD card working at all - that had several other pins defined and is no longer required, but the define remains for those that have an older setup). 

So if you really want a traditional SD card on the LCD (and need to or are happy to remove the SD card to get files on it from a PC) then go ahead and make the changes. But if you have a PC connected directly you may want to consider using the default setup and simply signore the LCD SD card reader.

gtrevi, 2019-01-07:
Thanks @gjdodd & @gloomyandy.

In the post update, I did mention "_The only heads-up is how to handle the 2 SD cards (see gloomyandy's point below)_".

That said, the very problem is to get the REPRAP_DISCOUNT_SMART_CONTROLLER's *screen* to display characters, before-hand all the SD card considerations.

So, taking aside the (correct) SD considerations for a moment, what are the correct modifications to make to EXT1 and/or EXT2 (besides flipping them 180 degrees of course), in order to at least see the menus?

boelle, 2019-02-21:
@gtrevi problem fixed?

gtrevi, 2019-02-22:
> @gtrevi problem fixed?

Unfortunately not: I ended up going back to my old Trigorilla, while waiting for my new Duet2 WiFi, which solved all at once.

To bad having 2 MKS-SBASE boards laying around :(

closed 2019-02-22 by gtrevi

boelle, 2019-02-22:
tried bugfix 2.0?

gtrevi, 2019-02-22:
> tried bugfix 2.0?

Yes, was using 2.0, fully compiled w/VS Code

thinkyhead, 2019-02-23:
> Yes, was using 2.0, fully compiled w/VS Code

tried bugfix 2.0 lately?

gjdodd, 2019-02-23:
I could not see any mention of it on the rest the the thread, but on my version of the mks sbase I have had to switch the little dip switches which are in the middle of the board, they have a little clear sticky plastic cover over them. These are used to enable the extra pins on exp2, without them the sd card on the full graphics version will not work. Once these were set then it all worked as expected.

RudyBenner, 2019-02-23:
Thanks for that information.

gloomyandy, 2019-02-23:
@gjdodd what problem did changing those switches actually fix for you? What display are you using? My understanding is that those switches are used to disconnect the SPI lines used to access the external SD card reader. As far as I know the the display not working problem is due to pins not being connected on EXP1 I would not have thought this switch would help with that, but I'm basing that off of the information in the schematic for the 1.2 version so perhaps something has changed?

gtrevi, 2019-02-23:
@thinkyhead yes, I was always on the latest 2.0, but of course back on when I was trying... now I'm running my delta on a duet2 wifi.

@gjdodd this is interesting, can you kindly detail more about your exact setup, including the dip-switch cfg (can't find any doc on that)?

I may want to recycle my SBASE 1.3 + Marlin for my CNC, which is now still running with my custom-designed board and FW w/g-code parser (developed 15 years ago when there were no cheap fancy microstepping drivers I had to implement myself with MCU + mosfets, and no other FW was yet available... of course developing all that crap was part of the project fun :) )

thinkyhead, 2019-02-25:
> now I'm running my delta on a duet2 wifi

I've got one of those to develop on now. Maybe we'll catch up with RepRapFirmware by 2020…

gtrevi, 2019-02-26:
> > now I'm running my delta on a duet2 wifi
> 
> I've got one of those to develop on now. Maybe we'll catch up with RepRapFirmware by 2020…

Terrific, very looking forward to that! ;)

Kyosuke33, 2019-07-19:
Hello, there is news about the problem ? I just buy a MKS SGen 32 bits for my Anycubic Linear Plus, Marlin 2.0 compile and installation is good, but my screen only show white square. Thank you.