MarlinFirmware/Marlin #12800: [2.0.x]  TMC Connection Error when using multiple TMC2208

ErikFontanel, 2019-01-02:
### Description

If I try to use multiple TMC2208 drivers (X, Y & Z) the LCD displays `TMC Connection Error`. M122 reports the following:

```
		X	Y	Z
Enabled		false	false	false
Set current	600	600	600
RMS current	1049	581	1049
MAX current	1479	819	1479
Run current	18/31	18/31	18/31
Hold current	9/31	9/31	9/31
CS actual		0/31	9/31	0/31
PWM scale	0	11	0
vsense		0=.325	1=.18	0=.325
stealthChop	false	true	false
msteps		256	16	256
tstep		0	max	0
pwm
threshold		98	98	658
[mm/s]		1613.52	100.85	48.06
OT prewarn	false	false	false
OT prewarn has
been triggered	false	false	false
off time		0	4	0
blank time	16	24	16
hysteresis
-end		-3	2	-3
-start		1	1	1
Stallguard thrs
DRVSTATUS	X	Y	Z
stst			X
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0xC0:09:00:00
		Z	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... OK
Testing Z connection... Error: All LOW
```

I know the drivers are working because if I switch out the X or Z driver with the Y driver, the driver UART works. But if I use a driver in the X or Z position, those are returning 0x00:00:00:00. Even though the UART connection does not work, the drivers do drive the motors.

I'm using cb643d87df16dd54d6e7779fe8565c9be04ec951 compiled with PIO for the [Fysetc F6](https://wiki.fysetc.com/F6_V1.3/), the original Watterott TMC2208 Silentstepsticks (jumper closed) and TMCStepper 0.2.3. The board allows for 'plug and play' use of the TMC2208, just one jumper on the board needs to be set (and the jumper on the stepsticks need to be closed).

I think it might be because of multiple drivers trying to use the same UART. But I do not know enough about this stuff at all. Or could the board be at fault?

[marlin2configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2721861/marlin2configuration.zip)

Ludy87, 2019-01-03:
is this your Pin definition?

```cpp
 #define X_SERIAL_RX_PIN       71
 #define X_SERIAL_TX_PIN       72

 #define Y_SERIAL_RX_PIN       73
 #define Y_SERIAL_TX_PIN       74

 #define Z_SERIAL_RX_PIN       78
 #define Z_SERIAL_TX_PIN       79

 #define E0_SERIAL_RX_PIN      75
 #define E0_SERIAL_TX_PIN      76
```

or this?

  ```c++
#define X_SERIAL_RX_PIN  71
  #define X_SERIAL_TX_PIN  72

  #define Y_SERIAL_RX_PIN  73
  #define Y_SERIAL_TX_PIN  75

  #define Z_SERIAL_RX_PIN  78
  #define Z_SERIAL_TX_PIN  79

  #define E0_SERIAL_RX_PIN 76
  #define E0_SERIAL_TX_PIN 77
```

Ludy87, 2019-01-03:
Do you use UART?

Ludy87, 2019-01-03:
A resistor is installed on TX?

ErikFontanel, 2019-01-03:
```
#define X_SERIAL_RX_PIN 71
#define X_SERIAL_TX_PIN 72

#define Y_SERIAL_RX_PIN 73
#define Y_SERIAL_TX_PIN 75

#define Z_SERIAL_RX_PIN 78
#define Z_SERIAL_TX_PIN 79

#define E0_SERIAL_RX_PIN 76
#define E0_SERIAL_TX_PIN 77

#define E1_SERIAL_RX_PIN 80
#define E1_SERIAL_TX_PIN 81

#define E2_SERIAL_RX_PIN 22
#define E2_SERIAL_TX_PIN 82
```

No resistor, as far as I understand from the [fysetc docs](https://wiki.fysetc.com/F6_V1.3/#3-tmc2208) it's not necessary.

ErikFontanel, 2019-01-04:
Update:
[image]

I had a look at the schematic to be sure, and there is a resistor after pin 3 on JP 1, which you need to short with pin 2 for the TMC2208.

boelle, 2019-01-06:
resistor is required, look here: https://www.instructables.com/id/UART-This-Serial-Control-of-Stepper-Motors-With-th/

ErikFontanel, 2019-01-07:
[This](https://github.com/MarlinFirmware/Marlin/issues/12800#issuecomment-451263870):

```
// Software serial
#define X_SERIAL_RX_PIN 71
#define X_SERIAL_TX_PIN 72

#define Y_SERIAL_RX_PIN 73
#define Y_SERIAL_TX_PIN 75

#define Z_SERIAL_RX_PIN 78
#define Z_SERIAL_TX_PIN 79

#define E0_SERIAL_RX_PIN 76
#define E0_SERIAL_TX_PIN 77

#define E1_SERIAL_RX_PIN 80
#define E1_SERIAL_TX_PIN 81

#define E2_SERIAL_RX_PIN 22
#define E2_SERIAL_TX_PIN 82
```

ErikFontanel, 2019-01-07:
I'm not sure if this is related per se, but I also can't communicatie with Marlin through Octoprint using the Raspberry Pi GPIO → UART/i2c/SPI. I've tried to enable a second serial port too, but no effect. Only USB works without problems.

ErikFontanel, 2019-01-08:
I currently can't check if the address id is the same, but as in my first message:

> I know the drivers are working because if I switch out the X or Z driver with the Y driver, the driver UART works. But if I use a driver in the X or Z position, those are returning 0x00:00:00:00. Even though the UART connection does not work, the drivers do drive the motors.

I'll check if the address is the same to be sure, but I suspect it is…

Tecumtha, 2019-01-09:
So, i testet my Stepper with the same Board, but using the Klipper firmware. With this Configuration, i can use the  Stepper without any Problems and all 4 runs simultaneously.

Domima, 2019-02-01:
Hello there,
I’m also having this problem, only the Y driver seems too get an address. 
If I change the steppers as Tecumtha suggested, only the stepper on the Y port gets an address the others don’t.
This is what I get after an M122
stealthChop	false	false	true	false	false	false
Driver registers:
		X	0x00:00:00:00	 Bad response!
		X2	0x00:00:00:00	 Bad response!
		Y	0xC0:0C:00:00
		Y2	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing X2 connection... Error: All LOW
Testing Y connection... OK
Testing Y2 connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

My set up is Marlin 2.0.X. on a Fysetc F6 with 5 TMC2208 stepperdrivers. 
If it works on Clipper that suggest a Marlin issue, am I right?

MSergW, 2019-02-02:
I, too, only Y works :(

MSergW, 2019-02-11:
FYSETC F6 V1.3
TMC2208 - UART
work only Y

Testing X connection... Error: All LOW
Testing Y connection... OK
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

bakteriak, 2019-02-13:
It seems I am impacted as well #13106

boelle, 2019-02-21:
@ErikFontanel problem fixed?

MSergW, 2019-02-22:
Commits on Feb 21, 2019 - not fixed :(

boelle, 2019-02-22:
have you checked that pins file is correct?
sometime you need to do a little research and testing yourself
use google
try facebook
....
....

ErikFontanel, 2019-02-22:
@boelle My pins are correct, see [my comment here](https://github.com/MarlinFirmware/Marlin/issues/12800#issuecomment-451969604). An interesting discussion is going on in #13197 though.

boelle, 2019-02-22:
is the cpu on your board 2560?

ErikFontanel, 2019-02-22:
Yep, you can find the info + schematic [here](https://wiki.fysetc.com/F6_V1.3/#hardware-reasources) btw

MSergW, 2019-03-17:
still not fixed, only Y works :(

naka123, 2019-04-29:
as far as I can understand TMCStepper lib uses SoftwareSerial, which only support limited number of pins as RX (see https://www.arduino.cc/en/Reference/SoftwareSerial )

[image]

Fysetc F6 v1.3 have these pins as RX:

PIN:  71   Port: G3        X_SERIAL_RX_PIN
PIN:  73   Port: J3        Y_SERIAL_RX_PIN
PIN:  78   Port: E2        Z_SERIAL_RX_PIN
PIN:  76   Port: J5        E0_SERIAL_RX_PIN
PIN:  80   Port: E7        E1_SERIAL_RX_PIN

so, TMCStepper lib can only communicate with TMC2208 drivers in slots Y and E0 

with this pins_arduino.h file - https://gist.github.com/naka123/7dccdeff141d4ba73f13bfeef8125dba - I have both Y and E0 slots working

`Send: M122
Recv: 		X	Y	Z	E
Recv: Enabled		false	false	false	false
Recv: Set current	700	700	700	350
Recv: RMS current	1215	673	1215	336
Recv: MAX current	1713	949	1713	474
Recv: Run current	21/31	21/31	21/31	10/31
Recv: Hold current	13/31	13/31	13/31	6/31
Recv: CS actual		0/31	13/31	0/31	6/31
Recv: PWM scale	0	15	0	7
Recv: vsense		0=.325	1=.18	0=.325	1=.18
Recv: stealthChop	false	true	false	true
Recv: msteps		256	64	256	32
Recv: tstep		0	max	0	max
Recv: pwm
Recv: threshold		158	39	31	16
Recv: [mm/s]		250.20	253.41	255.04	257.36
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		0	4	0	4
Recv: blank time	16	24	16	24
Recv: hysteresis
Recv: -end		-3	2	-3	2
Recv: -start		1	1	1	1
Recv: Stallguard thrs
Recv: DRVSTATUS	X	Y	Z	E
Recv: stst			X		X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0xC0:0D:00:00
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0xC0:06:00:00
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... OK
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... OK
Recv: ok`

J3 E2 E7 pins dont have PCINT, so Marlin can't communicate with TMC2208 drivers in these slots (X Z E1)

Klipper doesn't use SoftwareSerial or any other Arduino framework parts, so it can use all pins as RX

-----------------

best solution I found - is to buy BIGTREETECH-SKR-V1.3 board. it cheaper, faster (LPC1768 32bit MCU), have more memory and support 4 TMC2208 drivers without any issues

`Send: M122
Recv: 		X	Y	Z	E
Recv: Enabled		false	false	false	false
Recv: Set current	700	700	700	800
Recv: RMS current	673	673	673	795
Recv: MAX current	949	949	949	1121
Recv: Run current	21/31	21/31	21/31	25/31
Recv: Hold current	13/31	13/31	13/31	16/31
Recv: CS actual		13/31	13/31	13/31	16/31
Recv: PWM scale	15	15	15	19
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	false
Recv: msteps		128	128	128	64
Recv: tstep		max	max	max	max
Recv: pwm
Recv: threshold		0	0	0	0
Recv: [mm/s]		-	-	-	-
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		4	4	4	4
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		1	1	1	1
Recv: Stallguard thrs
Recv: DRVSTATUS	X	Y	Z	E
Recv: stst		X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0D:00:00
Recv: 		Y	0xC0:0D:00:00
Recv: 		Z	0xC0:0D:00:00
Recv: 		E	0x80:10:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: ok`

boelle, 2019-06-22:
@thinkyhead missing a label

tmaniac, 2019-07-07:
if i may chip in; (no expert at all here, just reporting my amaturistic findings :))

i just got the TMC connection error and i was also fiddling with the 
#define TX_BUFFER_SIZE property.

The default value is 0.
i changed this to 32 to make proper use of advanced ok and i got the TMC COMMUNICATION error.
When i set the property to 0 or 64, i do not get the TMC error.

boelle, 2019-07-21:
@ErikFontanel have you tried what @tmaniac did?

shitcreek, 2019-09-17:
Cleaning house. LMK if this issue needs to me reopened.

closed 2019-09-17 by shitcreek

otherguy, 2019-10-28:
@shitcreek just ran into this with a FYSETC F6 board and 5 TMC5160 drivers, latest commit on 2.0.x branch.

I have `ADVANCED_OK` enabled and `TMC_DEBUG` as well. Increased `TX_BUFFER_SIZE` to 128, but get the same error.

```
echo:Driver stepping mode:
echo:  M569 S1 X Y
Testing X connection... Error: All LOW
Testing Y connection... OK
Testing Z connection... Error: All LOW
Testing Z2 connection... OK
Testing E connection... OK```

otherguy, 2019-10-28:
Well, that's embarrassing. I forgot to set `default_envs = fysetc_f6_13` in `platformio.ini`.

I cannot test it at the moment but the driver registers look fishy to me:

```
                X       Y       Z       Z2      E
Enabled         false   false   true    true    false
Set current     800     800     800     800     900
RMS current     790     790     790     790     893
MAX current     1114    1114    1114    1114    1259
Run current     23/31   23/31   23/31   23/31   26/31
Hold current    11/31   11/31   11/31   11/31   13/31
Global scaler   130/256 130/256 130/256 130/256 130/256
CS actual       11/31   11/31   11/31   11/31   13/31
PWM scale       10      10      10      10      12
stealthChop     false   true    false   false   false
msteps          16      16      16      16      16
tstep           max     max     max     max     max
pwm
threshold       98      98      658     65535   31
[mm/s]          100     100     3       -       30
OT prewarn      false   false   false   false   false
OT prewarn has
been triggered  false   false   false   false   false
off time        3       3       3       3       3
blank time      24      24      24      24      24
hysteresis
-end            -1      -1      -1      -1      -1
-start          1       1       1       1       1
Stallguard thrs 0       0       0       0       0
DRVSTATUS       X       Y       Z       Z2      E
stallguard
sg_result       0       0       0       0       0
fsactive
stst            *       *       *       *       *
olb
ola
s2gb
s2ga
otpw
ot

Driver registers:
                X       0x80:0B:00:00
                Y       0x80:0B:40:00
                Z       0x80:0B:00:00
                Z2      0x80:0B:00:00
                E       0x80:0D:00:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK
```

Schild0r, 2019-11-17:
Having the same problem: Using an MKS Gen L with 5x TMC2208 in UART none of the drivers gets recognized, all drivers report ALL LOW.
With 1.1.9 everything worked fine. I copied the pin definitions from my 1.1.9 pins_ramps.h file into my config_advanced like this:

```
 #if AXIS_IS_TMC(X)
    #define X_CURRENT       800        //MS (mA) RMS current. Multiply by 1.414 for peak current. up to 950 with 1.5A steppers
    #define X_CURRENT_HOME  X_CURRENT  // (mA) RMS current for sensorless homing
    #define X_MICROSTEPS     16    // 0..256
    #define X_RSENSE          0.11
    #define X_CHAIN_POS      -1    // <=0 : Not chained. 1 : MCU MOSI connected. 2 : Next in chain, ...
    #define X_SERIAL_TX_PIN    40 //MS
    #define X_SERIAL_RX_PIN    63 //MS

  #endif

  #if AXIS_IS_TMC(X2)
    #define X2_CURRENT      800
    #define X2_CURRENT_HOME X2_CURRENT
    #define X2_MICROSTEPS    16
    #define X2_RSENSE         0.11
    #define X2_CHAIN_POS     -1
  #endif

  #if AXIS_IS_TMC(Y)
    #define Y_CURRENT       800
    #define Y_CURRENT_HOME  Y_CURRENT
    #define Y_MICROSTEPS     16
    #define Y_RSENSE          0.11
    #define Y_CHAIN_POS      -1
    #define Y_SERIAL_TX_PIN    59 //MS
    #define Y_SERIAL_RX_PIN    64 //MS
  #endif

  #if AXIS_IS_TMC(Y2)
    #define Y2_CURRENT      800
    #define Y2_CURRENT_HOME Y2_CURRENT
    #define Y2_MICROSTEPS    16
    #define Y2_RSENSE         0.11
    #define Y2_CHAIN_POS     -1
  #endif

  #if AXIS_IS_TMC(Z)
    #define Z_CURRENT       800
    #define Z_CURRENT_HOME  Z_CURRENT
    #define Z_MICROSTEPS     16
    #define Z_RSENSE          0.11
    #define Z_CHAIN_POS      -1
    #define Z_SERIAL_TX_PIN    19 //MS default 42
    #define Z_SERIAL_RX_PIN    11 //MS default 65
  #endif

  #if AXIS_IS_TMC(Z2)
    #define Z2_CURRENT      800
    #define Z2_CURRENT_HOME Z2_CURRENT
    #define Z2_MICROSTEPS    16
    #define Z2_RSENSE         0.11
    #define Z2_CHAIN_POS     -1
  #endif

  #if AXIS_IS_TMC(Z3)
    #define Z3_CURRENT      800
    #define Z3_CURRENT_HOME Z3_CURRENT
    #define Z3_MICROSTEPS    16
    #define Z3_RSENSE         0.11
    #define Z3_CHAIN_POS     -1
  #endif

  #if AXIS_IS_TMC(E0)
    #define E0_CURRENT      800
    #define E0_MICROSTEPS    16
    #define E0_RSENSE         0.11
    #define E0_CHAIN_POS     -1
    #define E0_SERIAL_TX_PIN   42 //MS default 44
    #define E0_SERIAL_RX_PIN   65 //MS default 66
  #endif

  #if AXIS_IS_TMC(E1)
    #define E1_CURRENT      800
    #define E1_MICROSTEPS    16
    #define E1_RSENSE         0.11
    #define E1_CHAIN_POS     -1
    #define E1_SERIAL_TX_PIN   44 //MS default -1
    #define E1_SERIAL_RX_PIN   66 //MS default -1**
  #endif
```

you can see I changed some pins to make all 5 drivers work but I made sure all RX pins are interrupt capable and since they have worked with Marlin 1.1.9 but with Marlin 2.0 none of them reports a connection.
I tried fiddling with the `TX_BUFFER_SIZE` and `ADVANCED_OK` as suggested by @tmaniac and even with `RX_BUFFER_SIZE` but it did not change anything.

Running the drivers in standalone mode now for the time being to make use of Marlin 2.0 but I'd like to be able to use `HYBIRD_THRESHOLD` etc. some time...

Schild0r, 2019-12-01:
I don't know how or why but when I switched from Arduino IDE to VSCode it worked suddenly without a problem.
Maybe Arduino did use some libraries that would not work since my LCD is not showing any artifacts anymore too (which it did with the Arduino build but I thought it might be just an electrical error because it did the same thing with 1.1.9 for obvious reasons but it was less prominent tho)