MarlinFirmware/Marlin #12802: Buzzer warnings on default lcd config for Anet A8

alekc, 2019-01-02:
### Description

Recently I tried to update my Anet A8 to latest firmware version of marlin. Since I didn't have any files available around I cloned bugfix branch (both 1.x and 2.x) and applied suggested configs.  After printer reboot, lcd was blank and a buzing beep was being produced every second or so. Took me several hours to find the issue, looks like default setting ```#define ZONESTAR_LCD ``` doesn't work for A8 with full graphic display (such as 12864) and  ```#define ANET_FULL_GRAPHICS_LCD``` is required instead. 

This whole situation can be scary for first time user because it looks like they have bricked their printer. Might be worth to put a comment near  teh default entry suggesting them to use the other setting in case of problem.

Ludy87, 2019-01-03:
Anet A8 default display is without encoder and buzzer.
`ANET_FULL_GRAPHICS_LCD` is for Anet A6 - some replicas have the big display installed (clone of the clone)

alekc, 2019-01-04:
I see.. I guess we can close this for now then, hopefully if someone encounter the same issue is me they will be able to find this thread.

closed 2019-01-04 by alekc

JeanCarloEM, 2020-07-22:
In Anet A8 PLUS, this BIP you mention occurs and the screen is blank, even with ANET_FULL_GRAPHICS_LCD. If you disable the ZONESTAR_LCD option, scketch will display an error; therefore, there must be something more to be done. For those who don't know, the PLUS screen is LCD-12864. What should I do?