MarlinFirmware/Marlin #12804: KILL called on G28 when job starts.

AndKe, 2019-01-02:
### Description

If printer is preheating, left alone for a while, starting a job result in KILL due to inactivity

### Steps to Reproduce
(last job finished many days ago, with M85 S0;  disable kill)
- switch on heating
-spend a while waiting
start a job, observe that M85 S720  is sent ...
then next command, G28 will move the head, then the job stops...

**Expected behavior:** [What you expect to happen]
expected it not to fail like that, there were no 720s of inactivity since that command were configured

**Actual behavior:** [What actually happens]
```
2018-12-29 10:49:57,296 - Connecting to: /dev/ttyUSB0
2018-12-29 10:49:57,312 - Changing monitoring state from "Offline" to "Opening serial port"
2018-12-29 10:49:57,318 - Connected to: Serial<id=0x7111e330, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2018-12-29 10:49:57,320 - Changing monitoring state from "Opening serial port" to "Connecting"
2018-12-29 10:49:57,332 - Send: N0 M110 N0*125
2018-12-29 10:49:58,400 - Recv: start
2018-12-29 10:49:58,403 - Recv: echo: External Reset
2018-12-29 10:49:58,406 - Recv: Marlin 1.1.9
2018-12-29 10:49:58,404 - Send: N0 M110 N0*125
2018-12-29 10:49:58,408 - Recv: 
2018-12-29 10:49:58,410 - Recv: echo: Last Updated: 2018-08-01 | Author: (none, default config)
2018-12-29 10:49:58,411 - Recv: echo:Compiled: Dec 15 2018
2018-12-29 10:49:58,413 - Recv: echo: Free Memory: 3152  PlannerBufferBytes: 1232
2018-12-29 10:49:58,414 - Recv: echo:EEPROM version mismatch (EEPROM=V47 Marlin=V55)
2018-12-29 10:49:58,415 - Recv: echo:Hardcoded Default Settings Loaded
2018-12-29 10:50:02,269 - Recv: ok
2018-12-29 10:50:02,275 - Changing monitoring state from "Connecting" to "Operational"
2018-12-29 10:50:02,284 - Send: N0 M110 N0*125
2018-12-29 10:50:02,780 - Recv: echo:SD card ok
2018-12-29 10:50:02,795 - Recv: ok
2018-12-29 10:50:02,797 - Send: N1 M115*39
2018-12-29 10:50:02,822 - Recv: FIRMWARE_NAME:Marlin 1.1.9 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:A10 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
2018-12-29 10:50:02,828 - Recv: Cap:SERIAL_XON_XOFF:0
2018-12-29 10:50:02,833 - Recv: Cap:EEPROM:1
2018-12-29 10:50:02,839 - Recv: Cap:VOLUMETRIC:1
2018-12-29 10:50:02,841 - Recv: Cap:AUTOREPORT_TEMP:1
2018-12-29 10:50:02,845 - Recv: Cap:PROGRESS:0
2018-12-29 10:50:02,846 - Recv: Cap:PRINT_JOB:1
2018-12-29 10:50:02,847 - Recv: Cap:AUTOLEVEL:1
2018-12-29 10:50:02,848 - Recv: Cap:Z_PROBE:1
2018-12-29 10:50:02,850 - Recv: Cap:LEVELING_DATA:1
2018-12-29 10:50:02,851 - Recv: Cap:BUILD_PERCENT:0
2018-12-29 10:50:02,853 - Recv: Cap:SOFTWARE_POWER:0
2018-12-29 10:50:02,854 - Recv: Cap:TOGGLE_LIGHTS:0
2018-12-29 10:50:02,855 - Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
2018-12-29 10:50:02,857 - Recv: Cap:EMERGENCY_PARSER:1
2018-12-29 10:50:02,860 - Recv: Cap:AUTOREPORT_SD_STATUS:0
2018-12-29 10:50:02,861 - Recv: Cap:THERMAL_PROTECTION:1
2018-12-29 10:50:02,863 - Recv: ok
2018-12-29 10:50:02,864 - Send: M21
2018-12-29 10:50:02,887 - Recv: echo:SD card ok
2018-12-29 10:50:02,887 - Recv: ok
2018-12-29 10:50:02,890 - Send: M20
2018-12-29 10:50:02,902 - Recv: Begin file list
2018-12-29 10:50:02,903 - Recv: End file list
2018-12-29 10:50:02,905 - Recv: ok
2018-12-29 10:50:02,912 - Send: M155 S2
2018-12-29 10:50:02,918 - Recv: ok

------(I DO SOME PREPERATIONS HERE, preheat it..)

2019-01-02 16:19:04,252 - Recv:  T:1.02 /0.00 B:3.06 /0.00 @:0 B@:0
2019-01-02 16:19:04,945 - Send: M104 S180
2019-01-02 16:19:04,970 - Recv: ok
2019-01-02 16:19:06,247 - Recv:  T:1.41 /180.00 B:3.06 /0.00 @:127 B@:0
2019-01-02 16:19:08,245 - Recv:  T:2.03 /180.00 B:3.25 /0.00 @:127 B@:0
2019-01-02 16:19:09,362 - Send: M140 S65
2019-01-02 16:19:09,390 - Recv: ok
2019-01-02 16:19:10,243 - Recv:  T:5.69 /180.00 B:3.69 /65.00 @:127 B@:0
2019-01-02 16:19:12,256 - Recv:  T:10.16 /180.00 B:2.94 /65.00 @:127 B@:0
......(HEATING - TEMPERATURES REMOVED )..........
2019-01-02 16:27:56,398 - Recv:  T:189.06 /189.00 B:64.57 /65.00 @:53 B@:0
2019-01-02 16:27:58,395 - Recv:  T:189.09 /189.00 B:64.51 /65.00 @:52 B@:127
2019-01-02 16:28:00,394 - Recv:  T:189.09 /189.00 B:64.59 /65.00 @:52 B@:127
2019-01-02 16:28:02,390 - Recv:  T:188.96 /189.00 B:64.81 /65.00 @:56 B@:127
2019-01-02 16:28:03,158 - Send: M104 S0
2019-01-02 16:28:03,175 - Recv: ok
2019-01-02 16:28:04,389 - Recv:  T:189.09 /0.00 B:64.89 /65.00 @:0 B@:127
2019-01-02 16:28:05,785 - Changing monitoring state from "Operational" to "Printing"
2019-01-02 16:28:05,830 - Send: N0 M110 N0*125
2019-01-02 16:28:05,850 - Recv: ok
2019-01-02 16:28:05,859 - Send: N1 M107*36
2019-01-02 16:28:05,875 - Recv: ok
2019-01-02 16:28:05,882 - Send: N2 M190 S65*89
2019-01-02 16:28:05,890 - Recv: ok
2019-01-02 16:28:05,894 - Send: N3 M104 S205*97
2019-01-02 16:28:05,906 - Recv: ok
2019-01-02 16:28:05,910 - Send: N4 M851 Z-1.84*111
2019-01-02 16:28:05,921 - Recv: ok
2019-01-02 16:28:05,924 - Send: N5 M85 S720*93
2019-01-02 16:28:05,938 - Recv: ok
2019-01-02 16:28:05,942 - Send: N6 G28*21
2019-01-02 16:28:05,968 - Recv: Error:KILL caused by too much inactive time - current command: M85 S720
2019-01-02 16:28:05,970 - Recv: Error:Printer halted. kill() called!
2019-01-02 16:28:05,973 - Changing monitoring state from "Printing" to "Error: KILL caused by too much inactive time - current command: M85 S720 - Printer halted. kill() called!"
2019-01-02 16:28:05,987 - Changing monitoring state from "Error: KILL caused by too much inactive time - current command: M85 S720 - Printer halted. kill() called!" to "Offline (Error: KILL caused by too much inactive time - current command: M85 S720 - Printer halted. kill() called!)"

```

#### Additional Information
 this is recent master 1.1.9

boelle, 2019-01-06:
@AndKe did @Ludy87 suggestion help?

AndKe, 2019-01-06:
I am not sure why that would solve anything, if time max_inactive_time was never used, it should not be "true" as-is ?   . .... it also seems to imply that the error can only spawn if this variable was never set?

The printer's low-power part is always-on, so it could also be I saw this eller after millis have overflowed  (after a week since last use or so?)

Regardless, I just had some other changes, so this change is to be flashed within minutes, not sure how if I can tell right away whatever it helped, as I am not sure whatever I can reproduce it every time at will.

AndKe, 2019-01-06:
ok, with @Ludy87 's fix:
this happened on my first attempt to print (and before that, heat for a while,, to reproduce)

```
Send: N5 M851 Z-1.84*110
Recv: ok
Send: N6 M85 S720*94
Recv: ok
Send: N7 G28*20
Recv:  T:31.12 /208.00 B:70.06 /70.00 @:127 B@:127
Recv: echo:busy: processing
Recv:  T:31.77 /208.00 B:70.29 /70.00 @:127 B@:127
Recv: echo:busy: processing
Recv:  T:33.80 /208.00 B:70.36 /70.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:37.19 /208.00 B:70.32 /70.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:41.08 /208.00 B:70.20 /70.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:46.20 /208.00 B:69.93 /70.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:51.05 /208.00 B:69.71 /70.00 @:127 B@:0
Recv: echo:busy: processing
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Printing" to "Error: Printer halted. kill() called!"
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"
Connection closed, closing down monitor
```

So I reconnected, and just started a print.. and it prints (I did not rey to reproduce second time)
Observe that this error appeared about the same time in process, but is telling even less about *why*

Ludy87, 2019-01-06:
This statement is too vague for me to trigger a PR.

Ludy87, 2019-01-06:
If the variable was created with `M85 S720` and with `M85 S0`
default is set, the variable has the value `0` and not `undefined`.

AndKe, 2019-01-12:
The suggested fix did not help, got the same error today.

boelle, 2019-02-21:
@AndKe problem fixed?

AndKe, 2019-02-21:
No, I just don't have a good way of reproducing it, observed it a few times when after a job, I only preheat, maybe make some steppers move, then a new job is expected to start.

boelle, 2019-02-21:
using bugfix 2.0?

AndKe, 2019-02-21:
no, 1.1.8 , maybe a master somewhere after 1.1.8 release.

boelle, 2019-02-21:
no use 2.0

AndKe, 2019-02-21:
"Not for production use. Use with caution!"  is kind of scary :)    - will try this weekend. 
I am surprised there is no summary of features except 32bit, which my controller can't do anyway.

boelle, 2019-02-21:
It's both 8 bit and 32 bit

thinkyhead, 2019-02-23:
> I am surprised there is no summary of features except 32bit

All the same features as 1.1.9, plus 32-bit, plus extra goodies.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-11 by thinkyhead

AndKe, 2019-12-25:
I have found one way to reproduce it on current 2.x bugfix:

1: Connect
2: M85 S60
3: M85 S60  ;  YES, repeat the command after one second or so.
4: observe KILL

Reconnect
do M999 in case that resets something
now you can reproduce it over again.

So: 
If your jobs start with M85 S300  and end with M85 S0  , then any cancelled job that do never get to S0 , will fail instantly on new M85 Sxxx

AndKe, 2020-01-06:
@Ludy87  / @boelle   ... please see above

thinkyhead, 2020-04-29:
@AndKe — Does the issue still exist in the current `bugfix-2.0.x` code?

AndKe, 2020-04-29:
Yes, I did not test in April, but narrowed it to this:
if I set temperature , like bed preheating from LCD, then start the job later, it will fail.
it seems that setting temperature/preheating via G-code (octopi interface) does not cause the same problem.

Also: if M85 is configured, and printer stops due to filament runout , it will be killed unless user changes filament within timeout.

thinkyhead, 2020-05-01:
Thanks. I'll look into those issues. They are subtle and will be interesting to solve.

madl3x, 2020-05-02:
The initialization of "max_inactive_time" looks to be commented. Couldn't be that one of the issues?

thinkyhead, 2020-06-29:
There is more insurance against timeouts in Marlin these days, but I might try adding some more.