MarlinFirmware/Marlin #12805: Power Loss Recovery Z-Axis Error

engineeren, 2019-01-02:
Hi! I have a problem with power loss recovery. When I start power loss resuming, X&Y axis go to home and LCD shows X&Y positions is "0". But for Z-axis it shows "???". Print resuming works but z-axis don't raise for next layers. I'm waiting for your helps.

I use bugfix-1.1.x version and the all files here:
[Marlin-bugfix-1.1.x Configure.zip](https://github.com/MarlinFirmware/Marlin/files/2722209/Marlin-bugfix-1.1.x.Configure.zip)

jhosorios, 2019-02-05:
I'm no body, but I request the help of @Roxy-3D for this Issue. Please!!

InsanityAutomation, 2019-02-05:
First off, nearly all of the work on this was done by @thinkyhead not Roxy. Second work has been done in this area in Marlin 2.0 so please validate the issue exists there still first.

InsanityAutomation, 2019-02-05:
Not intended to explicitly be unkind, simply that 1.1.9 is rather dated and the 1.1.9 branch is considered end of life. No more bugfixes or changes are going there. If it is something broken, it would only be fixed in 2.0 and knowing if the problem exists there or not is the important piece of information. Developers, especially the core maintainers, tend to have allot going on and get messages for help from every forum, social media platform, and other known contact point we have, so yes in some cases we tend to be short and to the point.

jhosorios, 2019-02-05:
Well, that´s right you can tend to be short and to the point, but please, in a good way. We know you have a lot of knowledge, but forget to be nice to those who do not have it and that is quite annoying. Knowledge is power but it is more important to be a person, a good person, generates better results. I hope you learn that before it's too late. 
Anyway, thanks for the information, it sure has been clear.

boelle, 2019-02-21:
@engineeren problem fixed?

engineeren, 2019-02-21:
No. Deactive the line;
//#define NO_MOTION_BEFORE_HOMING
did not any changing. 
Now I begin to use bugfix 2.0 version of marlin and did not try power loss features yet. I hope this issue get solution by this version. So I will reply again when I try power loss features on this version.

boelle, 2019-02-21:
only bugfix 2.0 is updated

and if this is a config error then its not a marlin issue

engineeren, 2019-02-21:
I don t thing config error. Because, I tried many config settings. One of them, just activate power_loss_resuming line.
Maybe, bugfix 1.9 or my sd card cause this issue, I dont know yet. Tomorrow I try somethings to fix this.

boelle, 2019-02-21:
remember to close this one and open a new issue if problem is there

engineeren, 2019-02-21:
Ok I will.

boelle, 2019-02-21:
you did not close

closed 2019-02-21 by engineeren