MarlinFirmware/Marlin #12808: [2.0] Dual extruder /single Nozzle support ?

Le-zOU, 2019-01-03:
### Description

Printing dog from my A20M Geeetech, it is suppose to start ptintying from E0, then mix E0/E1 and then finish with E1 only.

It only prints using E0

### Steps to Reproduce

1. use Marlin 2.0
2. print dog (color mixing)

**Expected behavior:** mix filament from both extruders

**Actual behavior:** only uses E0

#### Additional Information

print gcode
[Dog_Y20.zip](https://github.com/MarlinFirmware/Marlin/files/2723639/Dog_Y20.zip)
Marlin FW in use
[Marlin-2.0.X_A20M.zip](https://github.com/MarlinFirmware/Marlin/files/2723641/Marlin-2.0.X_A20M.zip)

AnHardt, 2019-01-03:
All your colour definition look about like
```
M163 S0 P0.64
M163 S1 P0.36
M164 S4 ; Save the color to tool 4
```

But you never actually select tool 4 - to be exact, you never select any tool.

Either save the collor to tool 0 (the default tool), or better select tool 4 with a `T4`.

Le-zOU, 2019-01-03:
Thank you.

Could it be that T4 is a virtual tool with pre-mixed filament settings from T0 and T1 ? (aka not using M163/M164)
That would have been configured in the original Marlin 1.1.8 configuration, which I am not using anymore since I'm using Marlin 2.

I know that I configured my Cura with 5 extruders
1 E1 100% E2 0%
2 E1 0% E2 100%
3 E1 50% E2 50%
4 E1 75% E2 25%
5 E1 25% E2 75%

I've not tried to use the mixed extruders since I upgraded to marlin 2.
But 1 and 2 are working just fine at the moment for non mixed dual color prints.

Cura star gcode:
```
M163 S0 P0.75
M163 S1 P0.25
M164 S2
M163 S0 P0.50
M163 S1 P0.50
M164 S3
M163 S0 P0.25
M163 S1 P0.75
M164 S4
T0
M109 S{material_print_temperature};
M104 T{initial_extruder_nr} S{material_print_temperature};
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
G92 E0
G1 F200 E3
G92 E0
``` 

My current configuration.h has:

```
#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
  #define GRADIENT_MIX             // Apply a gradient between mixes over several layers
#endif
```

Le-zOU, 2019-01-05:
ok, but it should work with extruder configured as mixed filament in the start Gcode ?

Le-zOU, 2019-01-05:
yes that was the file on the SD card provided by geetech to work on 1.1.8 I guess. And since i upgraded to 2.0... well unexpected result.
I will try with another model freshly sliced by myself :)

Le-zOU, 2019-01-14:
Hello,

I have a menu on Marlin 2 that allows me to load/unload new filament in the steppers.
So I can load/unload E1 and E2 independantly.
it works just fine for E1, but selecting E2 has no effect whatsoever. Option is present, but no bip or change when I press LCD knob.

Any file I could upload to help you look at that ?

Thank you

Le-zOU, 2019-01-25:
hopefully that'll fix this: https://youtu.be/SispDLjKXLI

boelle, 2019-02-21:
@Le-zOU problem fixed?

Le-zOU, 2019-02-25:
Not really, using that workaround:
https://github.com/Ultimaker/Cura/issues/2918#issuecomment-357540778

boelle, 2019-03-13:
@Le-zOU could you add [BUG] in title so it becomes [BUG][2.0] Dual extruder /single Nozzle support ?

shitcreek, 2019-03-14:
Forgot to update you on this incase you haven't noticed already, bugfix 2.0 now has gradient.

Le-zOU, 2019-03-14:
yes it does :)
I have yet to check it M163/M164 commands are working.
But yes, it is working from LCD .

boelle, 2019-03-14:
@Le-zOU so this one is "fixed"?

closed 2019-07-11 by thinkyhead