MarlinFirmware/Marlin #12809: Power Loss Recovery error printer start at

TB1405, 2019-01-03:
hi 
the printer simply restarts the gcode. 
He also wants to home the z axis and will do AUTO_BED_LEVELING_LINEAR. 
no new file will be created on the SD card

boelle, 2019-01-06:
attach your 2 config files as a zip file

TB1405, 2019-01-06:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2730488/Configuration.zip)

boelle, 2019-02-21:
@TB1405 and this is still a problem in bugfix 2.0?

maxlinux2000, 2019-04-07:
I think we can close this bug. Now the power loss, resume perfectly, but there is another problem,

please see: https://github.com/MarlinFirmware/Marlin/issues/13541

robbycandra, 2019-04-11:
The problem fixed in bugfix 2.0.
Print start over happened if you use cura as slicer and power failure happened at Z less than 15mm.
To fix this. Delete G1 Z15 at line 11 of your gcode files.

closed 2019-07-11 by thinkyhead