MarlinFirmware/Marlin #12812: Feature suggestion overheat protection

mchaos, 2019-01-03:
Marlin already has great thermal overrun protection however there is 1 more thing that can be done. If someone is using MOSFETS for ether of the heating elements there is a risk of it failing open and it running away anyway.  A feature to turn the layer fan on in this situation  will blow on the hot end nozzle and keep it from getting too hot. Typically once you get over 230 the layer fan can keep it from getting any hottoer.

AnHardt, 2019-01-03:
Or the fan will help the fire to ignite from smoldering.

InsanityAutomation, 2019-01-03:
Not to mention a well adjusted fan in an open area will blow below the heat block and not have any impact. My normal suggestion for beds is a snapdisc thermal couple. Maybe with the high is counts of some 32bit boards redundant mosfets with feedback from each to detect an issue could be done but that's a moonshot goal at best.

AnHardt, 2019-01-03:
Marlins 'normal' protection against defect (permanent on) MOSFETS is switching of (heater) power by PS_ON pin.

boelle, 2019-01-06:
:-) i was about to say that it would not help if people dont use an ATX psu, but a relay to the PS_ON pin could be used to kill the mains power

but i dont think that marlin as such can do much more, maybe this issue can be closed but serve as a reminder that if you use low quality hardware you are playing with fire :-)

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-11 by thinkyhead