MarlinFirmware/Marlin #12824: [2.0.x]STM32F4 L6470 error

ffnull, 2019-01-05:
Im write my own board variant for STEVAL-3DP001V1. All compiles fine. But in my board SPI drivers L6470.
When i use it in configuration - doesnt compile with errors.

c:\users\ffnul\appdata\local\temp\arduino_build_575834\sketch\src\core\drivers.h:30:29: error: expected identifier before numeric constant

 #define L6470               0x105

                             ^

D:\Documents\Arduino\libraries\Arduino-L6470-master\src/L6470.h:238:7: note: in expansion of macro 'L6470'

 class L6470 {

       ^~~~~

c:\users\ffnul\appdata\local\temp\arduino_build_575834\sketch\src\core\drivers.h:30:29: error: expected unqualified-id before numeric constant

 #define L6470               0x105

                             ^

D:\Documents\Arduino\libraries\Arduino-L6470-master\src/L6470.h:238:7: note: in expansion of macro 'L6470'

 class L6470 {

       ^~~~~

c:\users\ffnul\appdata\local\temp\arduino_build_575834\sketch\src\core\drivers.h:30:29: error: expected unqualified-id before numeric constant

 #define L6470               0x105

                             ^

c:\users\ffnul\appdata\local\temp\arduino_build_575834\sketch\src\module\stepper_indirection.h:86:10: note: in expansion of macro 'L6470'

   extern L6470 stepperX;

          ^~~~~

c:\users\ffnul\appdata\local\temp\arduino_build_575834\sketch\src\core\drivers.h:30:29: error: expected unqualified-id before numeric constant

 #define L6470               0x105

                             ^

c:\users\ffnul\appdata\local\temp\arduino_build_575834\sketch\src\module\stepper_indirection.h:121:10: note: in expansion of macro 'L6470'

   extern L6470 stepperY;

          ^~~~~

c:\users\ffnul\appdata\local\temp\arduino_build_575834\sketch\src\core\drivers.h:30:29: error: expected unqualified-id before numeric constant

 #define L6470               0x105

                             ^

c:\users\ffnul\appdata\local\temp\arduino_build_575834\sketch\src\module\stepper_indirection.h:156:10: note: in expansion of macro 'L6470'

   extern L6470 stepperZ;

          ^~~~~

c:\users\ffnul\appdata\local\temp\arduino_build_575834\sketch\src\core\drivers.h:30:29: error: expected unqualified-id before numeric constant

 #define L6470               0x105

                             ^

Bob-the-Kuhn, 2019-01-05:
Please attach your platformio.ini and configuration files & I'll see if I can reproduce it.

I've been playing with some L6470 drivers the last week.  What hardware are you using?

ffnull, 2019-01-05:
> Please attach your platformio.ini and configuration files & I'll see if I can reproduce it.
> 
> I've been playing with some L6470 drivers the last week. What hardware are you using?

Im doesnt use platformio becouse im use Arduino_Core_STM32. Im compile in arduino
Im attach my board variant for Arduino_Core_STM32 and project itself
[steval.zip](https://github.com/MarlinFirmware/Marlin/files/2729494/steval.zip)
[ST3DP_EVAL.zip](https://github.com/MarlinFirmware/Marlin/files/2729495/ST3DP_EVAL.zip)

ffnull, 2019-01-05:
Im using Marlin 2.0 files im wrote by my self. im try make port for this board. but stuck on motor drivers

Bob-the-Kuhn, 2019-01-05:
Now I understand.

I've been doing basically the same thing the last two weeks but for a Due based system.

The STEVAL-3DP001V1 schematics show that this board uses SPI daisy chaining.  Neither Marlin nor the Arduino L6470 library support this.

My software is mostly functional but is an UGLY RAT'S NEST.  If you want I can modify it so you can spin steppers with your board.  The wife wants me to do some things so it'll be a couple of hours before I can make the mods and post them.

ffnull, 2019-01-05:
I canty change motor drivers so if you can help - its will be great! Then i can move to future
)

Bob-the-Kuhn, 2019-01-05:
Do you know how SPI daisy chaining works?

ffnull, 2019-01-05:
> Do you know how SPI daisy chaining works?

Nope

Bob-the-Kuhn, 2019-01-05:
There's one slave select that's tied to the chip select on all the drivers.

MOSI from the CPU is tied to SDI on the first driver.  SDO on the first driver is tied to SDI on the second driver.  SDO on the second driver is tied to SDI on the third driver.  On the last driver in the chain the SDO is tied to MISO on the CPU.

Since you have 6 drivers all SPI transfers will be 6 bytes long. 

When select is low (active), each device echoes what it saw on it's SDI on the NEXT byte..  The result is each byte is passed along the chain sort of like a bucket line.  Data/commands for driver 6 (the one with it's SDO tied to MISO) is the first byte transmitted.  The last byte transmitted is for the first device in the chain (the one with its SDI tied to MOSI).  When select goes high each driver acts on the byte it last saw. Select stays low from start to finish.

The  [Maxim app note](https://www.maximintegrated.com/en/app-notes/index.mvp/id/3947 )has a good diagram of the data movement.  Just be aware that it's talking about 16 bit devices.

The very first data transmitted by each device is the response to the command it received in the previous 6 byte transfer.  After that it echoes the SDI input.  

The CPU always transmits commands.  If the CPU has no command for a particular device then the NOOP (data 0x00) is transmitted in that slot.

ffnull, 2019-01-05:
I'm not so good at programming. I'm just learning. Can you help me with porting? My board is on the list, but it appears to be not tested as the developers have no board yet. So I decided to help. But stuck with the engines because there is not enough knowledge. If the fee is working, I will send my code to the developers.

Bob-the-Kuhn, 2019-01-05:
Shoot - I was hoping that you would clean up my mess.  I'm in awe at how the big boys can make C++ sing.  My code is like fingernails scraping on a chalkboard.

I just got up from a nap.  I should be able to post a copy in the hour.  It'll take longer to write up what I've done.

ffnull, 2019-01-05:
thank you very much

Bob-the-Kuhn, 2019-01-05:
My software isn't sophisticated enough to handle a 6 device chain of which only 4 are active.  The work around is to assign the unused drivers to Y2 and Z2.  I think that's better than assigning them to E1 7 E2 (lfewer configuration options to set).

Here's your config added to my code. 

As feared, the STM environment/libraries are causing compile errors.  That's been an ongoing headache for the Marlin community.  I'll see what I can do after another nap.

[Marlin-Bob-2.zip](https://github.com/MarlinFirmware/Marlin/files/2729826/Marlin-Bob-2.zip)

Bob-the-Kuhn, 2019-01-05:
First some real top level items:

- I had to modify the Arduino L6470 library.  I couldn't figure out how to use the modified code as a library so it's now buried within Marlin as **\Marlin-Bob-2\Marlin\src\module\bob_L6470.cpp** and **\Marlin-Bob-2\Marlin\src\module\bob_L6470.h**.
- Most of the base code is in the directory **\Marlin\src\module\6470**
- I've written some utilities to help tune the system.  They're in **Marlin-Bob-2\Marlin\src\gcode\feature\L6470**
- I've hardwired in a special software SPI transfer routine.  to do the 6 byte transfers.  You may be able to use **\Marlin-Bob-2\Marlin\src\HAL\HAL_DUE\HAL_spi_L6470_Due.cpp** as is
- Creation of the stepper objects is done starting at line 792 in the file **Marlin-Bob-2\Marlin\src\module\stepper_indirection.cpp**
- Don't push the drivers yet.  Mine went up in smoke when I had them all very near the max power before an over temperature warning happened.  I need to write a routine similar to TMC_Monitor that'll get the status several times a second and reduce the power if the warning flag is set too often.

ffnull, 2019-01-05:
https://pastebin.com/mnyia6FY

Bob-the-Kuhn, 2019-01-06:
I've managed to confuse myself trying various platformio options.

I've run into an older issue on the same topic (issue #10739 ).  There's a comment towards the end that someone got it to compile but by then the person with the board had lost interest.  I'll try that & see what happens.

FYI - here's an updated pins file.  I've added the E1 & E2 socket definitions.

[pins_STEVAL.h.txt](https://github.com/MarlinFirmware/Marlin/files/2730160/pins_STEVAL.h.txt)

Bob-the-Kuhn, 2019-01-06:
Sorry - I just realized that you were the initiator of the earlier issue.

Bob-the-Kuhn, 2019-01-06:
@xC0000005 - your help would be appreciated.

Bob-the-Kuhn, 2019-01-06:
I've tried compiling the example STM32F4 config with platformio and Arduino 1.9.0 beta with equally disappointing results.

Time to wait until someone that knows the STM32 HAL joins the conversation.

xC0000005, 2019-01-06:
Your explanation of how this works is perfect. I'll see if I can get it to compile on the ST hal tomorrow, since this is something I'd actually like to do for the lerdge boards now that Marlin is nearly running on them (they don't have individual SPI select pins, either).

Bob-the-Kuhn, 2019-01-06:
What do they have that is SPI daisy chained?

Bob-the-Kuhn, 2019-01-06:
The latest Arduino 1.9 came with a virus.  How Nice.

When compiling the example I get:
**panic: interface conversion: error is *exec.Error, not *exec.ExitError**

Looks like the tool chain is having problems.

Bob-the-Kuhn, 2019-01-07:
Here's my updated L6470 library.
[L6470.zip](https://github.com/MarlinFirmware/Marlin/files/2731203/L6470.zip)

And here is my image that uses them,
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2731205/Marlin.zip)

They are not tested. I'm hoping to get in new hardware Wednesday ot Thursday so it'll be the weekend before i can have them tested.

---

This code will use an internal soft SPI or your external SPI, depending on how you initialize it.

xC0000005, 2019-01-07:
Here's what I'm seeing regarding @ffnull's error.
`In file included from /var/folders/2w/45wss0310nbfs7dbvg2mv01c0000gn/T/arduino_build_388469/sketch/src/inc/MarlinConfigPre.h:37:0,
                 from /var/folders/2w/45wss0310nbfs7dbvg2mv01c0000gn/T/arduino_build_388469/sketch/src/inc/MarlinConfig.h:28,
                 from /var/folders/2w/45wss0310nbfs7dbvg2mv01c0000gn/T/arduino_build_388469/sketch/src/Marlin.h:24,
                 from /var/folders/2w/45wss0310nbfs7dbvg2mv01c0000gn/T/arduino_build_388469/sketch/src/Marlin.cpp:31:
/var/folders/2w/45wss0310nbfs7dbvg2mv01c0000gn/T/arduino_build_388469/sketch/src/inc/../core/drivers.h:30:29: error: expected identifier before numeric constant
 #define L6470               0x105
                             ^
/var/folders/2w/45wss0310nbfs7dbvg2mv01c0000gn/T/arduino_build_388469/sketch/src/module/L6470.h:241:7: note: in expansion of macro 'L6470'
 class L6470 {
       ^~~~~`

I think this error is sort of teling us the truth.
In drivers.h, we have the line:
#define L6470               0x105
In l6740.h we have:
class L6470 {

Which should expand in the preprocessor to
class 0x105 {

I renamed the driver class to L6470Driver and was able to successfully compile (though I don't know if it works).

xC0000005, 2019-01-07:
> What do they have that is SPI daisy chained?

Nothing - I was thinking of SPI daisy chaining some drivers since the Lerdge-X can't use SPI due to not having a separate chip select. I had no idea daisy chaining existed, but the moment I saw it, I thought "Hmm. I could definitely use that to hook the pins up."

Bob-the-Kuhn, 2019-01-07:
That's strange.

I can compile in PlatformIO for the Due as is but not for STM32.

ffnull, 2019-01-09:
Sorry for delay. Still cant compile. Im stuck(( Here paste https://pastebin.com/D24J4RCE

Bob-the-Kuhn, 2019-01-10:
I'm at the same point as you.  False linker errors.  I've tried moving things around but nothing helped.

Here's my latest L6470 software modified for your system.

Since you physically have a 6 device chain,  my software demands that 6 steppers be defined.  I added Y2 & Z2 to your configuration.  That shouldn't affect anything.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2743969/Marlin.zip)

'You'll also need to replace your L6470 library files with these.

[L6470.zip](https://github.com/MarlinFirmware/Marlin/files/2743968/L6470.zip)

Bob-the-Kuhn, 2019-01-10:
Already found a problem.  I defined the chain wrong.  Chain positions start at 1, not 0.  

Time to call it a night

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2743983/Configuration_adv.zip)

xC0000005, 2019-01-10:
I’m trying to compile this and hitting errors like:
var/folders/2w/45wss0310nbfs7dbvg2mv01c0000gn/T/arduino_build_121505/sketch/src/HAL/shared/HAL_spi_L6470.cpp: In function 'void L6470_SPI_init()':
/var/folders/2w/45wss0310nbfs7dbvg2mv01c0000gn/T/arduino_build_121505/sketch/src/HAL/shared/HAL_spi_L6470.cpp:142:13: error: 'L6470_BUSY_PIN' was not declared in this scope
     pinMode(L6470_BUSY_PIN, INPUT);

Bob-the-Kuhn, 2019-01-10:
And yet another file I screwed up.
[HAL_spi_L6470.zip](https://github.com/MarlinFirmware/Marlin/files/2744025/HAL_spi_L6470.zip)

xC0000005, 2019-01-10:
Was it just the path, or are there other changes I need?

Bob-the-Kuhn, 2019-01-10:
Here's the file with that pin made optional.

[HAL_spi_L6470 (2).zip](https://github.com/MarlinFirmware/Marlin/files/2744059/HAL_spi_L6470.2.zip)

I'm sure there are bugs in the software but I think these changes should eliminate the compile errors.

xC0000005, 2019-01-10:
Ok, after commenting out the missing method calls, I was able to link a binary. I do not know that it works, but it definitely compiles.
However: My environment, I’ve updated the ST core package to work from my git repo for it. There may be fixes in that that are not in the PIO version. If you download the ST core repo and replace the entire contents of the pIO package, see if it still fails to link.

Bob-the-Kuhn, 2019-01-10:
You're talking about replacing something in my platformio directory?

I can't find any directory structures similar to what I just downloaded.

Bob-the-Kuhn, 2019-01-10:
Found it in Arduino

Bob-the-Kuhn, 2019-01-10:
Different errors.  Now it's complaining about syntax errors that I don't see a problem with.

```
In file included from C:\Users\bobku\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\cores\arduino\stm32\PinConfigured.c:30:0:

C:\Users\bobku\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\cores\arduino\stm32\PinConfigured.c:32:45: error: expected ')' before '*' token

 bool is_pin_configured(PinName pin, uint32_t* map) {

                                             ^

C:\Users\bobku\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\cores\arduino\stm32\PinConfigured.h:45:31: note: in definition of macro 'PINCONF_VAL'

 #define PINCONF_VAL(X, Y)   ((Y >> PINCONF_SHIFT(X)) & PINCONF_MASK)

                               ^

C:\Users\bobku\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\cores\arduino\stm32\PinConfigured.c:32:6: note: in expansion of macro 'is_pin_configured'

 bool is_pin_configured(PinName pin, uint32_t* map) {

      ^~~~~~~~~~~~~~~~~
```

xC0000005, 2019-01-10:
How are you building? I’ll see if I can set up another build machine to mimic what you’re seeing.

Bob-the-Kuhn, 2019-01-10:
A4rduino IDE

Yes - https://github.com/stm32duino/Arduino_Core_STM32

xC0000005, 2019-01-10:
All right. Let me see what I can do here. For board, I have selected the ST3DP Eval, and @ffnull’s variant is in my variants folder.

Bob-the-Kuhn, 2019-01-10:
I also made the following mods but they didn't seem to make a difference
variant - add pins PC14 & PC15
?? - uncommented flag GENERIC_STM32F401xx?  don't remember the details

Bob-the-Kuhn, 2019-01-10:
I'll add his variant.

How did you get  ST3DP Eval, as an option?  Did it popup after adding the variant?

xC0000005, 2019-01-10:
The easiest way to get it (and it’s sort of a hack) is to copy MY ST core in on top of your existing one (I know, I’m destroying everything). 
https://github.com/xC0000005/Arduino_Core_STM32/tree/ST_3DPEVALVariant <https://github.com/xC0000005/Arduino_Core_STM32/tree/ST_3DPEVALVariant>

This updates boards.txt, which is how the arduino IDE finds boards. Eventually, I’ll do a PR back to add in the Lerdge boards that are actually in this poorly named branch. Once you do that, copy @ffnul’s variant in on top of the ST3DP_EVAL variant files (I have no idea what he changed from mine, but if that’s what he’s trying to compile with, I’m fine with it).
Once you restart the Arduino IDE, you’ll see a bunch of new boards - A couple of Malyan ones, a couple of Lerdge ones, and ST’s eval board. Choose it.

Bob-the-Kuhn, 2019-01-10:
Need to go soon so it'll probably be this evening before I can report results.

Bob-the-Kuhn, 2019-01-10:
I'm getting 5 or 6 errors all along the line of:
```
C:\Users\bobku\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\cores\arduino\stm32\PinConfigured.h:50:3: note: in definition of macro 'set_pin_configured'

  (map[PINCONF_INDEX(pin)] = map[PINCONF_INDEX(pin)] | PINCONF_BIT(pin))

   ^~~

C:\Users\bobku\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\cores\arduino\stm32\PinConfigured.c:42:48: error: expected ')' before '*' token

 void reset_pin_configured(PinName pin, uint32_t* map) {
```

and

```
C:\Users\bobku\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\cores\arduino\stm32\digital_io.c:92:10: error: redefinition of 'digital_io_read'

 uint32_t digital_io_read(GPIO_TypeDef *port, uint32_t pin)

          ^~~~~~~~~~~~~~~

In file included from C:\Users\bobku\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\cores\arduino\stm32\digital_io.c:38:0:

C:\Users\bobku\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\cores\arduino\stm32\digital_io.h:72:24: note: previous definition of 'digital_io_read' was here

 static inline uint32_t digital_io_read(GPIO_TypeDef *port, uint32_t pin)

                        ^~~~~~~~~~~~~~~
```

Bob-the-Kuhn, 2019-01-11:
SUCCESS!! mostly

I go it to compile but I had to do the following:

- modified file **PinConfigured.h** to use the definitions in the source file for **is_pin_configured**, **set_pin_configured** & **reset_pin_configured**.  I have no idea which one is correct.
- Modified file **digital_io.h** to use the definitions in the source file for **digital_io_write** & **digital_io_read**.  I have no idea which one is correct.
- added USB full speed menu option to **boards.txt** for this board
- modified **usbd_cdc_if.h** to default to timer 10 .  The default of timer 6 doesn't exist on the 32F401 boards. TO DO:  find out where the right place is to set these definitions.
- fixed a logic error I made in the Marlin file **Conditionals_post.h**
- added the flags  **#define STM32GENERIC**,  **#define STM32F4**,  **#define STM32F4xx** &  **#define TEMP_TIMER 9** to the Marlin file **Configuration.h**. TO DO:  find out where the right place is to set these definitions.

I compared the RAM and FLASH usage to the same config if compiled to the Due:  I sort of expected the FLASH to be about the same.
```
                  FLASH     RAM
    STM32F401VE   89420   13136
    Due          119848    8136
```

I also switch from port-pin numbering to logical pin numbering.

[modified files.zip](https://github.com/MarlinFirmware/Marlin/files/2748363/modified.files.zip)

ffnull, 2019-01-11:
Will try today but later. Pin defenitions are good commented by me in board variant in folder of lybrary that i send. Native pins are xpwm xreset xdir....

Bob-the-Kuhn, 2019-01-19:
Here's the latest & greatest with your config files.:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2775155/Marlin.zip)

You'll need to replace the L6470 Arduino files with these:
[L6470.zip](https://github.com/MarlinFirmware/Marlin/files/2775157/L6470.zip)

I was able to compile using Arduino 1.9 beta.  Here's the image files:
[Steval_image.zip](https://github.com/MarlinFirmware/Marlin/files/2775158/Steval_image.zip)

The pins_STEVAL.h file has been updated.  I'll do a PR to move this into Marlin.

Bob-the-Kuhn, 2019-01-19:
I forgot to ask … there are some zero Ohm resistor/jumper locations to remove unused L6470s from the SPI chain.  Let me know if you install any of them so I can modify configuration_adv.h accordingly.

Bob-the-Kuhn, 2019-01-19:
Never mind - forgot to include a correction.

thinkyhead, 2019-01-20:
The above issue has now been patched.
```
 #define L6470               0x105
/var/folders/2w/45wss0310nbfs7dbvg2mv01c0000gn/T/arduino_build_388469/sketch/src/module/L6470.h:241:7: note: in expansion of macro 'L6470'
 class L6470 {
```

Bob-the-Kuhn, 2019-01-24:
Marlin 2.0.x will soon fully support your board.  PR #12895 has added the code.  All that remains is to merge PR #12994 (pins_STEVAL.h fix).

boelle, 2019-02-21:
@ffnull problem fixed?

closed 2019-03-03 by thinkyhead

ffnull, 2019-11-26:
@boelle @Bob-the-Kuhn Hello again. Im back from very long vacation. Sorry for ignoring. Im download latest branch and open it in platformio. How to compile it? Please give me simple how-to. I whant use board with delta printer. Thanx

boelle, 2019-11-26:
google search, there are many guides and videos or ask here

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

ffnull, 2019-11-26:
@boelle @Bob-the-Kuhn hah. sorry.Maybe I did not quite correctly put it. I can build this version of firmware for different motherboards. But with this motherboard, to which I started porting the firmware, I ran into a problem. I do not know what needs to be added to the Platformio.ini . They are needed to compile for this board. Or do I need to compile in arduino an ide with a plug-in library?

boelle, 2019-11-26:
please stop adding me, i'm not a code guy, i'm just the custodian of the issue list

ffnull, 2019-11-26:
@boelle ok. just reopen issue. thnx

ffnull, 2019-11-26:
Ok. im compile it great in arduino ide with stm32 arduino lybrary. But when im uncomment l6470 im got crash. im used latest lybrary for l6470. @Bob-the-Kuhn can you contact with me directly? Telegramm @ffnull

boelle, 2019-11-26:
@thinkyhead closed it as your board are not supported

so lets see what he have to say befor we do anything

boelle, 2019-11-26:
[image]

[72 more replies]

commit 52108192 (2019-01-20): Prevent L6470 class name conflict with drivers.h define

Fix #12824