MarlinFirmware/Marlin #12825: Delta calibration w/BLTOUCH: wierd jerky Z-probes after a first clean iteration

gtrevi, 2019-01-05:
ISSUE:
After setting up an original BLTOUCH in favor of a bulky clone I was using in the meanwhile, I get a very wired behavior that was not happening with the bulky "3D Touch".

After a first "clean" probing iteration, the subsequent probings are done by double-jerking the Z height before performing the probe, and a resulting Z-Height that is 10mm less than the actual.
This behavior is consistent and repeatable at each printer reset.

If the printer is not reset, then a subsequent 2nd autocalibration will perform the wired Z-move probing as the ones right from the 1st iteration.
I uploaded a video capturing the 1st & 2nd iteration after a printer reset: [currently uploading]

SETUP:
I have a delta (Anybubic Kossel Linear plus w/Trigorilla) with Marlin v1.1.9 and the same configuration (did not do any firmware/cfg update in the between of the probe swapping).

What could be the problem?
In the meanwhile, I re-swapped the bulky “3D Touch” and all is now working perfectly again.

Many thanks for any help.

boelle, 2019-01-06:
that looks normal, it probes the same spot twice. If i remember correct that is the default

i dont have a delta but i probe the same spot 4 times and marlin figures an avg value, i do it this way to even out tolerance issues

maybe it would help if you attached your 2 config files as a zip file

gtrevi, 2019-01-06:
Yes, the double-probing is indeed normal (and as configured in the MULTIPLE_PROBING pre-processor def)... it's the double-jerk (happening between the probes, and not in the initial first iteration) that is absolutely *not* normal… otherwise why the first iteration behaves differently? 
Please see carefully the video to notice the different behavior between the two iterations.
Also, this was not happening some time ago when I was using a bulky "3D touch" clone.

I attached my two configs if they could help shading some light on this phenome.

Thanks!
[Marlin 1.1.9 configs.zip](https://github.com/MarlinFirmware/Marlin/files/2730776/Marlin.1.1.9.configs.zip)

boelle, 2019-02-21:
@gtrevi problem fixed?

gtrevi, 2019-02-22:
> @gtrevi problem fixed?

Unfortunately not: I kept experiencing the problem despite several tweaks... now on a Duet2 WiFi, which solved all at once.

closed 2019-02-22 by gtrevi

boelle, 2019-02-22:
tried bugfix 2.0?

gtrevi, 2019-02-22:
> tried bugfix 2.0?

Yes, was using 2.0, fully compiled w/VS Code