MarlinFirmware/Marlin #12826: odd flashing lines

boelle, 2019-01-05:
so i got the re-arm setup complete and doing my first print with it (frame to hold it to printer)

i noticed that the lcd has some very strange flashinh lines on the display that was not there before

a 10 sec video of it: https://www.youtube.com/watch?v=uIjn_hgOQ9A

config files attached, the only odd change i had to do was adding a extra part cooling fan in the pins file so that one is included. i had to do that to be able to control an extra big fan i have that cools of the bed after print, without this change that fan would be on all the time
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2729554/Configuration.zip)

boelle, 2019-01-05:
side Q: is my config correct in regards that eeprom is saved to a file or where it is saved as the re-arm should not have an eeprom, the commands work, i'm just unsure as i have not seen any files on the sd card other than the firmware

gloomyandy, 2019-01-05:
On the Re-Arm and other LPC176x based boards eeprom is emulated (by default)  using flash memory pages. 

Do you still get the flashing lines if you do not have the extra fan defined?

boelle, 2019-01-05:
will have to test that, but its kind of painfull to live without the fan as the wait time after print gets longer before the PEI release its hold on the print,

2 hours left of print before i can experiment :-/

p3p, 2019-01-05:
It's unavoidable with an SD print from the controllers sd card, the displays do not adhere to the spi standard and listen to things they shouldn't on the shared bus (how the board is designed), If you are using the slot on the board (or a serial host) it shouldn't happen.

boelle, 2019-01-05:
btw... i closed the "issue" as there is nothing that can be done about it and i dont want to take up more issue footprint than need be

but i want to upgrade my 2 month old firmware and a new feature to me at least is junction deviation, are there any guides on how to use that?