MarlinFirmware/Marlin #12827: [2.0.x] Cap:BUILD_PERCENT wrong report

GMagician, 2019-01-05:
am I wrong or returned value must be:

```c++
// BUILD_PERCENT (M73)
cap_line(PSTR("BUILD_PERCENT")
#if ENABLED(LCD_SET_PROGRESS_MANUALLY) && (ENABLED(EXTENSIBLE_UI) || ENABLED(ULTRA_LCD))
  , true
#endif
);
```
and not:
```c++
// BUILD_PERCENT (M73)
cap_line(PSTR("BUILD_PERCENT")
#if ENABLED(LCD_SET_PROGRESS_MANUALLY)
  , true
#endif
);
```

since M73 is only defined with 1st statement

AnHardt, 2019-01-05:
Depedns on how `LCD_SET_PROGRESS_MANUALLY` is restricted. If that can't be set if there is nothing to display on, it does not need further checks for the same.

GMagician, 2019-01-05:
Not clear what you wrote due to my English knowledge but
`LCD_SET_PROGRESS_MANUALLY` is user defined and since 'Cap' state:
`// BUILD_PERCENT (M73)`
and M73 is not included when
`(ENABLED(EXTENSIBLE_UI) || ENABLED(ULTRA_LCD))`
is not true

why do Cap should return it support build percent via M73?

AnHardt, 2019-01-05:
Could also be my english.

Clear is:
If there is no M73 we should not inform the host to send updates via M73.
We should not have M73 if we don't want it ( //#define LCD_SET_PROGRESS_MANUALLY).
We also should not have M73 if we don't have a place to display the information we get with M73.

For that the right place to check if we have the ability to display the information is a sanity check for `LCD_SET_PROGRESS_MANUALLY`.
If that sanity check is passed we can later trust in that and omit all furter conditions when telling the host our capabilities.

So if we currently tell the host to send M73 but we can't display it, it's an error in the sanity check of 'LCD_SET_PROGRESS_MANUALLY'. Adding more conditions to 'capabilities' is just hiding an earlier error.

GMagician, 2019-01-05:
Ok that's clear.
So the 'not needed test code' is only in gcode.h, gcode.cpp where test is `ENABLED(LCD_SET_PROGRESS_MANUALLY) && (ENABLED(EXTENSIBLE_UI) || ENABLED(ULTRA_LCD))` and not simply `ENABLED(LCD_SET_PROGRESS_MANUALLY)`.

closed 2019-01-05 by GMagician