MarlinFirmware/Marlin #12830: unintentional retraction instead of priming

Lyr3x, 2019-01-05:
I have an issue with two printers when printing from the SD Card. Never happend with the third printer which is always printing via Octoprint. All three printers using TMC2130
Printer 1: CoreXY with E3D V6, Titan and EINSY 1.1
Printer 2: Ender 3 with E3D V6, Bondtech, MKS GEN 1.4
Printer 3: I3 style with E3D V6, Bondtech, MKS Gen 1.4
All printers running bugfix-2.0 (from a few weeks ago and yesterday)
Gcode is always sliced with Slic3r-PE

The problem: Sometimes (more about this later) the printer retracts the filament instead of priming the nozzle via the Start Gcode. Sometimes means that: Running the same gcode file sometimes the extruder is running backwards until priming is done and sometimes everything its working as intended.
Here is the Start Code:
```
M104 S[first_layer_temperature]
M140 S[first_layer_bed_temperature]
M109 S[first_layer_temperature]
M190 S[first_layer_bed_temperature]
G28
G29 ; auto bed level
G1 Y30.0 F1000.0 ; position y for purge line
G1 Z0.3 F1000.0 ; position z for purge line
G1 X10.0 F1000.0 ; position x for purge line
G1 Y150 E15 F500.0 ; start purge line
G92 E0 ; zero extruded length
```

GMagician, 2019-01-05:
Maybe it's better to execute G92 E0 even before moving extruder. You don't know its position and I'm not sure G28 zeroes it

Lyr3x, 2019-01-05:
I'll give it a try, but that would mean that octoprint is doing something about setting the extruder position. Using this startcode for 2 years now with the octoprint printer.

boelle, 2019-01-06:
My start code: 

```
G28 Z; Home Z
G28 Y; Home Y
G28 X; Home X
G29
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G92 E0 ; zero the extruded length
G1 X50 Y0 F3000 ; go to the middle of the front
G1 Z0.1 ; close to the bed
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G1 X100 E10 F200 ; make a thick line to prime extruder
G92 E0 ; zero the extruded length
G1 Z0.3 ; lift Z
G4 S2   ;wait 2 sec to bleed off excess pressure
```

explanation on why i home each axis in order: so that filement does not get in the way when i home y and x, i have a spool holder above the printer and if i home x first then its goes to make a mess when i home z

i bed level cold so that i get the same results every time, i use the pinda probe and those are sensitive to heat.

i then heat up my bed first, i then zero my extruder
when bed is at temp i move to start of prime line, but i go closer to the bed so i dont have ooze
i then start heating the nozzle and then its at temp i make a 50mm long and fat line
i zero the extruder again and lift the nozzle and wait 2 sec to get rid of excess pressure, if the excess pressure is not removed it will drag a string where print starts

hope that can give you some inspiration

PS. i print from sd card, i compared the same gcode printet from octoprint and from sd card and i got better results from sd card

Lyr3x, 2019-01-06:
The "problem" seems to be fixed with the G92 E0 moved up. I changed the three seperate moves to a one liner. 
It seems that octoprint is doing something about that. 
Thank you guys

closed 2019-01-06 by Lyr3x

boelle, 2019-01-06:
yet another reason to print from SD, nothing is being messed with without you knowing it :-D

Lyr3x, 2019-01-06:
But i love to control my printers remote :)

boelle, 2019-01-06:
you can still control it remotely, you just have your print job on sd card

i use octoprint too, but i have changed settings so it only display files on sd card and i can start print fron sd on octoprint or on printer if i want to see it start the first layer

the main thing is that the print  job is saved to sd card and not on to octoprint

you can try and compare if it matters to you

Lyr3x, 2019-01-06:
Interesting, i'll look into that. Thank you for the hint 👍