MarlinFirmware/Marlin #12831: [Bugfix 2.0 + Re-Arm] cant control 2nd extruder fan via Gcode

boelle, 2019-01-05:
so i have added a 2nd fan in the pins file like this:

#if ENABLED(IS_RAMPS_EFB) || ENABLED(IS_RAMPS_EFF)   // Hotend, Fan, Bed or Hotend, Fan, Fan
    #define FAN_PIN        RAMPS_D9_PIN
	  #define FAN1_PIN       P1_20   // (11)

i can control it from the LCD but when i send Gcode M106 P1 to turn it full on nothing happens

it did thou on my copy from 5/11-2018

what could be wrong this time?

config and pins file attached from todays copy of marlin

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2729952/Configuration.zip)

GMagician, 2019-01-05:
Maybe something related to #12820?

boelle, 2019-01-05:
Not sure, i think @thinkyhead is the better judge of that

but it demonstrates that its always a good idea to keep the last 2 working copies :-D

boelle, 2019-01-09:
can this be considered a bug? if so i would reword the title by adding [Bug] to it, if its not a bug i will close the issue so i dont take up more space than i need

GMagician, 2019-01-09:
Have you tryed version before #12820? (change is easy to revert).
Since that function can change used fan may give you an issue...
You may check in your code what's FAN_COUNT value. with 2 fans it should be 2

GMagician, 2019-01-09:
Have you checked what's the FAN_COUNT value?

SERVO0_PIN is same. Maybe fan value is overwritten? Try to change SERVO0_PIN

InsanityAutomation, 2019-01-09:
Does it work with S255 in the command? I'm not 100% sure full on without setting a speed was ever intended behavior.

boelle, 2019-01-11:
i tried to use google translate to translate from spanish but it made no sense

boelle, 2019-01-11:
think i got it...

you have the mks sbase and can control the extra fan from the lcd, so can i

but i can not control it via gcode

only thing i cant see in your pin example is where your extra fan is defined, that is what i did, i defined an extra part cooling fan that can be controlled seperately from the standard one

boelle, 2019-01-14:
i want @thinkyhead to answer this one:

Is it not possible to add the extra part cooling fan to the RAMPS_EFB setup like i did and still be able to control it? or do i need to add something extra?

closed 2019-01-22 by boelle