MarlinFirmware/Marlin #12832: [Q] Sensorlesshoming with two endstops

Ludy87, 2019-01-05:
Is it possible to trigger two z limit switches by sensorlesshoming?

thinkyhead, 2019-01-06:
Wire the endstops together…?

thinkyhead, 2019-01-06:
So, do you mean… When Z sensorless homing triggers, have both faux Z and Z2 endstops get set?

Ludy87, 2019-01-06:
If steps are lost, the Z-axis alignment will be wrong.

My idea is to calibrate two stops for Z and Z1 around the axis. When Z is triggered, Z1 is also stopped by default. Or am I wrong?

boelle, 2019-01-06:
@thinkyhead I think he means that he wants sensorless homing with dual z stepper drivers and dual z endstops

boelle, 2019-01-06:
@Ludy87 i have dual z stepper drivers and dual z endstops on  mendel90, the Mendel90 is a bit different in that it homes to max (up)

when i home the 2 z steppers are sync'ed autoamtic, i imagine that goes sensor less is no problem

it might be a problem on systems where the load on the steppers gets less when homing, in my case the load on the steppers will get higher

Ludy87, 2019-01-06:
@boelle 
I also use mechanical endstops at the moment. I have two Z motors with TMC2130 drivers, sensorlesshoming and I would like to use that.

boelle, 2019-01-06:
just as i figured, i dont know how to do this, but maybe  just connect stepper driver Z endstop output to Zmin or Zmax.. then connect output from Z2 stepper driver to the correct pin i think default is Xmax

but i'm sure @thinkyhead will pop in and confirm

kAdonis, 2019-01-06:
@Ludy87  I assume your printer uses leadscrews?
You need a lot of force to stop a leadscrew, did you try sensorless homing in z before?
You may not achieve a repeatable z- home position 
Be sure your Printer is rigid enough for this

boelle, 2019-01-06:
@kAdonis he wants to do dual z stepper driver and dual Z endstop, but with sensorless homing instead of normal endswitch

kAdonis, 2019-01-06:
Yes, I understand this
But I am concerned sensorless homing does not work at all with leadscrews

boelle, 2019-01-06:
depends if load on motor gets higher or lower when homing :-D

kAdonis, 2019-01-06:
depends on thread lead and pitch, friction, load on the Bed...

Ludy87, 2019-01-06:
@kAdonis 
> I assume your printer uses leadscrews?
Yes

> You need a lot of force to stop a leadscrew, did you try sensorless homing in z before?
yes, with one Endstop trigger from TMC worked.

boelle, 2019-01-06:
@kAdonis and if you home up or down :-D

boelle, 2019-01-06:
if homing towards the bed gives less load on the motor homing up might give more load so that the stepper can detect it and hence it should work

but since it it works with one stepper all he would have to do is to connect the other stepper driver to the endstop pins

kAdonis, 2019-01-06:
> yes, with one Endstop trigger from TMC worked

I never dared to try

>  but since it it works with one stepper all he would have to do is to connect the other stepper driver to the endstop pins

yes, I agree
Im curious, if its reliable
@Ludy87 please post your findings here

boelle, 2019-01-06:
me interested too as i plan on getting the tmc2130's maybe at end of this month or next month.
my printer is running 24V for the steppers allready so i would just have to make up the spi/cs wireharrness and connect them to the endstop pins, 

the bigger advantage is that if it works i could get rid of the inductive probe and just use the nozzle

and the even bigger benefit would be that the probing can be done with the nozzle hot and ready, with the pinda probe it has to be done cold as they are sensitive to heat

closed 2019-01-16 by Ludy87