MarlinFirmware/Marlin #12834: [1.1.9] #define LCD_LANGUAGE de (and others) ignore German umlauts in filenames on SD card

momo42, 2019-01-06:
Today I recognized that in Marlin 1.1.9 (stable) German umlauts are not printed in filenames.
When I put for example "ÄÖÜäöüßtestfile.gcode" on the sd card it is shown as "testfile.gcode"

Menu items like "Über den Drucker" are displayed correctly.

Then I played around with  "#define LCD_LANGUAGE de" in Configuration.h and found out that the test mode setting (#define LCD_LANGUAGE test) leads to correct filenames with umlauts:

I also tried some other languages (also en) but only "test", a weird mix of english and test menu items seems to be able to display umlauts. Very strange behavior ...
People living in the country called "test" can be lucky. ;-D

Could someone please look into this issue? :-)

momo42, 2019-01-06:
I testet the latest 1.1.x bugfix snapshot and this bug persists.

gloomyandy, 2019-01-06:
If you can it would really help if you tested issues like this with the latest bugfix-2.0x build. It looks like very few updates are now being made to the 1.1.x trees.

AnHardt, 2019-01-06:
This is unlikely to be solved until all operating systems use a common character encoding system.
Marlin tries to use utf8 as much as possible. 
But filenames in directories (especially if not created by Marlin) tend to be written in in one of several dozen 'codepages' (850, 832, ...).
We don't have space and patience to make dozens of  tables to convert back and forth.

boelle, 2019-01-06:
i agree that this will be a hard one to crack

every pc and every slicer have its way of saving files, if there was a common standard it would be far more easy

igorepi, 2019-01-07:
This problem is even worse for non-latin-based languages. You lose umlauts with German but with Russian you lose entire file name. What is the cause ?
Marlin use 8-bit fonts which are different for every language but utf-8 encoding for menu items, so some sort of mapping is required. This is done by charset_mapper() function declared in utf_mapper.h, same mapping is applied not only to items but also to file names read from SD card.
Long file names in FAT/FAT32 filesystems are always stored in UTF-16 encoding. Marlin in SdBaseFile just discards high byte of file name encoding; no mapping is done and 1-byte-for-symbol encoding is presumed in many places of code. So I can't find an easy way to add UTF16 file name symbol mapping without rewriting tons of code. Instead in order to display russian filenames along with russian menu items I have to use 1-byte (Win1251) encoding for menu items instead of (multibyte) utf-8.
Code changes are relatively simple but sort of quick-and-dirty. I added two mapping functions to uft_mapper.h,  charset_mapper() for mapping Win1251-encoded menu items to font symbols and fn_utf16_mapper() to mapping UTF-16-encoded file names :

    #elif ENABLED(MAPPER_1251)

    char charset_mapper(const char c) {
    	uint8_t d = c;
    	if (d >= 0xC0) {
    		HARDWARE_CHAR_OUT((char) (d - 0x10) );
    	}
    	else if (d == 0xA8) {
    		HARDWARE_CHAR_OUT((char)0xA1);
    	}
    	else if (d == 0xB8) {
    		HARDWARE_CHAR_OUT((char)0xF1);
    	}
    	else {
    		HARDWARE_CHAR_OUT((char)c);
    	}
    	return 1;
    }

    #ifdef FILENAME_MAPPER
    char fn_utf16_mapper(uint16_t ucs_character) {
    	char d;

    	if (ucs_character <= 0x7F)
    	{
    		// Plain single-byte ASCII.
    		return (char)ucs_character;
    	}
    	else if (ucs_character == 0x0401)
    		return (char)0xA8;
    	else if (ucs_character == 0x0451)
    		return (char)0xB8;
    	else if (ucs_character >= 0x0410 && ucs_character <= 0x044F)
    	{
    		return (char)(ucs_character + 0xC0 - 0x0410);
    	}
    	return '?';
    }
    #endif

Changed SdBaseFile.cpp to store mapped file name:

    //for (uint8_t i = 0; i < FILENAME_LENGTH; i++)
    //  longFilename[n + i] = (i < 5) ? VFAT->name1[i] : (i < 11) ? VFAT->name2[i - 5] : VFAT->name3[i - 11];
    for (uint8_t i = 0; i < FILENAME_LENGTH; i++) {
       uint16_t utf16_ch = (i < 5) ? VFAT->name1[i] : (i < 11) ? VFAT->name2[i - 5] : VFAT->name3[i - 11];
    #ifdef FILENAME_MAPPER
       longFilename[n + i] = fn_utf16_mapper(utf16_ch);
    #else
       longFilename[n + i] = utf16_ch;
    #endif
    }

and made new Win1251-encoded language_ru1251.h language file

    #ifndef LANGUAGE_RU1251_H
    #define LANGUAGE_RU1251_H

    #define MAPPER_1251                // For Cyrillic
    #define FILENAME_MAPPER
    char fn_utf16_mapper(uint16_t ucs_character);

    #define DISPLAY_CHARSET_ISO10646_5
 
    #define WELCOME_MSG .... (the rest is same as in existing language_ru.h, only in Win1251 encoding)

With this changes I now have russian file names and russian menu items on 12864 display. Same approach can be used for any national language and it doesn't break anything in existing Marlin (1.1.9) code.

This code work with long filenames only, 8.3 names are unaffected. 

If Marlin community is OK wtih changes in SdBaseFile.cpp I can try to port my code to Marlin 2.0 and  make an PR which will add two Russian languages for Marlin with graphical displays - russian SD card file names support with Russian and English menu. I personally use this for about a year and 4 Marlin 1.1.x releases without problems.

boelle, 2019-02-21:
@momo42 still an issue when using bugfix 2.0?

closed 2019-03-03 by thinkyhead