MarlinFirmware/Marlin #12835: Change filament procedure leave hotend on.

AndKe, 2019-01-06:
### Description

Change filament leaves extruder heat on

### Steps to Reproduce

With hotend off (non-printing),cChange filament using display feature (ADVANCED_PAUSE_FEATURE)

**Expected behavior:** [What you expect to happen]
after change , I would expect the hotend state to be the same as on start.

**Actual behavior:** [What actually happens]
on change, the hotend is left in PLA/ABS preheat temperature.

#### Additional Information

During the change, PAUSE_PARK_NOZZLE_TIMEOUT makes the hotend heating be switched off now-and-then.  ..for safety.

After the change, hotend is left hot, apparently indefinitly.

Solution:
Upon pause, store hotend temperature, on procedure completion, restore the previous temperature, be it a printing temperature, or "off"

boelle, 2019-01-06:
i would put [FEATURE REQUEST] on the title/subject line before your title and maybe reword it

[FEATURE REQUEST] Restore temp on filament change

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
This is a duplicate of another issue I saw awhile ago that got closed. The consensus there was if youre changing filament you plan to do something shortly. There was discussion somewhere of a heater timeout though which may get you closer to what you want though not exactly. It hasnt been implemented yet.

dewhisna, 2022-08-07:
Does this issue apply to the advanced pause feature when using a smart filament runout sensor, like the BTT?  I hit a weird case yesterday during a print.  A heat soak issue caused a filament jam and my BTT filament sensor correctly sensed it and paused the print for me to "change filament" (as the filament runout script is a M600, filament change).

While trying to clear the jam, I exceeded my `PAUSE_PARK_NOZZLE_TIMEOUT` value of 45 seconds.  It said it was shutting down the heaters, only it never did (a safety bug perhaps?).  In this specific case, that was actually a good thing as it kept the nozzle hot long enough for me to get the jam fixed and manually push some filament through to make sure things was clear (though it would have been a very bad thing if I didn't just happen to be in the room when the jam occurred).

After clearing the jam, I hit the button to resume the print.  It then said it was reheating the heaters (which didn't need doing since it never turned them off).  That message cleared quickly, which I assumed was because it was already up to temp.  But then, as it resumed the print,  it did actually turn them off, which proceeded to jam the hotend all over again.

It then got stuck in some strange loop trying to get me to load the filament again, only this time with the heaters still off, which made loading it impossible.  And it ignored any request to abort the print, and I eventually had to just hit the reset button and start all over.

Not sure if what I found is at all related, but running out of filament (at least by the filament runout sensors) is treated much the same as a filament change in the advanced pause logic (as the runout script is M600), and I was wondering if it shutting down the heaters when it resumed, in my case, was related to this "Restore temp on filament change" issue, or if what I found was just an unrelated bug or configuration quirk?