MarlinFirmware/Marlin #12836: KILL missing verbose reason

AndKe, 2019-01-06:
-->

### Description
Most likely related to extended cables;  my Y axis started to fail to home...
The lack of verbose output made it harder to troubleshoot.

The output should contain a reason, and refer to one particular axis.  (the usual G28 will fail at an unknown place, then require troubleshooting)

### Steps to Reproduce
run 
G28 Y
on a system with some endswitch problem

#### Additional Information

```
Send: G28 Y
Recv:  T:17.97 /0.00 B:38.30 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error: Printer halted. kill() called!"
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"
Connection closed, closing down monitor
```

boelle, 2019-01-06:
have you tested the endstops to check if they are broken, 2 meter of wire should not matter as its a simple switch circuit

Send M119 and see status of endstop
press the endstop with your finger and send M119 again, status should change

AndKe, 2019-01-06:
Whole story posted here, like a monologue , since I solved it.
https://reprap.org/forum/read.php?415,843653,843667#msg-843667

@boelle  Yes, incredibly enough, the long flat cable was to blame, I separated the endstop wires from stepper, and twisted them, that fixed the issue instantly for both axes.

boelle, 2019-01-06:
heheh, i have all wires twisted, less noise in and out of the wires :-D

for the stepper drivers i twist them in pairs

boelle, 2019-01-06:
but then again i think this thread is still valid, ie a reason should be given so that the user has just an idea of what goes wrong

boelle, 2019-06-22:
missing label

closed 2019-07-11 by thinkyhead