MarlinFirmware/Marlin #12837: Cleanup issue list

boelle, 2019-01-06:
I regular go through the issue list to see if i can help

but some issues seems solved, others do not have anything to do with marlin

would it be possible for either @thinkyhead or @Roxy-3D  to go through them and close those that are solved and label the rest so that its quick to see those that are marlin related?

boelle, 2019-01-09:
@thinkyhead if you need help putting on labels i'm up for it, i might be a bit off from time to time but i think it can benefit you in that you faster can get to issues that is marlin related and that are bug's rather than questions

boelle, 2019-01-09:
so me looking through them and figuring out if the issue is a Q or not related to marlin at all is less of a benefit?

just trying to figure where i can be of help to the project with the limited coding skills i have

DerAndere1, 2019-01-10:
Strictly seperating questions about usage from feature requests or bugs would help marlin users like me to see more quickly, which critical bugs we have to be aware of. In addition, for a software with a broad user base like marlin, it might be beneficial to [make use of issue templates](https://help.github.com/articles/creating-issue-templates-for-your-repository/) for those three categories to improve issue descrition quality.

boelle, 2019-01-10:
Yes and i know octoprint github has a bot that checks if an issue is filled
out correct. Might help too

swilkens, 2019-01-11:
Some sort of copy / pastable reply referring users to either the official facebook group or their printer's subreddit might be useful, large amounts of use-questions showing up after the christmas gift flood.

boelle, 2019-01-11:
there is already a template in place... this part of it looks very similar to what @swilkens just wrote

> Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.

complete template text:

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

closed 2019-01-14 by boelle