MarlinFirmware/Marlin #12838: [Marlin 1.1.9] auto leveling issue; downgrading to 1.1.0 solve the problem

Seraj22, 2019-01-06:
Hello guys,
 So recently I've bought a BL-Touch for my Tevo Black widow (with the board mks gen 1.4)
Mount it and made all the connections as instructed, the BL-touch appears to function well in terms of  standard deviation but the compensation is just wrong. the z compensation is definitely enabled (I can see the Z moving within the layer), The start gcode is also correct (G28 followed by G29).

In the beginning I though that the compensation plane was flipped but I was wrong, I tested by marking the probing points on the bed and tilt the bed to see how it will behave. what I think is that compensation is in the right direction but is just exaggerated.

after days of searching on the internet and triple checking the hardware (no I don't have an axis twist)/firmware configuration with no result, after using the BugFix version I started to downgrade marlin to 1.1.8 which also didn't solve the problem. after loosing hope I made a downgrade jump to 1.1.0 RCBugFix (since this is what the factory use) and BOOM it works great and the level is perfect all over the space

Notes:
-I'm using the bilinear autoleveling mode
-The configuration of all the version (Configuration.h and Configuration_adv.h) are made same
-The printer works well when not using autoleveling with all marlin versions
-I've attached my config files

Please help since downgrading is not a solution.

Thank you in advance/please excuse my English.
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/2730557/Configuration.files.zip)

vilsed, 2019-01-08:
I'm having the exact same issue, also on MKS gen 1.4. The G29 T gives me a somewhat sensible grid, but during print, in Y axis, it over compensates on one side, and under compensates on the other. Will try 1.1.0 RCBugFix branch today.

vilsed, 2019-01-09:
I've solved my issue by using UBL and a manually edited mesh. Will have to make separate meshes for different temperatures (610x610 bed, quite a lot of thermal movement), but for now it'll do. Something is probably wrong mechanically on my machine, but since it's a job printer, I could not be bothered to investigate.

xadox-1st, 2019-02-06:
> I'm having the exact same issue, also on MKS gen 1.4. The G29 T gives me a somewhat sensible grid, but during print, in Y axis, it over compensates on one side, and under compensates on the other. 

Same for me on MKS GEN 1.4 and BLtoch using UBL.
The grid is very accurate, but the compensation doesn't seem to work.

The bed is higher on the left than on the right, but the compensation is much too high on the left and much too low on the right.

This cannot be compensated by the Z-Offset either.

boelle, 2019-02-21:
@Seraj22 tried bugfix 2.0?

closed 2019-03-03 by thinkyhead