MarlinFirmware/Marlin #12839: Manual leveling with servo. (The bed will be moved)

Milu1980, 2019-01-06:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Milu1980, 2019-01-06:
Hi,

I need help for my projekt... i need somebody to write the script to the Marlin.
I want to have a 4 points bed leveling with Bltouch sensor and 3 servos.
1 srew is fix and the another 3 will be leveled with the touch sensor.
3 scresw have 3 servos with gear.( gear on the nut under bed, and gear on servomotor, all 6 gears)
The touch sensor do a first leveling on the fix screw side with the Z motor, and going to the another 3 screw and do it the same, but with the servos.
This methode will be do 2 times.
Thx.

Roxy-3D, 2019-01-07:
The easiest way to get features added to Marlin is to get interest in the idea and then make sure somebody that cares about the ideas has hardware.

I would like to see Marlin have full (and exotic) bed tilting under firmware control.   And most likely, there are a lot of other developers that would like to see that happen.   But the developers need hardware to code against.   

If you were representing a company, it might be easier.   You could supply a couple of units to interested developers and the code would magically appear.

boelle, 2019-02-21:
@Milu1980 problem fixed?

closed 2019-03-03 by thinkyhead