MarlinFirmware/Marlin #12840: [FR] LEVEL_BED_CORNERS with mutable Z height

swilkens, 2019-01-06:
### Description
When using manual leveling, a user might use LEVEL_BED_CORNERS to assist in automated moving to the corners of the bed. Currently, this method requires homing all axis before it can be done. It then jumps to all 4 corners of the bed while z-hopping to 4 mm above the bed for safety.

> static int8_t bed_corner;
> void _lcd_goto_next_corner() {
>   line_to_z(4.0);
> ...

Where I have understood that `line_to_z(foo);` is a raw move that doesn't update `current_position[Z_AXIS]`. I would like a possibility to set a certain offset from Z0 where this leveling is done. I do not want to configure the security z-hope value, I want to be able to modify the distance from the homed Z0 position in Z space where the bed is leveled. Let me explain why.

When leveling with a tool that has, for example, a 0.1 mm thickness - we make a 0.1 mm error in the leveling process. The printer (having just homed) believes is it at the Z0 position during leveling. A user manually leveling however will now level the bed with the nozzle at a what is effectively a Z0.1 position due to the thickness of the tool used. Thus a 0.1 mm error is made, as the printer still believes itself to be at Z0. The first layer would be printed at 0.1 mm + layer height, not the intended layer height.

I would propose to change this method by allowing a user to configure the firmware with the thickness of his / her leveling tool. The leveling method would then be:

1. G28
2. Go to the first corner **at a Z position of the thickness of the tool used**
3. Run through all corners on this Z plane
4. Finish

In this way, Z0 will actually be at the position where the nozzle is just touching the bed. Not at a position where the nozzle is floating 0.1 mm above the bed, where it was just touching the leveling tool.

### Proposed change

1. A configuration value `LEVEL_TOOL_THICKNESS`
2. A modification to the code in `Marlin\src\lcd\menu\menu_bed_corners.cpp` in the form of `current_position[Z_AXIS] = LEVEL_TOOL_THICKNESS;`

### Further evaluation

I believe this error in leveling is made nearly everywhere. While automated leveling systems usually have features in place to modify this (z offset, tool definitions or such), such features are not available when leveling manually.

It might be useful to bring a feature like this to all manual leveling workflows.

AnHardt, 2019-01-07:
As far i can see this is equivalent to a Z-probe-offset for manual probing.

swilkens, 2019-01-07:
@AnHardt thank you for thinking with me. Can you clarify how that workflow would be? I have been unable to find a feature that allows me to do this in combination with `LEVEL_BED_CORNERS`.

Features like `M851` for example are not available on my firmware, as it is only defined when I enabled a bed mesh leveling system of some kind. Which I have not, due to flash space restraints and the fact that my bed is perfectly level.

AnHardt, 2019-01-07:
An alternate workflow for you:
Use 1. to 3. as before
`G0 Z0`; to move the tool to the height of the levelling tool.
`G92 Z0.1`; to tell the system it's now at the height of the leveling tool.
Done.
For verification you could now send a `G0 Z0` again. The tool should move down 0.1mm now.

swilkens, 2019-01-07:
@AnHardt does that work with the endstops? As the nozzle would them have to move 0.1 mm down into the endstop?

To clarify, I have been doing:

1. G28
2. G1 Z0.1
3. G1 Xfoo Ybar (level corner 1-4)
4. Done

I have been doing that since this method does not work:

1. G28
2. G1 Xfoo Ybar
3. Set Z-offset to -0.1 (firmware refuses to do this, I presume endstops prevent this?)

Do I understand correctly that there is currently no configurable "level tool thickness" variable that is exposed to the user to automatically do the `G1 Z0.1` step in my first workflow? Is there some provision to define a "level tool" with a manual bed leveling approach that allows for automated compensation of the tool thickness?

If no do you see it as valuable to add such a feature? As I see it, everybody leveling with either a piece of paper or a thickness gauge is making an error now?

Excuse the long-winded posts, but I feel this is something so obvious that there surely must already be a method to do this in-firmware - but I just don't know it?

Hochmah, 2019-02-04:
I agree with that! 

It's important to offset the probe tool thickness from z axis without deactivate software endstops. 

Since I'm using a touch plate my bed level is below the touch plate, I want to be able to set my probe tool height in a way that after a G28 my z position is equal to that probe tool thickness and that without loose the safety of software endstops.

Previously, I used to call a G92 Zn (where n is my touch plate height), but now I can no longer use G92 to move "below the bed".

boelle, 2019-06-22:
missing label

thinkyhead, 2019-07-11:
Level tool thickness would not be relevant to Level Bed Corners, which is only intended to get your bed relatively flat. For more involved leveling of the manual kind, the manual procedure provided by Unified Bed Leveling is recommended, since it does provide for your shim / feeler gauge / "business card" thickness.

swilkens, 2019-07-11:
UBL typicy doesn't fit on the cheaper boards though, hence this suggestion

Zuckme, 2021-02-04:
Thanks for reporting this, until today I was using `LEVEL_CORNERS_HEIGHT `as the proposed `LEVEL_TOOL_THICKNESS` solution.
Silly me. 
Now I understand this LEVEL_CORNERS_HEIGHT is just the Z level during the corner level procedure.
I also observed that after G28, Z is the opposite of `LEVEL_CORNERS_HEIGHT`. Example `LEVEL_CORNERS_HEIGHT `is 0,1, after G28 the position is X0 Y0 Z-0.1. I am blessed to have a board which can support the UBL and I am doing it manually.
It is a pain in the ass but it works somehow.

A stop gap solution is to adjust the Z screw at the Z endstop, my Ender 5 has it:

[image]

So you could dial in the first layer thickness there, regarless the tool you used.

Related to that, I can't find a solution to keep my nozzle in the air after G28, let's say 10mm
See related discussion [here](https://reprap.org/forum/read.php?415,841049).

steve-zero, 2022-01-08:
Is there any progress on this?
I have a fairly simple 3d-printer (Anycubic Mega Zero 2.0) and tried to do manual bed leveling as good as possible.
But when using a feeler gauge, the assisted leveling gives me a bad leveling. The nozzle is simply 0.1mm too high, which is exactly the thickness of my feeler gauge. This ruins the prints. Instead, I hoped that a feeler gauge would give me more precision and thus better prints.
Instead, when using a sheet of paper for leveling, I have to ensure, that the bed is hit so hard by the nozzle that the paper is almost not be moveable. This lifts/bends the hotend upwards by exactly the paper width in order to get an accurate leveling. This is not good for the printer and not precise. But it is, what all users of non-expensive marlin printers have to do right now.

To make a long story short: Is there any aspiration to offer a _full-manual_ leveling assistant that is actually usable?
Currently the menu of my printer pretends to offer such functionality but does not work as expected. And generations of users struggle with this. The internet is full of leveling problem reports due to the lack of this feature.
Or is the project at a point, where only high-end hardware and automatic leveling are supported any more?
Any insights are appreciated.

swilkens, 2022-01-08:
@steve-zero `LEVEL_CORNERS_HEIGHT` was introduced to the firmware, which sets the Z level at the leveling points. It has been a while, but to my understanding, that should provide a solution here. 

If you set `LEVEL_CORNERS_HEIGHT` to the thickness of your gauge, you should find that a move to `Z 0` after leveling is actually with the nozzle against your bed.

Relevant PR: https://github.com/MarlinFirmware/Marlin/pull/13439

@thinkyhead @drzejkopf Unfortunately I don't use this firmware anymore, so I can't verify if the implementation in #13439 was indeed done in such a way that it compensates for the thickness of the tool. Therefore, I can't judge if I should close this.

steve-zero, 2022-01-08:
@swilkens Thank you for your quick and helpful reply. Sorry for letting my frustration shine through.
LEVEL_CORNERS_HEIGHT should indeed do the trick. Looks like I have to build my own firmware...
But from a users perspective, this is not an ideal solution. LEVEL_CORNERS_HEIGHT is accessible on source code level only. There is even no gcode for changing it. So it is purely for customized builds of the firmware and hence not a solution for the masses.
Is there any possibility to make this parameter user definable via gcode or the GUI in future Marlin releases?
That would actually solve some leveling problems for many users out there.

PS.
What about the statement from @Zuckme above?
"I also observed that after G28, Z is the opposite of LEVEL_CORNERS_HEIGHT. Example LEVEL_CORNERS_HEIGHT is 0,1, after G28 the position is X0 Y0 Z-0.1."
So changing the define may cause problems? Or is this already fixed?

thisiskeithb, 2022-01-08:
Added in #13439

closed 2022-01-08 by thisiskeithb

steve-zero, 2022-01-08:
"Closed"
So I need to raise a new feature request?

swilkens, 2022-01-08:
@thisiskeithb thanks 👍 

@steve-zero 
Making things configurable **after** compilation always requires more code, and therefore more flash space. Which is always difficult, especially for people that may use `LEVEL_BED_CORNERS` (probably small chips). The devs always needs to consider if the extra code required to make something configurable is worth it.

Personally, I imagine a person using this feature has a single leveling tool. Maybe a feeler gauge or other thin metal sheet. And that's it, that's what he / she uses. So the typical user would set `LEVEL_CORNERS_HEIGHT` once and never change it again.

Having said that, you could open a Feature Request. For example, an optional flag could be added to G28 that configures the tool thickness. Maybe something like `G28 T0.1` to auto-home with a tool of 0.1 mm thickness.