MarlinFirmware/Marlin #12841: Z-Min Endstop problem with workaround - Marlin 1.1

RC-Fly, 2019-01-06:
Hallo,

Subject:
I'm using Marlin for a simple 3-Axis milling application with min, max endstops 
for all axis. 5 of 6 endstops are working well but Z-min required a very special
workaround - modification of file endstopp.cpp - to become functional. 

General:
Hardware: Mega with Ramps 1.4 - Z-min Switch at the standard pins
Marlin Version 1.1

BEFORE workaround Behaviour:
With M119 the Z-min endstop behaviour shows that there is no hardware issue, but the
stepper motion do not stops if I trigger the Z-min switch and for all
configurations I tested it failted.

Workaround:
File endstops.cpp

  /**
   * Check and update endstops
   */
Line 492 subseq.

  #if HAS_Z_MIN
    #if ENABLED(Z_DUAL_ENDSTOPS)
      UPDATE_ENDSTOP_BIT(Z, MIN);
      #if HAS_Z2_MIN
        UPDATE_ENDSTOP_BIT(Z2, MIN);
      #else
        COPY_LIVE_STATE(Z_MIN, Z2_MIN);
      #endif
//    #elif ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
//      UPDATE_ENDSTOP_BIT(Z, MIN);
//    #elif Z_HOME_DIR < 0
    #else     //new
      UPDATE_ENDSTOP_BIT(Z, MIN);
    #endif
  #endif
 
AFTER workaround Behaviour:
The stepper motion stops if I trigger the Z-min switch but I don´t understand why.
The first #elif has to be worked, because my Z_HOME_DIR +1 make the secound #elif inactive. 

Theory:
I also tested Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN several times with several configurations
of all this probe special options, but now I stop to try with guess.

Question:
Why Z-min remains in the code as Endstop? For me it´s looks like that the Z-MIN-PROBE includes
a Z-min Endstop function. 
Why I didn't see Z-MIN-PROBE with M119 and what happend if both are defined?

Way forward:
This CNC Milling application is excotic for Marlin - please decide to investigate it, if
you like. If I can help, please be patient, I have started with Arduino and GitHub two mounth ago.

Best Regards  
 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2730843/Configuration.zip)

GMagician, 2019-01-06:
Better to test with bugfix 2.0.x to see if this issue is still present and

> Why I didn't see Z-MIN-PROBE with M119

because it's only visible when you have installed a probe (AFAIK)

> and what happend if both are defined?

don't know if probe take priority or some issue will appear

RC-Fly, 2019-01-07:
Quick Answer!

but I need help to understand - It is a bug or a feature?

I´m new and will do my best, but I have a switch - is this an "AFAIK"?

RC-Fly, 2019-01-07:
Subject: It is probably a BUG!

If I analyse my workaround than there are no problems with "Z_HOME_DIR -1" expected and this is the normal configuration in the examples.

I have "Z_HOME_DIR 1"

I searched in all example configuration "Z_HOME_DIR 1" and this is also for other printers exotic and I didn't find this #define combined with a Z-MIN Endstop. 

In bugfix 2.0 there  is no new endstops.cpp and in version 1.0 the endstops are not in a separate file.

For me it's looks like a rare configuration not covered by testing. 

Please request the programmer of endstops.cpp - for me it's looks like that:
"Z_HOME_DIR 1" and "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN" disabled is a valid and seldom configuration actually not covered and without side-effects it is possible to insert a additional "else" for it.

GMagician, 2019-01-13:
it's not up to me

boelle, 2019-02-21:
@RC-Fly tried bugfix 2.0?

closed 2019-03-03 by thinkyhead