MarlinFirmware/Marlin #12842: Extruder dosn't stop

dudman, 2019-01-06:
### Description
The extrude once started just dosn't stop, this has just started since I uploaded a new firmware to change the PID's of the heated bed. Before hand I've been printing fine. Have double checked the changes to code, and can't find anything different, let alone think what config changes might have changed this. I've tried plugging in a different stepper motor, and the same behavior is there.

version 010100

[semd.zip](https://github.com/MarlinFirmware/Marlin/files/2730979/semd.zip)

thinkyhead, 2019-01-07:
Please test the `bugfix-2.0.x` branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2019-02-21:
@dudman problem fixed?

closed 2019-03-03 by thinkyhead