MarlinFirmware/Marlin #12843: MMU 2.0 with a different board adaptation

xddarko, 2019-01-07:
Hi everybody,
I am willing to integrate an MMU 2.0 kit into my printer soon and would like to ask if it's possible to work with a different board than the Einsy used by Prusa Research. I am using MKS Gen 1.4 and wondering if am gonna bump into problems connecting the MMU board with mine. If so, how can I adapt marlin to work? Obviously am gonna re-define data pins but what else can you tell me?
Thanks.

rmoravcik, 2019-01-07:
@xddarko Try have a look: here https://github.com/MarlinFirmware/Marlin/issues/11859. I think some people got it working also on non Einsy boards.

xddarko, 2019-01-07:
@rmoravcik I thought that too but nobody explained thoroughly how to do so in the firmware. Communication appears to be the problem.

revilor, 2019-01-08:
Regarding firmware you can have a look at my [github](https://github.com/revilor/Marlin/tree/MMU2) There is not much documentation at the moment but the Marlin/MMU2 combination is woking fine with  RAMPS. As long as you have a free serial port on your MKS Gen board it should work too.

And please join the discussion in the already mentioned #11859 so we have all the MMU2-related stuff in one place.

boelle, 2019-02-21:
@xddarko still relavant to have this issue open?

closed 2019-03-03 by thinkyhead