MarlinFirmware/Marlin #12844: Error while compiling marlin.ino on arduino

The3rdDimension, 2019-01-07:
**_i am trying to upgrade my creality cr-10s firmware to the new marlin 1.1.9 and while compiling the file on arduino after making the necessary changes i get an error_**

In file included from sketch\MarlinConfig.h:27:0,

                 from sketch\power_loss_recovery.cpp:27:

macros.h:164:57: error: operator '&&' has no right operand

 #define PIN_EXISTS(PN) (defined(PN ##_PIN) && PN ##_PIN >= 0)

                                                         ^

sketch\power_loss_recovery.cpp:209:11: note: in expansion of macro 'PIN_EXISTS'

       #if PIN_EXISTS(POWER_LOSS)

           ^

macros.h:164:57: error: operator '&&' has no right operand

 #define PIN_EXISTS(PN) (defined(PN ##_PIN) && PN ##_PIN >= 0)

                                                         ^

sketch\power_loss_recovery.cpp:278:9: note: in expansion of macro 'PIN_EXISTS'

     #if PIN_EXISTS(POWER_LOSS)

         ^

exit status 1
operator '&&' has no right operand

**Apologies if i havent posted properly as this is my first time**

swilkens, 2019-01-07:
It seems some of your modifications have broken your copy of the firmware. Can you please zip your `configuration.h` and `configuration_adv.h` files and post them here?

boelle, 2019-02-21:
@The3rdDimension problem fixed?

closed 2019-03-03 by thinkyhead