MarlinFirmware/Marlin #12851: [Bugfix 2.0.x] Bug in Planner - JunctionDeviation calculates wrong cos(theta) - Fix included

HackingGulliver, 2019-01-07:
Hi everybody,
I found a bug in the junction deviation code in `planner.cpp` that may cause layer shifts / lost steps by allowing too high `vmax_junction_sqr`.

On my CoreXY machine (HyperCube Evolution) I like high accelerations and speeds for travel moves. Linear accelerations (i.e. without corners) of 4000mm/s² are no problem. However, with junction deviation turned on and angles between two movement vectors of more than 120° the print head slammed into corners and lost steps at high speeds.

I analysed the junction deviation code and found out that `junction_cos_theta` exceeds the allowed range of [-1, 1]. Digging further I realized that the vectors `unit_vec` and `previous_unit_vec` were not normalized as the name implies. And this is crucial for calculating the angle between them by their scalar product.

**Fix**
Add
`normalize_junction_vector(unit_vec);`
at line 2395 in planner.cpp.

With a bit more context:
```
   #else
      float unit_vec[] = {
        delta_mm[X_AXIS] * inverse_millimeters,
        delta_mm[Y_AXIS] * inverse_millimeters,
        delta_mm[Z_AXIS] * inverse_millimeters,
        delta_mm[E_AXIS] * inverse_millimeters
      };
    #endif

    normalize_junction_vector(unit_vec);  // We have to normalize here before calculation the scalar product

    // Skip first block or when previous_nominal_speed is used as a flag for homing and offset cycles.
    if (moves_queued && !UNEAR_ZERO(previous_nominal_speed_sqr)) {
      // Compute cosine of angle between previous and current path. (prev_unit_vec is negative)
      // NOTE: Max junction velocity is computed without sin() or acos() by trig half angle identity.
      float junction_cos_theta = -previous_unit_vec[X_AXIS] * unit_vec[X_AXIS]
                                 -previous_unit_vec[Y_AXIS] * unit_vec[Y_AXIS]
                                 -previous_unit_vec[Z_AXIS] * unit_vec[Z_AXIS]
                                 -previous_unit_vec[E_AXIS] * unit_vec[E_AXIS]
                                ;

```

Here are my debug outputs for a zig zag line with a speed of up to 400mm/s and a travel acceleration of 4000mm/s²:
(Format: [`junction_acceleration`] [- `junction_cos_theta`] - `vmax_junction_sqr` - `previous_unit_vec` -> `unit_vec`)
```
G1 X150 Y20 F24000
  - vmax_sqr:0.00 - [0.80 ,0.60 ,0.00 ,0.00] -> [-0.73 ,-0.68 ,0.00 ,0.00]
 G1 X140 Y40
 Acceleration:  4000.00 - cos(theta): 0.88 - vmax_sqr:13.22 - [-0.73 ,-0.68 ,0.00 ,0.00] -> [0.32 ,0.95 ,0.00 ,0.00]
 G1 X160 Y80
 Acceleration:  4000.00 - cos(theta): -0.60 - vmax_sqr:338.89 - [0.32 ,0.95 ,0.00 ,0.00] -> [0.95 ,0.32 ,0.00 ,0.00]
 G1 X140 Y120
 Acceleration:  4000.00 - cos(theta): -0.60 - vmax_sqr:338.89 - [0.95 ,0.32 ,0.00 ,0.00] -> [0.32 ,0.95 ,0.00 ,0.00]
 G1 X160 Y160
 Acceleration:  4000.00 - cos(theta): -0.60 - vmax_sqr:338.89 - [0.32 ,0.95 ,0.00 ,0.00] -> [0.95 ,0.32 ,0.00 ,0.00]
 G1 X140 Y200
 Acceleration:  4000.00 - cos(theta): -0.60 - vmax_sqr:338.89 - [0.95 ,0.32 ,0.00 ,0.00] -> [0.32 ,0.95 ,0.00 ,0.00]
 G1 X160 Y240
 Acceleration:  4000.00 - cos(theta): -0.60 - vmax_sqr:338.89 - [0.32 ,0.95 ,0.00 ,0.00] -> [0.95 ,0.32 ,0.00 ,0.00]
 G1 X140 Y280
 Acceleration:  4000.00 - cos(theta): -0.60 - vmax_sqr:338.89 - [0.95 ,0.32 ,0.00 ,0.00] -> [0.32 ,0.95 ,0.00 ,0.00]
```
These travel moves are now extremely smooth and my printer no longer rattles like crazy when flying around the corners.

Please let me know if you'd like me to create a pull request or if you want to insert that line of code yourself.

Best regard and always smooth printing
  Thomas

HackingGulliver, 2019-01-08:
> Does CORE kinematic need an different handling?
Yes, that may be the issue here. On CoreXY the steppers are calculated like this:
```
stepper_a = cartesian_x + cartesian_y
stepper_b = cartesian_x - cartesian_y
```
Let's assume the print head should move one mm in each x and y direction. Then `stepper_a`will be 2 and  `stepper_b`will be 0.

In cartesian the length of the vector will be sqrt(x²+y²) i.e. 1.414 and in CoreXY sqrt(a²+b²) is 2.

I currently have no time to inspect the code, but if `inverse_millimeters` is calculated based on x and y and `delta_mm` is filled with CoreXY stepper coordinates then the wrong scaling factor will be used for initializing `unit_vec`. My initial debug outputs before my bug fix pointed in that direction. Maybe I will have time for a closer inspection in the evening.

Cheers
Thomas

formadi, 2019-01-10:
Thanks your effort !.. great !!

landodragon141, 2019-01-11:
You should submit this as a pull request.

InsanityAutomation, 2019-01-11:
If youre not sure how to do so, I can do it for you. Im working on another today regardless.

AnHardt, 2019-01-11:
The suggested fix is 
a.) just a workaround for an error somewhere else.
b.) burning a lot of cycles for all kinds of kinematics - even the already working ones. Internally a sqrt() is used. The workaround has to be limited to the kinematics effected by the error.

InsanityAutomation, 2019-01-11:
@AnHardt agreed, id have put it in a core #if and left it for awhile to let people grab easily.  No rush to merge it if the root cause can be found.

HackingGulliver, 2019-01-11:
@landodragon141 , @InsanityAutomation 
No problem. I can create the pull request. As this is related to CoreXY I added some other lines of code anyway.

boelle, 2019-02-21:
@HackingGulliver problem solved?

HackingGulliver, 2019-02-22:
@boelle Yes, for me the problem is solved. I close this bug.

closed 2019-02-22 by HackingGulliver